myhdl/myhdl #222: VCD trace files only support 1ns timescale.

nfarring, 2017-05-02:
It appears that the current code to generate VCD trace files has hardcoded the value '1ns' as the timescale. Would be good to also support '1ps' timescales.

I have added a feature to allow the user to specify the timescale in the _Block.config_sim() call, e.g.

`inst.config_sim(trace=True, timescale='1ps')`

Will generate a pull request for this issue.

cfelton, 2017-05-02:
@nfarring, the 0.9 api supported user defined timescale and the new 0.10 api should as well ... at some point someone had a PR/proposal to add the rest of the sim options to the `config_sim` , it should work like the convert function.

As a work around you can `myhdl.traceSignals.timescale = '1ps'`.

Thanks for offering to generate the pull-request, we will keep our eyes out for it.

nfarring, 2017-05-02:
@cfelton, thank you for pointing out the workaround!