myhdl/myhdl #224: Translating to verilog fails when using ConcatSignal

arthur00, 2017-05-11:
When I attempted to translate one of my MyHDL designs, I got the following error:
"  File "/home/arthur/sim/myhdl/myhdl/conversion/_toVerilog.py", line 1627, in getName
    node.signed = _maybeNegative(node.obj)
  File "/home/arthur/sim/myhdl/myhdl/conversion/_toVerilog.py", line 1535, in _maybeNegative
    if hasattr(obj, '_min') and (obj._min is not None) and (obj._min < 0):
TypeError: unorderable types: member_descriptor() < int()
"

Upon close inspection, the issue happens because obj is the class ConcatSignal, which does have a **_min**, but it is a member_descriptor type. When trying to do obj._min < 0, you get a TypeError. The fix is to simply check whether obj is a class, and if it is, return False (unless I am missing something).

cfelton, 2017-05-12:
@arthur00 I think the correct way to do this is with the `signed` methods, same you would do with `concat` on `intbv`.

```python
x, y, z = [Signal(bool(0)) for _ in range(3)]
w = ConcatSignal(x, y, z)
w.signed()

Out[6]: intbv(0)
```

Regards,
Chris

arthur00, 2017-05-12:
Hi Chris, thank you for your quick reply! Let me clarify the issue a bit further: I am not using the 'signed' method of ConcatSignal, but rather getting an error when I try to convert my code that has a ConcatSignal to verilog.

I took some time to further narrow down the issue, and the problem happens on using a ConcatSignal for another ConcatSignal. Example:
```
@block
def mux(z, a, b, sel, WIDTH):
    x, y, zz = [Signal(intbv(0)[1:]) for _ in range(3)]
    w = myhdl.ConcatSignal(x, y, zz)
    @always_comb
    def combinational_logic():
        z.next = myhdl.ConcatSignal(w,Signal(intbv(0)[1:])[0])
        if sel == 0:
            z.next = a
        else:
            z.next = b
    return myhdl.instances()
```

I believe the reason is that the AST tries to read each argument and determine whether they are signed or unsigned, but when the argument is a ConcatSignal object **somehow** (not sure why) the ast visits the class ConcatSignal and tries to compare its __min_ to 0, ultimately failing with an exception. The above code is a modified version of the mux example, and generates the same exception as posted on original post.

Thank you,

--
Arthur Valadares

arthur00, 2017-05-12:
While my submitted pull request fixes this particular exception, I have not extensively tested whether ConcatSignal of a ConcatSignal correctly maintains the sign. It seems intuitive to me that concatenating signals - whether unsigned or signed - would generated an unsigned signal (which is what my patch ultimately does by returning False), unless indicated otherwise, but I might be overlooking use cases I am not aware of.

cfelton, 2017-05-12:
@arthur00 The `ConcatSignal` is only to be used outside of the myhdl generators (in elaboration) in the `concat` function is used.  This can be a little confusing.

arthur00, 2017-05-12:
That was not clear to me from the documentation. A ConcatSignal is a particular kind of SliceSignal, which  in turn is shadow signal. Would that mean shadow signals are forbidden in generators? They are very useful  and present in good part of my code. My understanding of shadow signals is that they behave like signals (great for serving as inputs to wire/registers) but are not real signals, thus, cannot receive values.

Thank you for your clarification, I am not an expert in digital design so I may be misusing these features. I would like to understand why using ConcatSignals in generators is confusing, if you wouldn't mind expanding a bit more on this affirmation.

cfelton, 2017-05-12:
[This probably explains it the best](http://docs.myhdl.org/en/stable/manual/structure.html#converting-between-lists-of-signals-and-bit-vectors).

To use shadows in a generator is not forbidden to create is.

arthur00, 2017-05-12:
I had read that document before, it is where I learned shadow signals. I read again, more carefully :)

I can see now that it is under Structural Modeling as opposed to RTL modeling (and thus, ConcatSignals were not meant to be used in generators). Thanks for the input! I would imagine this should have its own exception, as it works in simulation and fails in verilog translation with a weird exception. But that is beyond the issue at hand. Closing the issue down.

Thanks again!

closed 2017-05-12 by arthur00