myhdl/myhdl #226: Wrong substitution in verilog_code of memory arrays.

CaptainJK, 2017-05-12:
Hello.
Template substitution does not work as expected, when passing array:
```
from myhdl import *
from math import ceil

def _mem_init_gen(mem):
    @instance
    def mi():
        """Initialize memory for simulation here"""
        yield delay(0)

    return instances()

_mem_init_gen.verilog_code = \
"""
initial begin
    $$readmemb("mem_init_fname", $mem)
end
"""

if __name__ == '__main__':
    def tb():
        mem = [Signal(intbv(0)[8:]) for _ in xrange(256)]
        i = _mem_init_gen(mem)
        return i

    toVerilog(tb)
```
This results mem being substituted with [Signal(intbv(0)),Signal(intbv(0)),Signal(intbv(0))....].
First attempt was to subclass the list with custom_list and append _name to it's instance, which of course failed. A quick look through the code reveals, that memory and signals are processed separately and differently. Thereby I don't see easy conservative fix for the issue.

CaptainJK, 2017-05-12:
Well, here is my dirty fix on the issue:
_analyze.py
```
<             m.name = _makeName(n, prefixes, namedict)
---
>             if hasattr(m.mem[0], '_name'):
>                 m.name = m.mem[0]._name
>             else:
>                 m.name = _makeName(n, prefixes, namedict)
```
This assumes we use zeroth array element to contain memory name, which is ugly and in some way implicit. I guess, the consistent workaround would be to introduce new public subclass of list with _name attribute. This will also require fixing template substitution (as I understand now it simply parses whole namespace).

josyb, 2017-05-12:
I experimented with your code and think it is rather to do that a: *mem* is not used by any other logic and b: the converter has an issue with replacing *$mem*, whether *mem* is used or not.

cfelton, 2017-05-12:
@CaptainJK and @josyb, in general this won't convert, I have a ~~fix~~ change for it in my local repository and I will try and push it ... some day :)