myhdl/myhdl #229: Conversion exception

occoder, 2017-06-20:
Just copy the conversion example from 
http://docs.myhdl.org/en/stable/manual/conversion_examples.html
The "A small combinatorial design" example raised an exception.
I was using Python 3.6 on WIN10.

josyb, 2017-06-20:
It would have been helpful if you had copied the error message(s)

occoder, 2017-06-21:
The log is as follows.
C:\Users\www\AppData\Local\Programs\Python\Python36-32\python.exe D:/PycharmProjects/myhdl/example_conversion_comb.py
Traceback (most recent call last):
  File "D:/PycharmProjects/myhdl/example_conversion_comb.py", line 19, in <module>
    bin2gray_inst = toVHDL(bin2gray, B, G, width)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\conversion\_toVHDL.py", line 168, in __call__
    genlist = _analyzeGens(arglist, h.absnames)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\conversion\_analyze.py", line 164, in _analyzeGens
    v.visit(tree)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\ast.py", line 261, in generic_visit
    self.visit(item)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\conversion\_analyze.py", line 290, in visit_FunctionDef
    self.visitList(node.body)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\conversion\_misc.py", line 162, in visitList
    self.visit(n)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\conversion\_analyze.py", line 261, in visit_Assign
    self.visit(node.value)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\ast.py", line 263, in generic_visit
    self.visit(value)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\conversion\_analyze.py", line 264, in visit_Call
    if node.starargs:
AttributeError: 'Call' object has no attribute 'starargs'

Process finished with exit code 1

occoder, 2017-06-21:
I've also tried the "StopWatch" example.
http://www.myhdl.org/docs/examples/stopwatch/
And it raised conversion exception either.
C:\Users\www\AppData\Local\Programs\Python\Python36-32\python.exe D:/PycharmProjects/myhdl/StopWatch.py
Traceback (most recent call last):
  File "D:/PycharmProjects/myhdl/StopWatch.py", line 19, in <module>
    convert()
  File "D:/PycharmProjects/myhdl/StopWatch.py", line 16, in convert
    startstop, reset, clock = [Signal(bool(0) for i in range(3))]
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\_Signal.py", line 98, in Signal
    return _Signal(val)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\site-packages\myhdl\_Signal.py", line 126, in __init__
    self._init = deepcopy(val)
  File "C:\Users\www\AppData\Local\Programs\Python\Python36-32\lib\copy.py", line 169, in deepcopy
    rv = reductor(4)
TypeError: can't pickle generator objects

Process finished with exit code 1

NicoPy, 2017-10-03:
The examples on the MyHDL website are for the 0.9 version of MyHDL. This is the one you get using pip or another similar tool.
I guess you are using the git version of MyHDL which requires the use of @block decorator.