myhdl/myhdl #236: Conversion, two dififerent versions, preferably to be the same

joshuisken, 2017-09-27:
Hi,

Attached are two versions myhdl code which I think should generate identical code (1.0dev).
- f0.py gives:  ToVerilogWarning: Output port is read internally: i
- f1.py produces, in my opinion, the correct verilog/vhdl.

The differences between the two versions are the signal arguments to the generators.
In f0.py a signal is passed normally, whereas in f1.py the same signal is passed as dict.
For this example I prefer the result of f1.py

The two example files: [f0.py and f1.py](https://gist.github.com/joshuisken/5de43f79c2b111d9301fb0e0172d4345)

joshuisken, 2017-09-27:
Try it:
```
python3 f0.py
mv dut.v dut_f0.v
mv dut.vhd dut_f0.vhd
python3 f1.py
mv dut.v dut_f1.v
mv dut.vhd dut_f1.vhd
diff  dut_f0.v  dut_f1.v
```

josyb, 2017-11-01:
Actually none of the two produces valid VHDL (I don't do Verilog), both are missing the *clk* and *reset*.
If you re-write the ClkRst class as follows:
```
class ClkRst(object):
    def __init__(self):
        self.clk = Signal(bool(False))
        self.rst = ResetSignal(True, active=True, async=True)
```
The clock and reset signals appear in the entity's portlist.

In *f0.py* ypou supply *i* as an argument to the top level, as such it should and will appear in the portlist. You however hand this *i* to lower level routines once as input once as output, hence the signal will be promoted to *inout*.
In *f1.py* you package *i* in a dictionary.  Apparently this directory gets silently discarded by ```_AnalyzeTopFuncVisitor```.