myhdl/myhdl #237: Support to VHDL-AMS

fedeman, 2017-10-24:
Hi,
I would like to know if [VHDL-AMS ](https://en.wikipedia.org/wiki/VHDL-AMS) support is on the Roadmap?

josyb, 2017-10-24:
It currently isn't.
And as we have quite a backlog, it will take a while ...
What is your goal:
Use MyHDL to simulate (in-place) or use MyHDL to generate VHDL-AMS code to simulate in an external Simulator?

fedeman, 2017-10-24:
Thank you for the answer. 
Ideally would be nice to be independent of such expensive VHDL-AMS simulators and bring this language to the open source community. Maybe 2 phases could be introduced:

Phase 1: Generate VHDL-AMS code for external simulators
Phase 2:  Simulate in-place

This, of course, if phase 1 "helps" phase 2. If not, directly phase 2.

josyb, 2017-10-25:
I think 'simulate in place' in MyHDL using the power of Python is a possibility. Generating VHDL-AMS code for use with an external $$$ commercial simulator is pointless?
Can you show a (slightly) more elaborated example than that on the Wikipedia page? It should contain both digital and analogue processes.

fedeman, 2017-10-25:
Hi @josyb,
I prepared an example file (mixed_signal_example.vhd) with a testbench (tb_top,vhd), see attachment.
The example is from a nice presentation that you can consult [here](http://denverpels.org/Downloads/Denver_PELS_20071113_Cooper_VHDL-AMS.pdf).

This is a working code and produces the following output in a 100 µs simulation run:
[image]

We have here:

- constants
- digital signals
- terminals (i.e. analog ports)
- quantities (which requires ODE solver)
- entity instance
- processes with and without trigger

This list is quite complete, if this is simulable in myhdl would be great. Later functions can be introduced but I think that for python will be not difficult.

[test_mixed_signal.zip](https://github.com/myhdl/myhdl/files/1415005/test_mixed_signal.zip)

Let me know your thoughts.