myhdl/myhdl #245: Slicing not okay when multiple interface objects are present

rmuthukrishnan, 2017-11-13:
Hi All,

It appears slicing (shadow signals) does not work if there are multiple interface objects.
For the example below, the line "if a_slice == 3" is getting translated into verilog as "if (0 == 3)".
Could you check? It works if there is only one interface object.

[m_ex1.txt](https://github.com/myhdl/myhdl/files/1468874/m_ex1.txt)

from myhdl import *

# interface 1
class MyObj(object):
   def __init__(self):
     self.x = Signal(intbv(0)[8:])
     self.y = Signal(intbv(0)[4:])
     self.z = Signal(intbv(0)[9:])

# interface 2
class InObj(object):
   def __init__(self):
     self.a = Signal(intbv(0)[8:])

# module uses xyz and abc interface
def m_ex1(clk, xyz, abc):
   a_slice = abc.a(2,0)

   @always(clk.posedge)
   def hdl():
     # verilog output for a_slice comes out as 0
     if a_slice == 3:
         xyz.z.next = xyz.x + xyz.y
     else:
         xyz.z.next = xyz.x - xyz.y

   return hdl

# verilog conversion code
def convert():
    clk = Signal(False)
    xyz = MyObj()
    abc = InObj()
    toVerilog(m_ex1, clk, xyz, abc)

if __name__ == "__main__":
    convert()

Thank you,
Raman

rmuthukrishnan, 2017-11-13:
The version information of the myhdl I am using:
pip show myhdl

Name: myhdl
Version: 0.9.0
Summary: Python as a Hardware Description Language
Home-page: http://www.myhdl.org
Author: Jan Decaluwe
Author-email: jan@jandecaluwe.com
License: LGPL
Location: /usr/local/lib/python2.7/dist-packages

rmuthukrishnan, 2017-11-15:
I tried in myhdl.0.10dev, and the same issue is present.

With further debugging, I found that if the slice signal is only used as a condition with if-statement and is not used anywhere else, then the problem occurs. If the slice signal is also used in one of the assignments, then the problem does not occur.

The code below translates x_slice to 0 in verilog.
```
from myhdl import *
import pdb

class MyObj(object):
   def __init__(self):
     self.x = Signal(intbv(0)[8:])
     self.y = Signal(intbv(0)[4:])
     self.z = Signal(intbv(0)[9:])

@block
def m_ex1(clk, xyz):

   x_slice = xyz.x(2,0)

   @always(clk.posedge)
   def hdl():
     if x_slice == 3:
         xyz.z.next = xyz.y
     else:
         xyz.z.next = xyz.z

   return hdl

@block
def test():
    clk = Signal(False)
    xyz = MyObj()
    dut = m_ex1(clk, xyz)
    dut.convert('verilog')
    return dut

if __name__ == "__main__":
    inst = test()
```
Verilog output:
```
// File: m_ex1.v
// Generated by MyHDL 1.0dev
// Date: Wed Nov 15 12:55:44 2017

`timescale 1ns/10ps

module m_ex1 (
    clk,
    xyz_y,
    xyz_x,
    xyz_z
);

input clk;
input [3:0] xyz_y;
input [7:0] xyz_x;
output [8:0] xyz_z;
reg [8:0] xyz_z;

always @(posedge clk) begin: M_EX1_HDL
    if ((0 == 3)) begin
        xyz_z <= xyz_y;
    end
    else begin
        xyz_z <= xyz_z;
    end
end

endmodule
```

rmuthukrishnan, 2017-11-15:
The above problem is there only when interface is defined. If interface is not used, then the problem does not happen.
```
from myhdl import *
import pdb

@block
def m_ex1(clk, xyz_x, xyz_y, xyz_z):

   x_slice = xyz_x(2,0)

   @always(clk.posedge)
   def hdl():
     if x_slice == 3:
         xyz_z.next = xyz_y
     else:
         xyz_z.next = xyz_z

   return hdl

@block
def test():
    clk = Signal(False)
    xyz_x = Signal(intbv(0)[8:])
    xyz_y = Signal(intbv(0)[4:])
    xyz_z = Signal(intbv(0)[9:])
    dut = m_ex1(clk, xyz_x, xyz_y, xyz_z)
    dut.convert('verilog')
    return dut

if __name__ == "__main__":
    inst = test()
```
Verilog output:
```
// File: m_ex1.v
// Generated by MyHDL 1.0dev
// Date: Wed Nov 15 13:55:14 2017

`timescale 1ns/10ps

module m_ex1 (
    clk,
    xyz_x,
    xyz_y,
    xyz_z
);

input clk;
input [7:0] xyz_x;
input [3:0] xyz_y;
output [8:0] xyz_z;
reg [8:0] xyz_z;

always @(posedge clk) begin: M_EX1_HDL
    if ((xyz_x[2-1:0] == 3)) begin
        xyz_z <= xyz_y;
    end
    else begin
        xyz_z <= xyz_z;
    end
end

endmodule
```

josyb, 2017-11-16:
Apparently there is a problem here.
As a workaround, avoid using slices of top-level interface ports.
```
#     x_slice = xyz.x(2,0)
    
    @always(clk.posedge)
    def hdl():
        if xyz.x[2:0] == 3:
            . . .
```