myhdl/myhdl #247: Error in always_comb when internal signal used as inout

Zylanx, 2018-02-20:
Always_comb will except when an internal signal is used both as in and out in the same process, even though this is entirely valid in VHDL. (No idea about Verilog)

The workaround I have used is to just split it off into _another_ always_comb

cfelton, 2018-02-20:
I am fairly sure this was intentional design decision to avoid latches, it is only an issue when you have a complex set of statements, e.g. use one comb signal as a conditional for another.  Yes, the correct design pattern would be to use multiple `always_comb` statements.

josyb, 2018-02-20:
It doesn't avoid latches. Nor will it avoid loops. Not even when using multiple ```@always_comb``` blocks.
We've had this discussion before (I just don't remember where and when).
I suggested removing the exception, at least for VHDL (I don't know about Verilog either) (I did in my local code)

Zylanx, 2018-02-20:
I'm likely going to do some testing and thinking myself to see if it is worth changing. It could be that changing would add a lot more analysis during the elaboration, due to edge cases (this could even be considered an edge case)

cfelton, 2018-02-20:
"helps" but not sufficient.

josyb, 2018-02-21:
@cfelton It was there early on, so our BDFL @jandecaluwe made that decision. I opted to remove it in some discussion somewhere but it didn't happen.
@Zylanx taking out the check doesn't influence the analysis nor the elaboration (IMHO)

NicoPy, 2018-03-20:
I also had this exception with trivial assignments. I solved the problem by splitting the always_comb statement in two. Like Josy, I don't understand to purpose of this limitation.

josyb, 2018-03-20:
It is an issue when the signal is a actually a port in a top-level file; this changes the direction from *out* to *inout*. Although making it a *buf* was the indicated solution. As of VHDL2008 this can remain an *out* and we can remove the check. For lower level blocks/modules/functions this is never an issue as the *offending* MyHDL signal will become a local VHDL signal .
I have no idea how Verilog feels about this all, and it may well be inherited from Verilog (as Verilog was the guideline when our BDFL created MyHDL)

NicoPy, 2018-03-21:
@josyb Maybe we can narrow the exception for signals used as ports.