myhdl/myhdl #248: Enum is massively flawed (In VHDL at least)

Zylanx, 2018-02-21:
So, the enum type MyHDL provides is supposed to be better than just assigning a range to a set of constants.
However, the way enum has been implemented and the way the synthesis is done, once you choose enum you are locked into using it.
Now how is this a problem?
Say you have a rom storing instructions. if part of that instruction contains something that you want to represent with an enum, then you can't really convert it around. You can't subslice the signal and convert it to an enum. It stays as an intbv or whatever the data type being used is.

The VHDL way is to use either enum'val or enum'pos for converting from and to enums. This would be perfect because then you don't have to worry about hard coding values, it is all handled by the one enum statement. MyHDL does not allow that.

I however don't have enough knowledge of the internals in order to say, but I imagine if enums allowed indexing with integers then at least half of this problem would basically go away.

josyb, 2018-02-21:
The MyHDL ```enum``` is designed for writing state-machines. If you look it up in the [reference section of the documentation ](http://docs.myhdl.org/en/stable/manual/reference.html?highlight=enum#myhdl.enum) this becomes a bit clearer, perhaps.

There are other ways to achieve what you want using (standard) Python constructs.

P.S. It would have been nice if you did add your code to see what you are trying to achieve.
P.S. Because something doesn't do what you want doesn't mean it is massively flawed,
P.S. Also, the[ MyHDL Discourse forum](http://discourse.myhdl.org) is better place for this kind of discussion.

Zylanx, 2018-02-21:
Okay, that makes a _lot_ more sense now. I think most of my confusion came from wanting to have a mapping in a singular location. And seeing the enum, well I assume it is pretty obvious.
Also, yes. That would be a much better place to discuss it. I just didn't even realise it was there.

As for the rest. I was tired at the time.
We all occasionally have those moments of tiredness-induced blindness and stubbornness. If it was any other time, I would have noticed.

Gonna close the "issue"

Sorry about any annoyance this may have caused

closed 2018-02-21 by Zylanx