myhdl/myhdl #249: Support for SystemVerilog?

dj-on-github, 2018-02-22:
To be useful to me I need to be able to output to SystemVerilog instead of Verilog. I've added support for SV here: https://github.com/dj-on-github/myhdl which is working with basic examples but needs more extensive testing.

josyb, 2018-02-22:
I think that adding a ```standard``` attribute in the _toVerilog code would be better? It avoids unnecessarily duplicating of code.

dj-on-github, 2018-02-22:
That would work also. It would remove the many points around the codebase where I had to add a SystemVerilog case of some Verilog related code. I'll give it a go.

cfelton, 2018-10-17:
I think this could be a good addition, and I like the idea of a conversion argument.  However (as can be seen by the delayed responses) we don't have a ton of resources to implement such features right now.

josyb, 2018-10-17:
It can be done gradually ...
First output an .sv instead of .v, then add a few SystemVerilog constructs in _toVerilog.py. and we are on our way.
SystemVerilog is an essential requirement if we add structured types: struct/record and multidimensional arrays, perhaps even *true* interfaces (although these don't have a VHDL counterpart ...)
I'm not volunteering, but still would be meddling :)

dj-on-github, 2018-10-17:
This has been on my todo list for a long time. I have my own version
working and I use it, but the maintainers want it done this other way for
the mainline. That’s fine and I’ll get to it the next time it matters to me
a lot if nobody else does.

josyb, 2018-10-18:
This means that you are actually using what you published on GitHub:  https://github.com/dj-on-github/myhdl ? That gives us good checking point.

dj-on-github, 2018-10-18:
I don't use it much anymore. My RTL writing days are mostly behind me. But
I have used it for many projects.