myhdl/myhdl #258: Lattice VHDL: max function not found

tehKaiN, 2018-03-19:
In python I do something like this:

``` python
def test(in1, out1):
    minPacketLength = 48
    def process():
        out1.next = max(minPacketLength, len(in1))
    return process;
```

In VHDL I get:

``` vhdl
out1 <= max(minPacketLength, 25);
```

latest Lattice Diamond says: `	2049990	ERROR - *pathWasHere* (VHDL-1241) max is not declared
`

I've corrected it to:

``` vhdl
function max(
	a: in integer;
	b: in integer
) return integer is
begin
	if a > b then
		return a;
	else
		return b;
	end if;
end function max;

-- somewhere else:
out1 <= to_unsigned(max(minPacketLength, 25), out1'length);
```

So I guess that MyHDL should export max function in same way as it exports reverse_bits - one for each use case.

As a side note, doing stuff like `out1.next = len(in1)` throws `(VHDL-1278) type unsigned does not match with the integer literal` and it can be hackfixed by `out1.next = len(in1) + 0`.

josyb, 2018-03-19:
You may want to take a quick look at slides 19 and 20 of [this presentation](https://www.slideshare.net/GuyEschemann/2014-all-programmabe-days-fpga-design-with-python)
Anyway, MyHDL doesn't handle the built-in ```max()``` function, because it is a software solution that doesn't fit hardware. 
IMHO MyHDL cannot directly export a```max```function because then the conversion to V* would have to spell it out for every case. Image the next situation: you feed it a list of 1024 Signals to later find out that the design fails timing, or even doesn't fit.
It would be a lot easier to help if you supply a completer extract of your actual code. We than understand you intentions better, and run it locally to test our suggestions.

The word *hackfixed*.
1. *hack*: an inelegant but effective solution
2. *fix*: a solution
So either *hack* or *fix* is redundant. I'd say *hack* expresses your message. 
Unless it applies to the programming language Hack by Facebook

tehKaiN, 2018-03-20:
I have one component passing a bit vector to another, and also number of bits which are actually filled (say 40 or 64 or ... or 128 bits). Second component is an SPI, which transmits data over MOSI and also receives 48 bytes over MISO, hence minimal amount of data transmitted through SPI - so that I can receive full packet.

To keep SPI component generic I'd rather enforce minimal send size in first component, so what I did is wroting `sendSize.next = max(trueSize, minSize)` where minSize is 48, ofc I can also do it with `if` or something but using max was way easier in my current code. If you don't want to support max it's fine with me, as long as V* converter would yell at me if it doesn't know what to do with it - I've used it in defining size of `intbv` and it expanded nicely, thus I thought it's legal everywhere, where it can be determined during conversion:

``` python
spiBfrSize = max(len(packet1), len(packet2), len(packet3))
bvSpiBfr = Signal(intbv(0, _nrbits=spiBfrSize))
```

As for _hackfix_, yeah, it's kinda wrong but I like that word and I'm almost certainly sure I've seen it in many places on the net. ;)

NicoPy, 2018-03-20:
> As a side note, doing stuff like out1.next = len(in1) throws (VHDL-1278) type unsigned does not match with the integer literal and it can be hackfixed by out1.next = len(in1) + 0.

This is fixed in PR#259

josyb, 2018-03-20:
@NicoPy Great!

tehKaiN, 2018-03-20:
Yup, I'm more of a software guy - doing most of work on microcontrollers and recently a small project on FPGA. I can't really share my code because it's something internal in my company, but relevant part is that I have 3 components which are generating data of different length to be sent over SPI. There's also a scheduler which takes an intbv from each of them in a cyclic manner and passes it to SPI component as well as number of bits to be sent. The scheduler has to have intbv output of size equal to the greatest input data, hence its declaration with size of max from 3 intbvs, cited by me in my last post, and also number of bits to send by SPI component. This is the data coming through MOSI.

One of input packets is only 3 bytes long, whereas recently we've added MISO packets coming to FPGA, which have 10 bytes of size - thus requiring transmission to be of at least 80 bits long. One fix to this would be to write:

``` python
minTransmissionLength = 80

# later in process:
if packet = tPacket.A:
  o_bvData.next[len(i_bvDataA):] = i_bvDataA
  o_bvSize = max(len(i_bvDataA), minTransmissionLength)
  packet.next = tPacket.B # keep state machine on walking
elif packet = tPacket.B:
  # etc.
```

This way the SPI component could remain without minimum transfer length logic, making it generic for future use. Also, original size of bitvector is never used, but mentioned in code in a way that a reader understands why it wasn't used. Putting o_bvSize equal to minTransmissionLength in one of cases would be a bit less intuitive and also less resistant to future enhancements - that single packet could get longer during later development, way after everyone in team forgot about that piece of code.

Hope it helps... somehow? ;)

NicoPy, 2018-03-21:
@tehKaiN You can precompute the sizes :
```python
minTransmissionLength = 80

MaxSizeA = max(len(i_bvDataA), minTransmissionLength)
MaxSizeB = max(len(i_bvDataB), minTransmissionLength)

# later in process:
if packet = tPacket.A:
  o_bvData.next[len(i_bvDataA):] = i_bvDataA
  o_bvSize = MaxSizeA
  packet.next = tPacket.B # keep state machine on walking
elif packet = tPacket.B:
  # etc.
```
This way, the max function is the Pyhon one and no conversion is done on it.

tehKaiN, 2018-03-21:
Yes, that's very simple and elegant way to solve this problem. I guess MyHDL could warn during conversion that this kind of function is used inside process with suggestion to try to precompute result outside process. My way of thinking is that generated V* code should always work and if it isn't generate a hint to user why.