myhdl/myhdl #261: @always_seq with multiple events other than reset?

tehKaiN, 2018-04-17:
There is one bigger and two minor thingies here but idk if creating 3 separate issues is a good thing in this case.

So I'm implementing DDR mode where data is sent on both clock edges. Unfortunately `@always_seq` only accepts 2 args (edge, rst) and I needed 3. Would be cool if it could be expanded to multiple events, something like `@always(evt1, evt2, ..., evtn, reset=rst)` but I guess it could introduce massive incompatibility. Perhaps a breaking change for 1.0 release?

For now I've helped myself with `@always(clk.negedge, clk.posedge, i_reset.negedge)` and I'm quite happy with it, but it converts to a bit ugly VHDL: `STUFF_PROCESS: process (clk, clk, i_reset) is`. It would be cool if MyHDL could remove that duplicate in sensitivity list. Or am I doing things the wrong way by passing both edges to `@always`?

Also, there's an inconsistency in the doc: [here](http://docs.myhdl.org/en/stable/manual/rtl.html#alternative-template) it says to use `if not reset:` but it generates `No proper edge value test` error. [Here](http://myhdl-list.narkive.com/er6zb7ed/no-proper-edge-value-test) it says to use `if reset == 0` and it fixes problem for me. I don't really know if it's doc typo or if this feature is simply missing in HDL converter.

josyb, 2018-04-18:
I would just use two ```@always_seq``` processes, one for the *negedge* and the second for the *posedge*, both without a reset.
I don't think that expanding ```always_seq``` to handle multiple events is a sound idea. I believe that separate clock domains should be kept separate.
I don't think that 0.10 has touched this.
Are you using this in an FPGA?

tehKaiN, 2018-04-18:
Yup, on Lattice one.

I've went with dual process too, but AFAIR there was an error about one signal driven by both processes because I also needed to generate clock in sync with data "unpacked" from DDR - so on posedge process I needed it to go high and on negedge - low. If I understand things correctly, I couldn't just feed later blocks with input clock and output data, but needed to do it through clock delay to keep it with sync with DDR-unpacked data. Now that I think of it, I could do a separate process for clock propagating, but I guess my single-process solution is way more compact and easier to understand. ;)

josyb, 2018-04-18:
I am confused.
I associate DDR with physical pins of the FPGA. And I just use  the vendor primitives to do the work.
Now it is always difficult to comment without seeing actual code.
Maybe we should make it a rule that you have to add relevant code when raising an issue or a support request (on http://discourse.myhdl.org/)

```

tehKaiN, 2018-04-18:
It's not really about DDR RAM where there are plenty solutions but [dual pumping](https://en.wikipedia.org/wiki/Double_data_rate) in general (here QSPI with DDR functionality which is supported on the other end by e.g. STM32L4). In FPGA, I'm trying to do as much in vendor-agnostic code as possible because I don't really like the idea of being too attached to given vendor/IDE/whatever. Also, I'm almost certain that not every vendor supplies primitives for implementing every DDR-based transmission protocols imaginable. :) Also I really like the current state of having 99% of code in MyHDL and only a tiny layer of VHDL coded by hand in my project for interfacing with .jed export or ActiveHDL sim (one with PLL IP and one without it).

I don't see if discussing this specific implementation is really necessary here. What I'd like to emphasize is, as `@always` construct allows describing VHDL processes in 1:1 and `@always_seq` is a shorthand for that, I think it's a bit crippled because it lacks ability to support even a bit more elaborate sensitivity list than reset and one event. If it is meant to be so, then I'm fine with it. I guess I'm just posting my feedback here.

josyb, 2018-04-18:
L, A and X supply plenty DDR primitives, and if you have a reasonably high data rate you are better of using these. I am a bit curious why the MyHDL generated DDR code work especially for the output function.
I thought a bit about the ``@always``` sensitivity list and getting rid of the double *clk* signal is easy to do. If only I could copy your code, so I don't have to make a test-case for it myself ...