myhdl/myhdl #262: @always does not correctly convert to VHDL

NicoPy, 2018-06-11:
The following code :
```python
from myhdl import block, Signal, always

@block
def flipflop(s, r, q):
    @always(s.negedge, r.negedge)
    def proc():
        if s == 0:
            q.next = 1
        else :
            #if r == 0 :
            q.next = 0
    return proc

s = Signal(bool(0))
r = Signal(bool(0))
q = Signal(bool(0))
ff = flipflop(s, r, q)       
ff.convert(hdl='VHDL', name="ff", path=".", initial_values=True)
ff.convert(hdl='VERILOG', name="ff", path=".", initial_values=True)
```
converts to :
```vhdl
...
FF_PROC: process (s, r) is
begin
    if (s = '0') then
        q <= '1';
    elsif falling_edge(r) then
        q <= '0';
    end if;
end process FF_PROC;
...
```
This is not correct. The falling edge on s is omitted.
Conversion to verilog seems to be ok.

josyb, 2018-06-11:
The VHDL conversion only accepts a single edge. It does some hocus-pocus with the first Signal in the sensitivity list, and because there are 2 or more Signals it skips something, and then I got lost ...
The conversion to Verilog  is OK because the edges are in the emitted ```@always()``` declaration.
Unfortunately in VHDL we need to add the ```clock_falling()``` or ```clcok_rising()``` inside the process body.

NicoPy, 2018-06-12:
In VHDL, we can model logic currently impossible to model with MyHDL (to my knowledge) because of the way rising/falling edges are specified. MyHDL modelling is more like verilog.
However, with VHDL, we can model logic à la verilog :
```python
FF_PROC: process (s, r) is
begin
    if falling_edge(s) or falling_edge(r) then
        if (s = '0') then
            q <= '1';
        else
            q <= '0';
        end if;
    end if;
end process FF_PROC;
```
But maybe the synthesiser rejects this way of writing ( I did not tried it).

josyb, 2018-06-12:
MyHDL is (unfortunately?) inspired by Verilog, VHDL conversion came much later.
The VHDL code you present looks plausible to me.
The next approach will generate close to what you want, but only for VHDL. :)
```
def flipflop(s, r, q):
    @always(s, r)
    def proc():
        if s.negedge:
            q.next = 1
        elif r.negedge:
            q.next = 0
        else:
            pass

    return proc
```
```
FF_PROC: process (s, r) is
begin
    if falling_edge(s) then
        q <= '1';
    elsif falling_edge(r) then
        q <= '0';
    else
        null;
    end if;
end process FF_PROC;
```
I have no idea  whether MyHDL simulates that as intended .

or:
```
    @always(s, r)
    def proc():
        if s.negedge:
            q.next = 1
        else:
            if r.negedge:
                q.next = 0
```
```
FF_PROC: process (s, r) is
begin
    if falling_edge(s) then
        q <= '1';
    else
        if falling_edge(r) then
            q <= '0';
        end if;
    end if;
end process FF_PROC;
```
```        if s.negedge | r.negedge: ``` 
does convert, but MHDL wraps a bool() around it which doesn't work if the arg already is a bool.
```        if bool((falling_edge(s) or falling_edge(r))) then ```

```        if s.negedge or r.negedge: ``` doesn't convert because MyHDL erroneously thinks that ```s.negedge``` and ```r.negedge``` are not bool (but they are ...)

josyb, 2018-06-13:
So: your original module simulates but doesn't convert; my approach converts but doesn't simulate.
Looks to me that someone should correct/enhance the conversion. Or the simulation?

Out of curiosity: what is your intention with this design, is it something you actually need in your work?

cfelton, 2018-06-13:
@josyb 

> MyHDL is (unfortunately?) inspired by Verilog,

That is not correct, if you read @jandecaluwe writings or look at the early newsgroup he has an fondness for VHDL, and other than the decorator names (which make sense for event-oriented description), there is much more influence on the design from VHDL than Verilog.  The Verilog conversion came first, simply because Verilog makes more sense as an intermediate format, and is easier to convert to.   

Especially on the internals of the simulator, which is evident from these: [Verilog's major flaw](http://insights.sigasi.com/opinion/jan/verilogs-major-flaw.html) and [VHDLs jewel crown](http://insights.sigasi.com/opinion/jan/vhdls-crown-jewel.html)

cfelton, 2018-06-13:
> The VHDL conversion only accepts a single edge

We should list this in the FAQ and conversion limitations, I will add it to my TODO list (if I can find the end of it).

josyb, 2018-06-13:
@cfelton 

> That is not correct, 

As an ASIC designer he was using Verilog? (which explains why toVerilog() came first; he had an actual use case for it)
I must admit that you are right, he modelled the Signal and the simulation after VHDL, but he chose to set the decorator names and the posedge, negedge as their direct Verilog equivalents? I re-read the original article in the LinuxJournal issue of Nov 2004 (which I should have a copy of at home).

And @jandecaluwe wrote/published those blogs after he joined Sigasi  (AFAIK, but what do I know?)

josyb, 2018-06-13:
> The VHDL conversion only accepts a single edge
> We should list this in the FAQ and conversion limitations, I will add it to my TODO list (if I can find the end of it).

One might actually want to use the both edges of the clock, say if one wants to model a DDR port.

cfelton, 2018-06-13:
>
@josyb sure but I don't know of a quick fix, I think this is a feature addition and I imagine it will take some time (time none of us seem to have, where is the myhdl foundation when you need it).

In the near-term I think we should add this to: http://docs.myhdl.org/en/stable/manual/conversion.html#known-issues

And rename it to *Known issues and limitations*.    Adding to the list I can do quickly, extending the conversion to more than one edge sensitivity, I cannot do quickly.

NicoPy, 2018-06-13:
@josyb 
> Out of curiosity: what is your intention with this design, is it something you actually need in your work?

I need an asynchronous FIFO. 
I'm trying to model one from these articles :
[http://www.sunburst-design.com/papers/CummingsSNUG2002SJ_FIFO1.pdf](url)
[http://www.sunburst-design.com/papers/CummingsSNUG2002SJ_FIFO2.pdf](url) 
For now, I translated verilog code given in the second pdf document to MyHDL and made some tweaking to let the MyHDL code simulate and convert.
These kind of function uses asynchronous logic like the one I have given in my original post. This is where problems arise. 
I see 3 possibilities :
- Fix VHDL conversion
- Fix simulation
- Modify the FIFO model to remove, if possible, the use of asynchronous logic.
First option is risky.  Even if conversion is fixed, I'm not sure synthesis will pass.
Last option is my preferred.

josyb, 2018-06-13:
@cfelton I'm sure that adding it to the list of *Known Issues and Limitations* will suffice for quite some time to come. I'm not planning on making a DDR controller in MyHDL, and if I were at least not the physical IO.
I also wonder how much merit the OP's edge SR-latch has?
The conversion looks not easy to me either, and what to do if one finds more than 2 edges in the sensitivity list?

josyb, 2018-06-13:
@NicoPy 
Asynchronous as in Dual Clocked FIFO or something truly asynchronous?
I found the paper you refer to.
I also needed a Dual Clocked FIFO and after searching and seeing those Gray counter issues, I decided to make it packet based:
You write a number of words, say 8, and then send a single pulse to the other clock domain to add to his packet-count.
As soon as the same number of words a read the other clock domain sends a pulse back.
Both clock domains maintain their own address generator and used-word counter.
This only works if  a complete transmission is a multiple of the number of words per packet, which in my line of work (digital cameras) is always the case. In the other case you may add a bit to the data to signal the end of transmission.
The only drawback is that the generated VHDL code is not recognised by Vivado as a *true* dcfifo, I doubt Altera/Intel will do better.

cfelton, 2018-06-13:
@NicoPy I probably modified, and I didn't convert to vhdl, and it's not fully tested - but I have used it.
I generated an async fifo based on one of those references (i forget which one).
https://github.com/cfelton/rhea/blob/master/rhea/cores/fifo/fifo_async.py

I believe it is the second because the second tries to address issues in the first.  I didn't find any limitations, given the disclaimer.

NicoPy, 2018-06-14:
@josyb 
>I found the paper you refer to.

I corrected the links, They are functional now.
There are other very interesting documents on [this page](http://www.sunburst-design.com/papers/).

>Asynchronous as in Dual Clocked FIFO or something truly asynchronous?

Dual Clocked FIFO.

I often use FIFOs (sync or async) in my designs. Async ones are especially useful when crossing clock domains. Like recording/playing video/audio data through PCIe DMA.

>The only drawback is that the generated VHDL code is not recognised by Vivado as a true dcfifo, I doubt Altera/Intel will do better.

Most of the time, the FIFO model I use (generated with Xilinx logicore) do not use integrated FIFOs but BRAMs with additional logic around them (integrated FIFOs are limited in functionality). 
We can achieve the same result with MyHDL since BRAMs are instantiated when arrays are found by synthesiser.

@cfelton I'll have a look at it. Thanks.

NicoPy, 2018-06-20:
@cfelton I now have a functional dual clock FIFO. I made a mix of my code and yours.