myhdl/myhdl #263: Verilog Testbench Conversion

GProtoZeroW, 2018-06-29:
Oky I am still new to testbenchs in verilog 
but I am trying to converty my in python testbench to a verilog testbench 

```
from myhdl import *
from myhdlpeek import Peeker
import pandas as pd

TwoBitAdderTT=pd.DataFrame()
TwoBitAdderTT['x1']=[0,0,1,1]
TwoBitAdderTT['x2']=[0,1,0,1]
TwoBitAdderTT['Sum']=[0,1,1,0]
TwoBitAdderTT['Carry']=[0,0,0,1]
TwoBitAdderTT

@block
def HalfAdder(x1, x2, Sum, Carry):
    """Half adder in myHDL"""
    @always_comb
    def logic():
        Sum.next=x1^x2
        Carry.next=x1 & x2
    return logic

###########
# in python testbench

Peeker.clear()
x1=Signal(bool(0)); Peeker(x1, 'x1')
x2=Signal(bool(0)); Peeker(x2, 'x2')
Sum=Signal(bool(0)); Peeker(Sum, 'Sum')
Carry=Signal(bool(0)); Peeker(Carry, 'Carry')

DUT=HalfAdder(x1, x2, Sum, Carry)

def HalfAdder_TB():
    
    @instance
    def Stimules():
        for _, row in TwoBitAdderTT.iterrows():
            x1.next=row['x1']
            x2.next=row['x2']
            yield delay(1)
        raise StopSimulation()
    
    return instances()

sim = Simulation(DUT, HalfAdder_TB(), *Peeker.instances()).run()

################
# Verilog testbench

@block
def HalfAdder_TBV():
    x1=Signal(bool(0)); x2=Signal(bool(0))
    Sum=Signal(bool(0)); Carry=Signal(bool(0))

    DUT=HalfAdder(x1, x2, Sum, Carry)
    
    @instance
    def Stimules():
        for _, row in TwoBitAdderTT.iterrows():
            x1.next=row['x1']
            x2.next=row['x2']
            yield delay(1)
        raise StopSimulation()
    
    return instances()

#create instaince of TB
TB=HalfAdder_TBV()
#convert to verilog with reintilzed values
TB.convert(hdl="Verilog", initial_values=True)
```

and I am getting all sorts of error with everything I have tried.  Here are the things I have tried
1.  Exstact x1, x2 to a list and change the for loop to a zip over list 
2. Change from a for loop to a while loop add an extra signal to then iteration contorl of the list 

So what do I need to do get this susfuly convert to verilog, thanks guy

GProtoZeroW, 2018-06-29:
also, I know I forgot to add the print out in the Verilog version sorry

josyb, 2018-06-29:
Steven,

this (heavily simplfied) code will convert to valid Verilog, if only the *initial_values* would work :(
```python
@block
def HalfAdder_TBV():
    x1 = Signal(bool(0))
    x2 = Signal(bool(0))
    Sum = Signal(bool(0))
    Carry = Signal(bool(0))

    DUT = HalfAdder(x1, x2, Sum, Carry)

    testx1 = [Signal(bool(val)) for val in [0, 0, 1, 1]]
    testx2 = [Signal(bool(val)) for val in [0, 1, 0, 1]]
    testsum = [Signal(bool(val)) for val in [0, 1, 1, 0]]
    testcarry = [Signal(bool(val)) for val in [0, 0, 0, 1]]
    
    @instance
    def Stimulus():
        for i in range(len(testx1)):
            x1.next = testx1[i]
            x2.next = testx2[i]
            yield delay(1)
            
        raise StopSimulation()
    
    return instances()

# create instaince of TB
TB = HalfAdder_TBV()
# convert to verilog with reintilzed values
TB.convert(hdl="verilog", initial_values=True)
```
Conversion to VHDL is only slightly better as the initial values work but the values are wrong; *1X"False"* isn't a valid initialiser ...

GProtoZeroW, 2018-06-30:
Thanks, Josy, I would add that myHDL needs to have the testbench converter pull the values into the list from outside the testbench def, cause hand typing values into a list NOOO

GProtoZeroW, 2018-06-30:
@josyb So took your fix and moded to get what I neded
```
@block
def HalfAdder_TBV():
    x1 = Signal(bool(0))
    x2 = Signal(bool(0))
    Sum = Signal(bool(0))
    Carry = Signal(bool(0))

    DUT = HalfAdder(x1, x2, Sum, Carry)

    testx1=Signal(intbv(int("".join([str(i) for i in TwoBitAdderTT['x1'].tolist()]), 2))[4:])
    testx2=Signal(intbv(int("".join([str(i) for i in TwoBitAdderTT['x2'].tolist()]), 2))[4:])

    
    @instance
    def Stimulus():
        for i in range(len(testx1)):
            x1.next = testx1[i]
            x2.next = testx2[i]
            yield delay(1)
            
        raise StopSimulation()
    
    @always_comb
    def print_data():
        print(x1, x2, Sum, Carry)
    
    return instances()

# create instaince of TB
TB = HalfAdder_TBV()
# convert to verilog with reintilzed values
TB.convert(hdl="verilog", initial_values=True)
```

josyb, 2018-06-30:
@GProtoZeroW 
I see that you coerced the list of 4 bits into a 4-bit vector. This probably won't work once you have a list of vectors instead of a list of bits ... I don't know Verilog well (probably never will) but a quick search tells me initialisation is somewhat contrived
Regards,
Josy