myhdl/myhdl #265: Cannot run myhdl cookbook example on Windows 10

EvgenyMuryshkin, 2018-07-09:
Hi,
I am trying to run StopWatch example and having some issues
What I did is:
1. Download and install Python 3.7
2. Installed MyHDL via pip install myhdl
3. Run any test or examples and get next error in console
[image]

4. Try to install from github using setup.py, got next errors
[image]

[image]

[image]

What am I doing wrong?

Thanks,
Regards,
Evgeny

josyb, 2018-07-09:
You have tripped over the *async* reserved keyword in Python 3.7 which conflicts with the ```async=``` argument in MyHDL's ```ResetSignal()```
Currently the best option is to revert to Python 3.6

cfelton, 2018-10-04:
We need a FAQ for this :)

closed 2018-10-04 by cfelton