myhdl/myhdl #266: Incorrect generation of array assignments for VHDL

xd009642, 2018-08-02:
Below is a relevant section of the python I use to generate VHDL and the generated VHDL. The key point is that where assignments should `<=` they are instead `:=`. This only seems to affect arrays from what I can see.

```python
    x = [Signal(intbv(0)[phasebits+1:]) for i in range(stages)]
    y = [Signal(intbv(0)[phasebits+1:]) for i in range(stages)]
    p = [Signal(intbv(0)[phasebits+1:]) for i in range(stages)]

    @always(clk.posedge)
    def update():
        if rst == 0:
            y[0] = 0
            x[0] = 0
            p[0] = 0
```

```VHDL
type t_array_y is array(0 to 8-1) of unsigned(15 downto 0);
signal y: t_array_y;
type t_array_x is array(0 to 8-1) of unsigned(15 downto 0);
signal x: t_array_x;
type t_array_p is array(0 to 8-1) of unsigned(15 downto 0);
signal p: t_array_p;

begin

CORDIC_GEN_UPDATE: process (clk) is
    variable temp_phase: integer;
begin
    if rising_edge(clk) then
        if (rst = '0') then
            y(0) := to_unsigned(0, 16);
            x(0) := to_unsigned(0, 16);
            p(0) := to_unsigned(0, 16);
        end if;
    end if;
end process CORDIC_GEN_UPDATE;
```

josyb, 2018-08-02:
Add the '.next' like y[0].next =0

xd009642, 2018-08-02:
Thank you :+1: should this error maybe generate a failure to convert to VHDL? The synthesis tool threw out errors when it was confronted with the generated core. Or are there situations where this would be the desired behaviour?

josyb, 2018-08-03:
You are right that MyHDL should flag the error.
Our BDFL does like variables and and as such MyHDL has provisions to infer them; the assignment without the ```.next``` is one of the use cases, but apparently there is work to be done.
I almost never (never say never?) use variables.