myhdl/myhdl #272: error in myhdl\_always_seq.py

chb111, 2018-09-20:
Hi, 

I am new with myHDL and relatively new with Python.
I have just installed myHDL.

I tried to run the firfilt example in Christopher Felton's github repo:
https://bitbucket.org/cfelton/examples/src/tip/firfilt/

I get the following error: 
..\cfelton-examples\firfilt>python firfilt.py

Traceback (most recent call last):
  File "firfilt.py", line 2, in <module>
    from myhdl import *
  File "C:\python37\lib\site-packages\myhdl\__init__.py", line 162, in <module>
    from ._always_seq import always_seq, ResetSignal
  File "C:\python37\lib\site-packages\myhdl\_always_seq.py", line 48
    def __init__(self, val, active, async):
                                        ^
SyntaxError: invalid syntax

This error appears when executing the following line:
from myhdl import *

Any clue?

Python ver: 3.7.0
myHDL ver: 0.10

Thanks, 
Christian

chb111, 2018-09-21:
Actually, this has nothing to do with the examples in the Christopher Felton's github repo. The error occurs just when executing "from myhdl import *" from Python.

josyb, 2018-09-21:
Python 3.7 promoted ```async``` to *reserved keyword*, and this causes the conflict. You have two options:
1. Revert to Python 3.6
2. (preferred) Update your MyHDL package tot the latest commit. In your (or any) myhdl code replace ```async``` by ```isasync``` . See PR #267

closed 2018-10-04 by cfelton