myhdl/myhdl #273: Failed to make myhdl.vpi based on Makefile of Icarus Verilog

isanmu, 2018-09-29:
### Environment:
Icarus Verilog : 10.1(stable)
MyHDL : 0.10
Make: i686-pc-mingw32 (mingw32-make)
OS: windows7-64bit

### First Try:
Just type 'make' as "README" said.
Cause errors as bellow:
```cmd
iverilog-vpi myhdl.c myhdl_table.c
Compiling myhdl.c...
gcc: fatal error: input file 'myhdl.c' is the same as output file
compilation terminated.
Compiling myhdl_table.c...
myhdl_table.c:3:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
  extern void myhdl_register():

myhdl_table.c:5:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
  void (*vlog_startup_routines[])() = {
iverilog-vpi: 1 file(s) failed to compile.
Makefile:4: recipe for target 'myhdl.vpi' failed
make: *** [myhdl.vpi] Error 2
```

### Second Try:
I doubt that "iverilog-vpi file1.c file2.c" is not supported in 10.1 version. 
So, i try iverilog-vpi myhdl.c directly, but resulting some "makefile undefined reference to vpi_xxx" error, that means the vpi_user.h was not included correctly, meanwhile, transfer code in myhdl_table.c to myhdl.c  also bring some conflicts.

### Request
Could you just provide myhdl.vpi in git since so many problems may happen due to different environments? 
Or how can i solve these problems?  degradation which program?

cfelton, 2018-10-02:
Unfortunately it is not feasible for us to provide a vpi for each platform.

I have not tried to build it with mingw without cygwin or some other posix emulator.  

I compiled with 10.1 on my system fine.

isanmu, 2018-10-02:
@cfelton 
What emulator did you use to compile? And your os system is windows or Unix? I want try to build it again.

cfelton, 2018-10-04:
I used iverilog 10.1 on osx, I might be able to try on a windows machine.  I use a windows machine quite a bit for development at work, however I use modelsim and I run cosimulation on a unix machine (centos or redhat usually).

At one point someone started implementing a socket version with the hopes of more windows friendly and other reasons - but I believe they abandoned the effort.  Maybe @josyb has tried cosim on windows recently?

josyb, 2018-10-04:
Actually I haven't tried cosim on Windows yet. And I am actually trying to get a Linux machine set up for development work on  MyHDL, as I even't can get the testsuite running properly.

isanmu, 2018-10-18:
@AngelTerrones 
Thanks for replying.
What version of iverilog did you use? I will try in cygwin later, thanks gain!

AngelTerrones, 2018-10-18:
@isanmu the master branch.

isanmu, 2018-11-09:
@AngelTerrones 
I tried cygwin and gcc provided by it, if version of iverilog is larger than 0.7, "myhdl.vpi" still can not be built , and only been built successful in  iverilog-0.7.  At last,  I 'd better try other way to verify verilogHDL code...

simux, 2019-11-22:
I have the same issue:

$ make
iverilog-vpi myhdl.c myhdl_table.c
gcc: fatal error: input file 'myhdl.c' is the same as output file
compilation terminated.
myhdl_table.c:3:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
    3 | extern void myhdl_register();
      | ^~~~~~
myhdl_table.c:5:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
    5 | void (*vlog_startup_routines[])() = {
      | ^~~~
Compiling myhdl.c...
Compiling myhdl_table.c...
iverilog-vpi: 1 file(s) failed to compile.
make: *** [Makefile:4：myhdl.vpi] error 2

Windows 10 Ver.1903
MSYS2 with minGW64
myhdl 0.11
icarus 10.3
python 3.7.4

huangqi412, 2020-02-18:
i have the same problem, and finally I gave up.

DavidC-75, 2022-02-06:
This is a bug in `driver-vpi\main.c`
For a simple fix, see https://github.com/steveicarus/iverilog/issues/602