myhdl/myhdl #276: Name mangling Function and Procedure

RPeschke, 2018-10-03:
Hello great product! 
I would like to be able to use classes (list of signals) also with function and procedures. Is there any change to implement the name mangling scheme which works fine for entities also for function etc?

cfelton, 2018-10-04:
you can use a generator in a class / object.  The only existing limitation is a function, myhdl only support combinatorial function conversion, similar to verilog.

josyb, 2018-10-07:
If I read you well, you want to use e.g. ```AxisStreamInt``` as an argument to a function? I gave it a quick try and it won't convert ...
```python
class AxisStreamInt:

    def __init__(self):
        self.tvalid = Signal(bool(0))
        self.tlast = Signal(bool(0))
        self.tready = Signal(bool(0))
        self.tdata = Signal(intbv(0)[8:])

@block
def Reverse(AxIn, AxFeatureOut):

    def _reverse(AxiIn, AxiOut):
        TDATABITS = len(AxiIn)
        for i in range(TDATABITS):
            # this doesn't convert
            AxiOut.tdata.next[i] = AxiIn.tdata[TDATABITS - 1 - i]
            # this does
#             AxiOut.next[i] = AxiIn[TDATABITS - 1 - i]

    @always_comb
    def assign():
        AxiOut.tvalid.next = AxiIn.tvalid
        AxiOut.tlast.next = AxiIn.tlast
        AxiIn.tready.next = AxiOut.tready
        # this doesn't convert
        _reverse(AxIn, AxFeatureOut)
        # this does
#         _reverse(AxIn.tdata, AxFeatureOut.tdata)

    return assign

if __name__ == '__main__':
    AxiIn = AxisStreamInt()
    AxiOut = AxisStreamInt()

    dfc = Reverse(AxiIn, AxiOut)
    dfc.convert(hdl='VHDL')
```
giving this error:
```
myhdl.ConversionError: in file C:\Users\josy\myhdlsupport\issues\rpeschke\intffunc.py, line 25:
    Unsupported attribute: tdata
```
In contrast with @always decorated functions the AST tree of a function is not operated upon by the Attribute Transformer (which makes the int_signal out of intf.signal) and subsequently  the attribute is not found.

Can you share more of your code? We may give you some better help.

RPeschke, 2018-10-07:
Hello.

Please find the example at my github page:
https://github.com/RPeschke/VHDL_Feature_extraction/blob/master/axiStreamTest.py

It is fine to use the signals as a function argument but as soon as I try to use the entire object it fails:
https://github.com/RPeschke/VHDL_Feature_extraction/commit/d9d7eff78ccd3c54c992e74fd93d4ecf6f44bd68#diff-21c49a0840dd1bc4fe24a5484947f663

I think the main issue is that myhdl has two different methods for converting function arguments. For example an ordinary function gets translated at:

https://github.com/myhdl/myhdl/blob/e22840209398e09fbf1d0ee7ba1931946d3c854b/myhdl/conversion/_toVHDL.py#L1962

while the entity conversion is written completely independent of what is used for function and procedures:

https://github.com/myhdl/myhdl/blob/e22840209398e09fbf1d0ee7ba1931946d3c854b/myhdl/conversion/_toVHDL.py#L336

I think whatever is done for entities needs a bit of cleanup and then should be the default for all calls. 

BTW. Do you know why my two python objects (AxiStimilus, FindPeak) end up in one entity? 

Thanks

josyb, 2018-10-08:
HI,

If I read everything well; the objective is to monitor an AxiS stream passing by and finding the maximum value in each *packet* and forwarding this maximum via an outgoing AxiS stream.
I notice that you are ultimately converting both the target logic and the test-bench in one go. Is this your intention?
This also answers your question why both AxiStimulus and FindPeak end up in a single entity.
Most of us simulate the design in Python and only convert the target logic, not the test-bench.
Some of us (excluding me) also run co-simulation and in that case the test-bench must be convertible too.
Note that the *problem point* with the function is only in the test-bench, so there will be no issue simulating as that is running pure Python.

Regards,
Josy

RPeschke, 2018-10-09:
Hi,

So if I understand this correctly that means that everything I export with one call to the “toVHDL” function will end up in one entity and there is nothing I can do to change that behavior? 

 

> Note that the problem point with the function is only in the test-bench, so there will be no issue simulating as that is running pure Python. 

Don’t pay to much attention on where I called the function. I wanted to have the function exported to the vhd file. 
Basically what I am trying to achieve is getting some sort of OOP style abstraction in my VHDL code.  I think what is done with entities is already a good start but from my point of view this functionally needs to be extended to also include function/procedures. 

Do you see any possibility that this feature gets added to myHDL?

Cheers,
Richard

josyb, 2018-10-09:
Hi Richard,

MyHDL flattens the hierarchy; so everything ends up in a single entity. This is only an issue when you want (need) to explore the schematic in Quartus/Vivado after compilation.  We are discussing to implement hierarchical conversion, but it will take time before we really get there.

IMHO in  MyHDL design you don't often, or I'd better say rarely, need functions and I wouldn't say that functions/procedures will give you OOP abstraction. 
Perhaps you want to take a look at these gists: 
https://gist.github.com/josyb/d119a2f3f12dcc28533ac612e30576af
https://gist.github.com/josyb/2e43c9ad7ffa772d52dfd66cb660dc4a
The first is *class-based* design where the second is true OO but of rather academic value only and not so practical.

Adding that feature to properly access class attributes in function / procedures definitely is doable, but I wonder where it will land in the priority ranking and whether we find a volunteer.

Regards,
Josy

RPeschke, 2018-10-16:
Hi,

Thanks for the answer and the links. It will take me some time to fully understand this but it looks really interesting. 
When I discovered myHDL I was thinking I can write my entire Firmware in python and just export it to vhdl. But without these features (name Mangling/ deep hierarchies) it will not be possible. Which is a pity since I really enjoyed working with the lib. 

But anyway Thanks for your help.

Cheers,
Richard

josyb, 2018-10-16:
Hi Richard,

>But without these features (name Mangling/ deep hierarchies) it will not be possible.

I do not readily see why the flattening of  the hierarchy would be a show-stopper. E.G. I have a large (class based) MyHDL design that generates a 30,000 line VHDL output which eventually also consumes some 30,000 LUT. 

Regards,
Josy

RPeschke, 2018-10-17:
Hi,

Maybe it’s just my lack of understanding but what I wanted to do was writing more or less the entire firmware in myHDL and then convert it to VHDL. From how I understand it now I have basically two options: 
Either, I write the firmware in myhdl and convert it to VHDL which would result in one GOD entity which handle all the signals and all the logic. I assume it would easily become so big that it is basically write only. Which means I can’t mix myhdl generated code with handwritten VHDL. 

Or, I am converting only small sections of myHDL code into individual entities and then connect them manually in VHDL.  

The second solution sounds to me more practical but it seems like there is a lot of extra work involved which makes it unhandy to use. 

I would be very happy if you have a better way of doing it.

Thanks for your help.

Cheers,
Richard

lomax, 2018-10-17:
Maybe i'd say something obvious, but VHDL and so myHDL is the Hardware and not Firmware description language.

RPeschke, 2018-11-18:
@josyb Thanks again for your examples. Maybe if I give you some more examples it will become clearer.

First let me start with the general overview over all my programs. I like to use a very simple map in which I give every entity (not VHDL entities but rather objects, components or symbols) a dedicated role.

[image]

that means it can either be responsible for a given recourse. A good example would be a file Handler, or a memory handler. something that makes sure that a given recourse is always in a valid state (or reports an error). 

 the next is a processor. It takes data in and processes. For example a sort function is a processor. 

and the last one is the data object. its purpose is to store and move information in the program.  

now what i want to be able to do, is write an axistream handler (for example) that allows me to work with it like this:

[image]

I have created this little code snipped which seems to work but it is a lot to write and since VHDL does not allow me to write generic code i have to rewrite it every time i am going to change something. 

```vhdl 

entity master is 
  port(
    clk : in sl;
    -- Outgoing response
    fromMaster : out  AxiMonoFromMaster_t;
    -- Incoming data
    toMaster   : in AxiMonoToMaster_t
  );
end master;

architecture rtl of master is 

begin
  seq : process(clk) is
    variable RXTX : AxiMonoSendReceiveMaster ;
   

  begin
    if (rising_edge(clk)) then

      AxiMonoMasterPullData(RXTX, toMaster);

      if txIsDataReady(RXTX) then 
        txPushData(RXTX, 1,0);
        txPushData(RXTX, 2,1);
        txPushData(RXTX, 3,2);          
        txPushData(RXTX, 4);      
        txPushData(RXTX, 5);      		
        txPushData(RXTX, 6);      
        txPushLast(RXTX);
      end if;

      AxiMasterPushData(RXTX, fromMaster);

    end if;
  end process seq;

end rtl;
```

the full code is at  [axiStream](https://github.com/RPeschke/AxiStream)

If i would have better support for functions/procedures i would be able to write these Handlers in python and it would automatically determine which signal it needs to use in this function/procedure and that would get me a huge step closer to object oriented programming. 

I hope this is helpful for you. and thanks again for the examples

josyb, 2018-11-18:
In the second picture there is no need for an Axi Stream Handler between Entity A and Entity B; it is the AxiStream object which connects the two together; code to 'push' is in Entity A and code to 'pull' is in Entity B

> I have created this little code snipped which seems to work but it is a lot to write and since VHDL does not allow me to write generic code i have to rewrite it every time i am going to change something.

Surely VHDL allows you to write generic code, and I am sure you can  reduce the writing a lot.

I perused your axiStream project for over an hour but still failed to understand what it does, gotten lost in functions/procedures chaining on to more functions/procedures.

If I read the testbench well it has two unrelated streams: one between master and slave, and another between a csv reader and a csv writer. The ,txt files are not included but are secondary anyway.

About subtypes; it is not because you can use them that you must; a subtype is only warranted when it actually adds knowledge:
  ```subtype data_t is integer;```
doesn't improve
but 
  ```subtype fingercount_t is integer range(1 to 10);```
can be useful
If you are using subtypes to reduce the typing work (as you have 'slv' for 'std_logic_vector'); I suggest you find a better editor that offers word-completion (I use Eclipse with PyDev and Sigasi plug-ins for Python and VHDL resp.)

RPeschke, 2018-11-22:
Sure i can set the axiStream object directly but that would leave me very open to human mistakes. 

The following table shows all the different states the AxiStream Object can be in. Only the ones marked in green are valid. All the others are either undefined or errors. This means i need to check on every assignment that the axiStream object is in a valid state. therefore it makes i huge difference for me to be able to use classes.  

[image]
U... Unset
S...  Set

In my example above if i would have a real AxiObject it would have a set data function.

the set data function would check if the AxiObject is in a state to except data. If it is not it would communicated the error back to the users. If it is in a state to except data, it would also set the TxValid to high. This way i can prevent the user from ever operating in an invalid state.  

we are currently in the evaluation of Sigasi. which version would you recommend to get? 

Thanks,
Richard

josyb, 2018-11-23:
I think your goal is rather simulation and not so much synthesis?
IMO you are over-engineering/over-constraining the AXI4S protocol.
Sigasi: it depends, if you restrict to VHDL the Sigasi Studio Creator (or even perhaps the free) version will suffice.

RPeschke, 2018-11-24:
maybe i am over engineering this one but as i said i am taking the axi stream example because it is the easiest i can think of. 

My point is it is an extreme simple protocol and yet it can go wrong in many different ways. On the other hand all of these problems have been solved by OOP a couple of decades ago. For me the only question is how to utilize this knowledge/techniques for this language. I thought MyHDL would be a good starting point since it has already a huge part of the functionality. for my point of few having the possibility to use function and procedures with objects (as it is already possible with processes) would be a central point to conveniently write good OO code.  

is there any argument i can make to convince you that adding this functionality is actually worth while?