myhdl/myhdl #278: Simulation & Conversion problems

kjleeic, 2018-10-14:
I wrote down two files: Convolution_2x2.py (module), tb_Convolution_2x2.py (testbench)

I want to do below four things in one code **when I execute my testbench code (e.g. Command: python tb_Convolution_2x2.py)**.

1. Make waveform files (.vcd) from myhdl simulation in './result' directory
2. Convert my module(Convolution_2x2) to verilog code in './Verilog' directory
3. Convert my testbench code to verilog in './Verilog/Testbench' directory
4. Don't copy whole module code in verilog conversion code of testbench.

But the function 'toVerilog', '.convert()' do not work well.
I use **python 3.6**.
Could you give a simple example with my below code?

My module code (file: Convolution_2x2.py) is:

```
from myhdl import *

def Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN):
        
    @always_comb
    def logic():
        a = not(x0 ^ w0)
        b = not(x1 ^ w1)
        c = not(a ^ b)
        out_DATA.next = not(x0 ^ w0)
        out_EN.next =  not( (not(x0 ^ w0)) ^ (not(x1 ^ w1)) )
        
    return logic
```

My testbench code is:
```
from Convolution_2x2 import *
from myhdl import *

@block
def tb_Convolution_2x2():
    x0 = Signal( bool(0) )
    x1 = Signal( bool(0) )
    w0 = Signal( bool(0) )
    w1 = Signal( bool(0) )
    out_DATA = Signal( bool(0) )
    out_EN = Signal( bool(0) )
    
    conv_2x2 = Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN)
    
    test_input = [Signal(bool(val)) for val in [0, 1]]
    
    @instance
    def Stimulus():
        for a in range(len(test_input)):
            for b in range(len(test_input)):
                for c in range(len(test_input)):
                    for d in range(len(test_input)):
                        x0.next = test_input[a]
                        w0.next = test_input[b]
                        x1.next = test_input[c]
                        w1.next = test_input[d]
                        
                        yield delay(10)
            
        raise StopSimulation()    
    
    return instances()

        
def simulate(timesteps):
#     traceSignals.filename = ''
    traceSignals.directory = './result'
    traceSignals.timescale = '1ns'
    tb = traceSignals(tb_Convolution_2x2)
    sim = Simulation(tb)
    sim.run(timesteps)
    
simulate(1000)

```

josyb, 2018-10-15:
This is what I made from it in the weekend:
> 1. Make waveform files (.vcd) from myhdl simulation in './result' directory
> 2. Convert my module(Convolution_2x2) to verilog code in './Verilog' directory
> 3. Convert my testbench code to verilog in [...]
1,2 and 3: Done
4: Because MyHDL flattens the design all module code (to be simulated) is inside the generated test-bench.  Which for me is no issue, as I never generate test-benches anyway :)

You will notice that everything is in a single module with the simulation and conversion code below the ```if __name__ == '__main__':``` *divider* neatly separating the to-be-imported code from the test-code

```python
from myhdl import Signal, block, always_comb, instance, StopSimulation, instances, delay

@block
def Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN):

    @always_comb
    def logic():
        out_DATA.next = not(x0 ^ w0)
        out_EN.next = not((not(x0 ^ w0)) ^ (not(x1 ^ w1)))

    return logic

if __name__ == '__main__':

    @block
    def tb_Convolution_2x2():
        x0 = Signal(bool(0))
        x1 = Signal(bool(0))
        w0 = Signal(bool(0))
        w1 = Signal(bool(0))
        out_DATA = Signal(bool(0))
        out_EN = Signal(bool(0))

        conv_2x2 = Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN)

        test_input = [Signal(bool(val))for val in [0, 1]]

        @instance
        def Stimulus():
            for a in range(len(test_input)):
                for b in range(len(test_input)):
                    for c in range(len(test_input)):
                        for d in range(len(test_input)):
                            x0.next = test_input[a]
                            w0.next = test_input[b]
                            x1.next = test_input[c]
                            w1.next = test_input[d]

                            yield delay(10)

            yield delay(10)
            raise StopSimulation

        return instances()

    def convert():
        x0 = Signal(bool(0))
        x1 = Signal(bool(0))
        w0 = Signal(bool(0))
        w1 = Signal(bool(0))
        out_DATA = Signal(bool(0))
        out_EN = Signal(bool(0))

        conv_2x2 = Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN)
        conv_2x2.convert(hdl='Verilog', name="Convolution_2x2", path="Verilog", testbench=False, initial_values=True)
        conv_2x2.convert(hdl='VHDL', name="Convolution_2x2", path="VHDL", initial_values=True)

    # simulate
    tb = tb_Convolution_2x2()
    tb.config_sim(directory='result', trace=True)
    tb.run_sim(None)

    # convert the 'main' function
    convert()

    # convert the testbench
    tbconv2x2 = tb_Convolution_2x2()
    tbconv2x2.convert(hdl='Verilog', name="tb_Convolution_2x2", path="Verilog/testbench", initial_values=True)
    tbconv2x2.convert(hdl='VHDL', name="tb_Convolution_2x2", path="VHDL/testbench", initial_values=True)

```

josyb, 2018-10-15:
I added a *separated* test-bench too
```python
from myhdl import Signal, block, always_comb, instance, StopSimulation, instances, delay

@block
def Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN):

    id1 = Signal(bool(0))
    id2 = Signal(bool(0))

    @always_comb
    def interim():
        id1.next = x0 ^ w0
        id2.next = x1 ^ w1

    @always_comb
    def logic():
        out_DATA.next = not id1
        out_EN.next = not((not id1) ^ (not id2))

    return interim, logic

if __name__ == '__main__':

    @block
    def tb_Convolution_2x2():
        x0 = Signal(bool(0))
        x1 = Signal(bool(0))
        w0 = Signal(bool(0))
        w1 = Signal(bool(0))
        out_DATA = Signal(bool(0))
        out_EN = Signal(bool(0))

        conv_2x2 = Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN)

        test_input = [Signal(bool(val))for val in [0, 1]]

        @instance
        def Stimulus():
            for a in range(len(test_input)):
                for b in range(len(test_input)):
                    for c in range(len(test_input)):
                        for d in range(len(test_input)):
                            x0.next = test_input[a]
                            w0.next = test_input[b]
                            x1.next = test_input[c]
                            w1.next = test_input[d]

                            yield delay(10)

            yield delay(10)
            raise StopSimulation

        return instances()

    @block
    def tb_Convolution_2x2_sep(x0, x1, w0, w1, out_DATA, out_EN):
        test_input = [Signal(bool(val))for val in [0, 1]]

        @instance
        def Stimulus():
            for a in range(len(test_input)):
                for b in range(len(test_input)):
                    for c in range(len(test_input)):
                        for d in range(len(test_input)):
                            x0.next = test_input[a]
                            w0.next = test_input[b]
                            x1.next = test_input[c]
                            w1.next = test_input[d]

                            yield delay(10)
                            # perhaps read the results here
                            # and print something
                            print(out_DATA, out_EN)  # not sure about this

            yield delay(10)
            raise StopSimulation

        return instances()

    def convert():
        x0 = Signal(bool(0))
        x1 = Signal(bool(0))
        w0 = Signal(bool(0))
        w1 = Signal(bool(0))
        out_DATA = Signal(bool(0))
        out_EN = Signal(bool(0))

        conv_2x2 = Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN)
        conv_2x2.convert(hdl='Verilog', name="Convolution_2x2", path="Verilog", testbench=False, initial_values=True)
        conv_2x2.convert(hdl='VHDL', name="Convolution_2x2", path="VHDL", initial_values=True)

    def convert_tb():
        x0 = Signal(bool(0))
        x1 = Signal(bool(0))
        w0 = Signal(bool(0))
        w1 = Signal(bool(0))
        out_DATA = Signal(bool(0))
        out_EN = Signal(bool(0))

        tbconv2x2 = tb_Convolution_2x2_sep(x0, x1, w0, w1, out_DATA, out_EN)
        tbconv2x2.convert(hdl='Verilog', name="tb_Convolution_2x2_sep", path="Verilog/testbench", initial_values=True)
        tbconv2x2.convert(hdl='VHDL', name="tb_Convolution_2x2_sep", path="VHDL/testbench", initial_values=True)

    # simulate
    tb = tb_Convolution_2x2()
    tb.config_sim(directory='result', trace=True)
    tb.run_sim(None)

    # convert the 'main' function
    convert()

    # convert the testbench
    tbconv2x2 = tb_Convolution_2x2()
    tbconv2x2.convert(hdl='Verilog', name="tb_Convolution_2x2", path="Verilog/testbench", initial_values=True)
    tbconv2x2.convert(hdl='VHDL', name="tb_Convolution_2x2", path="VHDL/testbench", initial_values=True)

    def convert_tb():
        x0 = Signal(bool(0))
        x1 = Signal(bool(0))
        w0 = Signal(bool(0))
        w1 = Signal(bool(0))
        out_DATA = Signal(bool(0))
        out_EN = Signal(bool(0))

        tbconv2x2 = tb_Convolution_2x2_sep(x0, x1, w0, w1, out_DATA, out_EN)
        tbconv2x2.convert(hdl='Verilog', name="tb_Convolution_2x2_sep", path="Verilog/testbench", initial_values=True)
        tbconv2x2.convert(hdl='VHDL', name="tb_Convolution_2x2_sep", path="VHDL/testbench", initial_values=True)

    # convert the alternative testbench
    convert_tb()
```