myhdl/myhdl #279: Verilog parameter support

kjleeic, 2018-10-14:
Is there any similar syntax in myhdl like **parameter** in verilog?

Examples:
```
module dff **#( parameter N = 8)**
(
    input trigger,
    input enable,
    input      [N-1:0] d,
    output reg [N-1:0] q
);
    always @(posedge trigger) begin
        if(enable) begin
            q <= d;
        end
    end
endmodule
```

cfelton, 2018-10-14:
@kjleeic you can pass *ports* and *parameters* as arguments in an myhdl block, however these parameters will not be converted in the generated code.

```python
@hdl.block
def myblock(port1, port2, port3, param1=1, param2=False):
    # block code ...
```

kjleeic, 2018-10-15:
> @kjleeic you can pass _ports_ and _parameters_ as arguments in an myhdl block, however these parameters will not be converted in the generated code.
> 
> ```python
> @hdl.block
> def myblock(port1, port2, port3, param1=1, param2=False):
>     # block code ...
> ```
> 
> This example might help:
> http://docs.myhdl.org/en/stable/manual/conversion_examples.html#a-small-combinatorial-design

Thanks!

nanduraj1, 2020-11-07:
@cfelton this will replace the parameter with its values, in the generated verilog, isn't it?

NicoPy, 2020-11-07:
Yes, that the way it works.