myhdl/myhdl #280: Tracing internal signals of mudule w/o input or output port

kjleeic, 2018-10-15:
Is there any way get internal signals of module (e.g. reg in module) in .vcd file without making input or output port when I use **traceSignals()**?

In SystemC, for example, LogicPoet(http://www.logicpoet.com/systemc/) project is supporting this.

Vivado also support  this thing.

josyb, 2018-10-15:
Internal *Signals* are included in the .vcd output.
I slightly modified the Convolution example to show this:
```python
@block
def Convolution_2x2(x0, x1, w0, w1, out_DATA, out_EN):

    id1 = Signal(bool(0))
    id2 = Signal(bool(0))

    @always_comb
    def interim():
        id1.next = x0 ^ w0
        id2.next = x1 ^ w1

    @always_comb
    def logic():
        out_DATA.next = not id1
        out_EN.next = not((not id1) ^ (not id2))

    return interim, logic
```
and this the resulting waveform showing the two internal signals *id1* and *id2*:
[image]

kjleeic, 2018-10-15:
> waveform

Thank you for your kind answer.
What is the program that you use to see a waveform?
It looks different from GTKwave.

josyb, 2018-10-15:
Impulse by toem.de as a plug-in in Eclipse IDE (which also runs PyDev for Python development and Sigasi for VHDL development and a few other editors)