myhdl/myhdl #281: array of module or generate syntax in myhdl

kjleeic, 2018-10-30:
Is there any way to express array of module (or generate syntax in verilog) in myhdl?

Array of module:

```
wire DFF_i[15:0];
wire DFF_o[15:0];

DFF d[15:0] (clk, DFF_i, DFF_o);
```

Generate Syntax:

```
wire DFF_i[15:0];
wire DFF_o[15:0];

generate
  genvar i;
  for (i=0; i<15; i=i+1) begin : dff
    custom i_custom(
       .clk(clk)
      ,.input(DFF_i[i])
      ,.output(DFF_o[i])
      );
  end
endgenerate
```

josyb, 2018-10-31:
This shows you how:
```python
from myhdl import block, Signal, intbv, always_comb, always_seq

@block
def DFF(Clk, D, Q):

    @always_seq(Clk.posedge, reset=None)
    def seq():
        Q.next = D

    return seq

@block
def DFF_Register(Clk, D, Q):
    NBR_BITS = len(D)
    # we need an array of bits to receive the output of each instantiated DFF
    dffq = [Signal(bool(0)) for _ in range(NBR_BITS)]

    # make a placeholder for the instantiated DFFs
    dffs = [None for _ in range(NBR_BITS)]

    # instantiate a singe DFF for each bit in the input vector
    for i in range(NBR_BITS):
        dffs[i] = DFF(Clk, D(i), dffq[i])

    # assign the output bit of the instantiated DFFs to the output vector
    @always_comb
    def assign():
        for i in range(NBR_BITS):
            Q.next[i] = dffq[i]

    return dffs, assign

Clk = Signal(bool(0))
D = Signal(intbv(0)[16:])
Q = Signal(intbv(0)[16:])

duc = DFF_Register(Clk, D, Q)
duc.convert(hdl='verilog', path='Verilog')

```

bat52, 2023-02-26:
Hello, I ran this example and getting the following error.
I bumped into this discussion because I originally had the same on a design of mine, that included the same genvar structure.
Can you give some advice ?

Traceback (most recent call last):
  File "/media/marco/DATA/programming/rtl/ddpm/myhdl/./genvar.py", line 40, in <module>
    duc.convert(hdl='verilog') # , path='Verilog')
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/_block.py", line 342, in convert
    return converter(self)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_toVerilog.py", line 177, in __call__
    genlist = _analyzeGens(arglist, h.absnames)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 170, in _analyzeGens
    v.visit(tree)
  File "/usr/lib/python3.10/ast.py", line 410, in visit
    return visitor(node)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 1119, in visit_Module
    _AnalyzeBlockVisitor.visit_Module(self, node)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 1070, in visit_Module
    self.generic_visit(node)
  File "/usr/lib/python3.10/ast.py", line 418, in generic_visit
    self.visit(item)
  File "/usr/lib/python3.10/ast.py", line 410, in visit
    return visitor(node)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 1099, in visit_FunctionDef
    self.visit(n)
  File "/usr/lib/python3.10/ast.py", line 410, in visit
    return visitor(node)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 721, in visit_For
    self.visit(stmt)
  File "/usr/lib/python3.10/ast.py", line 410, in visit
    return visitor(node)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 521, in visit_Assign
    self.visit(target)
  File "/usr/lib/python3.10/ast.py", line 410, in visit
    return visitor(node)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 962, in visit_Subscript
    self.accessIndex(node)
  File "/home/marco/.local/lib/python3.10/site-packages/myhdl/conversion/_analyze.py", line 986, in accessIndex
    self.visit(node.slice.value)
AttributeError: 'Name' object has no attribute 'value'

josyb, 2023-02-26:
Hi Marco,

I re-tested this against the current master branch and it converts fine.
I surmised (ascertained?) that you are running an 'older' version of MyHDL 0.11
Which you probably have installed from PyPi.
Apologies, but the PyPi is not kept up to date by the previous maintainers and I don’t have the ‘key’ - we are working on this
You can install the master branch directly from the GutHub repository:
`pip install git+https://github.com/myhdl/myhdl.git@master`
The master branch is tested against CPython 3.7, 3.8, 3.9, 3.10, 3.11 and PyPy 3.9

Best regards,
Josy

bat52, 2023-02-26:
Thank you very much @josyb for the quick reply.
I confirm that indeed the example worked after installing myhdl from github. 
Perhaps you could add a disclaimer on the github README ;-).

Anyways... thank you very much for the good work! I came back to myHDL, after giving a quick try to amaranth HDL that seems to get some traction recently, but in my opinion results in a much less readable RTL code than myHDL.