myhdl/myhdl #283: Help with design low-level HDL language

XVilka, 2018-11-16:
FPGA  world suffers a lot from fragmentation - some tools produce Verilog,  some VHDL, some - only subsets of them, creating low-level LLVM-like  alternative will help everyone, so HDL implementations will opt only for  generating this low-level HDL and routing/synthesizers accept it. LLVM  or WebAssembly - you can see how many languages and targets are  supported now by both. With more open source tools for FPGA this is more  feasible now than ever.

See SymbiFlow/ideas#19

cfelton, 2018-11-27:
A quick glance at the [SymbiFlow/ideas#19](https://github.com/SymbiFlow/ideas/issues/19), looks like it is going to be difficult, if not impossible, to get a consensus on anything more than trivial items.  

We have limited resources but the topic of a usable IR rears its ugly head often.  It sounds like a reasonable idea but the practicality seems uncertain.

XVilka, 2018-11-28:
Well, everyone more or less agree that FIRRTL is a good candidate with a condition that some features will be added in it and it will be implemented as C/C++ library, so every project can reuse it no matter what language it was written in.

josyb, 2018-11-28:
No, not everyone ... I'd say "au contraire"

**FIRRTL eventually translates to Verilog**, so what's the point?

XVilka, 2018-11-28:
@josyb as for current implementation it does. Of course at this point it is not suitable as is. The idea is to take the language design and concepts and reimplement in C++, eliminating the need of Verilog middle man. Integrating it across the tools will help to phase Verilog/VHDL out. They themselves can be compiled into FIRRTL (or whatever name this IR will have) at that time.

josyb, 2018-11-28:
I won't hold my breath ...
Which vendor tools (will) accept FIRRTL?