myhdl/myhdl #288: Verilog Conversion - Unable to use 'initial_values' with TristateSignals

pete-restall, 2018-12-12:
Converting a block to Verilog that takes a TristateSignal does not work if you also set 'initial_values=True'.  The resulting error:

    n = None, radix = ''
        def _intRepr(n, radix=''):
            # write size for large integers (beyond 32 bits signed)
            # with some safety margin
            # XXX signed indication 's' ???
    >       p = abs(n)
    E       TypeError: bad operand type for abs(): 'NoneType'

    /usr/local/lib/python3.6/site-packages/myhdl/conversion/_toVerilog.py:483: TypeError

I have attached a ZIP of a simple DUT and test fixture that illustrate the issue.  I'm using Icarus Verilog as the simulator, so place myhdl.vpi in the same directory and then run:

    $ pytest tristate_output_driver_test.py

If you set 'initial_values=False' in the test (marked with a comment) then the simulation progresses further, but there is another issue with TristateSignals - I can raise another issue for this if necessary.

Any help on either issue is appreciated - I've been struggling with Tristates for a while now, and they should just be a daft bit of periphery glue for my I2C bus.

Cheers.
[tristate_bugs.zip](https://github.com/myhdl/myhdl/files/2673412/tristate_bugs.zip)

pete-restall, 2018-12-14:
I should probably say that I hacked '_intRepr()' in _toVerilog.py to work around this, but it most likely has knock-on effects that I'm not yet aware of:

    def _intRepr(n, radix=''):
        if n is None:
            return "'bz"
        ...

hashhsah, 2019-02-07:
@pete-restall I wonder if the discussions in #190 and #21 are relevant to the problem you encounter?

pete-restall, 2019-02-09:
> @pete-restall I wonder if the discussions in #190 and #21 are relevant to the problem you encounter?

Doesn't look like it.  For #21 that was to do with interfaces by the looks; my scenario didn't involve interfaces.

For #190 the issue was trying to use a non-None initial value for a tristate.  My issue was that None is the (default) initial value of my TristateSignal but it doesn't convert because the converter is trying to get the integer value of the signal.  It should convert the None to 'bz, which is what my _intRepr() hack does.

I've found a few other issues with TristateSignals as well, such as passing them into a Cosimulation doesn't work, but I've managed to work around the other issues without modifying the MyHDL code.