myhdl/myhdl #291: Incorrect Simulation

LeHoangLong, 2018-12-17:
I created a help block for edge detection:

@block
def RisingEdgeDetector(iClk, iData, oTrigger):
    data_prev = Signal(bool(iData))
    
    @always(iClk.posedge)
    def edgeDetector():
        data_prev.next = iData
        
    @always_comb
    def output():
        if data_prev == 0 and iData == 1:
            oTrigger.next = 1
            pass
        else:
            oTrigger.next = 0
            pass
    return edgeDetector, output

Then in my design, i instantiated 4 of them. 3 of them was working fine. But for some reason, the last one was not working as expected. data_prev changes at the same edge as iData instead of 1 cycle delay. I have checked the converted vhdl output file and everything seems correct. Only the simulation is wrong
[image]

josyb, 2018-12-17:
At first sight everything looks fine. Can you give us the complete file/project? Then I will check the simulation also.

Regards,
Josy

LeHoangLong, 2018-12-17:
Hi Josy,

Attached are the files.  The first 4 signals come from 1 of the 3 other block. The last 4 signals is from the incorrect instance. As you can see, the data_prev changes instantly with iData.

[temp.tar.gz](https://github.com/myhdl/myhdl/files/2686921/temp.tar.gz)

Regards,
Long

LeHoangLong, 2018-12-17:
Update: 

I was able to get around it by inserting a temporary wire:

@block
def RisingEdgeDetector(iClk, iData, oTrigger):
    data_prev = Signal(bool(iData))
    temp = Signal(bool(iData))
    
    @always(iClk.posedge)
    def edgeDetector():
        data_prev.next = temp
        
    @always_comb
    def output():
        temp.next = iData
        if data_prev == 0 and iData == 1:
            oTrigger.next = 1
            pass
        else:
            oTrigger.next = 0
            pass
    return edgeDetector, output

josyb, 2018-12-17:
In your testbench you generate the input signal for the 4th edge detector  via a delay(100). This is exactly 10 clock cycles, and you now have a race condition, which in my simulation showed up as a 0-width glitch.
I replaced the 100 delay by a for-loop, and everything works as expected.
```
                    uart_sendDone.next = 0
#                     yield delay(100)
                    for _ in range(10):
                        yield clk.posedge
                    uart_sendDone.next = 1
```
The temporary wire you inserted provokes a *delta cycle* which made it work.

LeHoangLong, 2018-12-18:
Okay, thank you for the information.

closed 2018-12-18 by LeHoangLong