myhdl/myhdl #306: Verilog conversion : Sometimes, wires should be regs

NicoPy, 2019-03-12:
When converting a design to Verilog, a non driven signal is declared as ```wire```.
Sometimes, it should be declared as ```reg```. Typically, when instantiating a ROM.

## System information
```
MyHDL Version: 0.10
```

cfelton, 2019-03-12:
@NicoPy do you have an example we can use as a test case in the test suite?

josyb, 2019-03-12:
Can you add a relevant example?

josyb, 2019-03-12:
Perhaps check this first: http://docs.myhdl.org/en/stable/manual/conversion_examples.html
At the end it shows the ROM inference.

NicoPy, 2019-03-12:
Yes, but do vendor tools consider a big case as a ROM ?

josyb, 2019-03-12:
Check it with the vendor tools :)
If you add the example someone will?

NicoPy, 2019-03-12:
Here they are :
[Test.zip](https://github.com/myhdl/myhdl/files/2957170/Test.zip)
rom.py is the problematic example where 'rom_array' is converted to 'wire' instead of 'reg'.
rom_tuple.py is the source Josy will use to check vendor tool behaviour ;)

josyb, 2019-12-24:
Hi Nicolas,

It's been a while ...
I checked your code (again): I think the vendor tools will not instantiate a RAM-block-based ROM because you specified a reset signal.