myhdl/myhdl #308: Support older Revisions of VHDL explicitly

jlusiardi, 2019-03-21:
Hi,

I have hit the following issue with Signals with a set value (like `lsr = Signal(intbv(0x2eb1edd0, _nrbits=30))`. If, on conversion to VHDL, I use the parameter `initial_values=True` it generates this:
```vhdl
signal pseudo_random0_lsr: unsigned(31 downto 0) := 30X"aeb1edd0";
```
This is valid VHDL 2008 code but invalid in VHDL 1993 and VHDL 1987 (at least if I trust Quartus II here). 

The following fragment seems valid in VHDL 1993 and VHDL 1987:
```vhdl
signal pseudo_random0_lsr: unsigned(31 downto 0) := 30X"2eb1edd0";
```
**Proposal**
I don't know of anymore of these differences between the various revisions of VHDL. There might be more things, that can be expressed in a better way in newer version of VHDL. So my proposal would be to add a new option to specify the revisions of VHDL we want to have. Default should be 2018 though.

What do you think?

joachim

NicoPy, 2019-07-09:
It has been decided, long time ago, that VHDL2008 is required.
Many vendor tools accept VHDL2008 source files today.
Is VHDL2008 a problem for you ?

jlusiardi, 2019-07-09:
No, VHDL2008 is not an issue. I did not find any explicit statement in the documentation with a quick view. Perhaps this should be more emphasized? 
From my point of view, we can close here then.

NicoPy, 2019-07-09:
You are right, this should be mentioned in the documentation.