myhdl/myhdl #311: How to define Array Port in myhdl

yangjin07, 2019-04-20:
I'd like to use an 2D arrays  port:

output [7:0] data_out[0:3];

Is there any example for generating such port definition in myhdl?
-----------------------------------------------------------------------------
Alternatively, I try to use another expression. 

reg [8*4-1:0] data_out;
reg [7:0] data_buf[0:4];

always @* begin
    for (i=0;i<4;i=i+1) begin
        data_out[i*8+8:i*8] = data_buf[i];
    end
end

But I've got Error message compiling the code using xvlog (Xilinx Vivado).
Error: data_out[i*8+8:i*8] range must be bounded by constant expressions

NicoPy, 2019-05-17:
Lists (arrays) are not supported at the top level (design ports) for now.
You have to use "simple" signals at the top level. You can use a top level wrapper converting "simple" signals to lists of signals.
2D arrays are not supported anywhere in the design (for now).

NicoPy, 2019-06-21:
I have a fix for arrays in ports. I just need time to create a pull-request.

olisnr, 2023-04-06:
im interested: is it possible today to make array of buses in the toplevel?

josyb, 2023-04-06:
No, currently a List Of Interfaces (a bus is an Interface) is not supported at the toplevel.
Can you elaborate a bit more?

olisnr, 2023-04-11:
translation from a list of signals to VHDL it should be possible, but not in verilog. so may be its not a good idea to implement it. 

but it would make multichannel code more readable. and [NicoPy](https://github.com/NicoPy) was writing he has done it allready (if i understand his statement correct). i would like to use it and change to VHDL...

josyb, 2023-04-11:
@NicoPy  used a VHDL package - which obviously doesn't work for Verilog; there may b alike possibilities in SystemVerilog but I still have to learn a lot there
The obvious alternative is to enumerate the array, that will work for both VHDL and Verilog - I have  experimental code for that
Most users declare a large std_logic_vector and then make an array out of that in their MyHDL code.

olisnr, 2023-04-11:
i would be interested in using this package, where can i find it?