myhdl/myhdl #314: AttributeError: 'Name' has no attribute 'obj'

sww1235, 2019-05-11:
I am testing out MyHDL using the implementation of a simple 8 tap FIR filter. I am uncertain if this is a bug or a mistake on my part, but I followed the readthedocs conversion example as closely as I could. 

This only happens when specifying VHDL in the convert function. When using Verilog it converts perfectly.

The code is in two files and is shown below after the error log:

Thank you

## System information
```
MyHDL Version: f8efa3d041194fade09109a9748be3476ec0096f installed using pip3 install --upgrade git+git://github.com/myhdl/myhdl@master
Python Version: 3.7.3 (mac homebrew) and 3.7.2 (windows 10)
```

```
Ω:python3 firFilterConverter.py 
Traceback (most recent call last):
  File "firFilterConverter.py", line 24, in <module>
    convert_FIRFilter(hdl='VHDL')
  File "firFilterConverter.py", line 21, in convert_FIRFilter
    firFilter_1.convert(hdl=hdl)
  File "/usr/local/lib/python3.7/site-packages/myhdl/_block.py", line 342, in convert
    return converter(self)
  File "/usr/local/lib/python3.7/site-packages/myhdl/conversion/_toVHDL.py", line 207, in __call__
    _annotateTypes(genlist)
  File "/usr/local/lib/python3.7/site-packages/myhdl/conversion/_toVHDL.py", line 2437, in _annotateTypes
    v.visit(tree)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ast.py", line 262, in visit
    return visitor(node)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ast.py", line 270, in generic_visit
    self.visit(item)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ast.py", line 262, in visit
    return visitor(node)
  File "/usr/local/lib/python3.7/site-packages/myhdl/conversion/_toVHDL.py", line 2177, in visit_FunctionDef
    self.visit(stmt)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ast.py", line 262, in visit
    return visitor(node)
  File "/usr/local/lib/python3.7/site-packages/myhdl/conversion/_toVHDL.py", line 2265, in visit_For
    self.generic_visit(node)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ast.py", line 270, in generic_visit
    self.visit(item)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ast.py", line 262, in visit
    return visitor(node)
  File "/usr/local/lib/python3.7/site-packages/myhdl/conversion/_toVHDL.py", line 2189, in visit_AugAssign
    self.visit(node.target)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ast.py", line 262, in visit
    return visitor(node)
  File "/usr/local/lib/python3.7/site-packages/myhdl/conversion/_toVHDL.py", line 2274, in visit_Name
    node.vhd = inferVhdlObj(node.obj)
AttributeError: 'Name' object has no attribute 'obj'
```

```python
"""Python Implementation of a 8 tap FIR filter.
Designed to be implemented in hardware through postprocessing in MyHDLself.
"""

from myhdl import block, always_seq, Signal, intbv, downrange

@block
def FIRFilter(clock, reset, input, output):
    """FIR Filter.
    clock -- master clock input
    reset -- reset input, active low
    input -- signed 8 bit value
    output -- signed 8 bit value
    Due to constraints in myHDL, you need a top level block without
    function arguments in order to convert properly
    """
    # python list of signed 8 bit integers
    delayLine = [Signal(intbv(0, min=-128, max=127)) for i in range(8)]

    # @always_seq generates reset logic automagically
    @always_seq(clock.posedge, reset=reset)
    def FIRProcess():
        # process output
        for tap in range(8):
            output += delayLine[tap]//8
        # shift all taps up one
        for tap in downrange(8, 1):  # 7 downto 1
            delayLine[tap] = delayLine[tap - 1]
        delayLine[0] = input

    return FIRProcess
```

```python
""Converter for Python Implementation of a 8 tap FIR filter.
Designed to be implemented in hardware through postprocessing in MyHDLself.
"""

from myhdl import Signal, ResetSignal, modbv, intbv

from FIRFilter import FIRFilter

def convert_FIRFilter(hdl):
    """Convert FIR Filter to verilog or VHDL."""
    clock = Signal(bool(0))
    reset = ResetSignal(0, active=0, isasync=True)
    input = Signal(intbv(0, min=-128, max=127))
    output = Signal(intbv(0, min=-128, max=127))

    firFilter_1 = FIRFilter(clock, reset, input, output)
    firFilter_1.convert(hdl=hdl)

convert_FIRFilter(hdl='VHDL')
```

josyb, 2019-05-11:
This is definitely *not* a bug. I don't do Verilog, but if your code converts to Verilog, it will probably not synthesize, perhaps not even simulate ...

This may come closer to what you intended:
```python
@block
def FIRFilter(clock, reset, input, output):
    """FIR Filter.
    clock -- master clock input
    reset -- reset input, active low
    input -- signed 8 bit value
    output -- signed 8 bit value
    Due to constraints in myHDL, you need a top level block without
    function arguments in order to convert properly
    """
    # python list of signed 8 bit integers
    delayLine = [Signal(intbv(0, min=-128, max=128)) for _ in range(8)]
    sumfir = Signal(intbv(0, min = -128 * 8, max = 128 * 8))
    @always_comb
    def tally():
        sumfir.next = delayLine[0] + delayLine[1] + delayLine[2] + delayLine[3] + \
                      delayLine[4] + delayLine[5] + delayLine[6] + delayLine[7]
    
    # @always_seq generates reset logic automagically
    @always_seq(clock.posedge, reset=reset)
    def FIRProcess():
        # process output
        output.next = sumfir // 8
        # shift all taps up one
        for tap in downrange(8, 1):  # 7 downto 1
            delayLine[tap].next = delayLine[tap - 1]
        delayLine[0].next = input

    return tally, FIRProcess
```
This converts
The main thing is the **.next**  which you need to assign new values.
BTW: *input* and *output* are *reserved* words in Verilog

sww1235, 2019-05-13:
Thank you for the hint around the .next operator. I didn't notice that when reading the examples the first time.

I had already fixed the input and output reserved word issues but forgot to paste them into the issue.

Is there any reason why I cannot do += for the sum?

josyb, 2019-05-13:
There can be several reasons:
1. It was deferred until someday ...
2. The construct is *not pure*; i.e. ``` sum.next += val[i]``` can be rewritten as ``` sum.next = sum.next + val[i]```  which is not an exact replacement of ``` sum.next = sum + val[i]```.  Not sure whether the simulator would handle this Python construct correctly.
3. There is no exact match in Verilog (nor in VHDL)

josyb, 2019-05-13:
You can change above code in such a way that you don't need the for-loop summing at all, at the same time reducing the resources to one addition and one subtraction in stead of seven additions.

sww1235, 2019-05-14:
thanks for the help. i will go ahead and close this now

closed 2019-05-14 by sww1235