myhdl/myhdl #321: Icarus cosimulation on Windows 10 broken

philipaxer, 2019-09-30:
Hi All, I was recently playing with cosimulation on windows 10 and noted that the pipe communication between myhdl.vpi and myhdl doesnt work. Actually this was somewhat noticed by a broken testcase in 2016 (this is not 100% the same, as this doesnt directly involve myhdl.vpi): http://discourse.myhdl.org/t/myhdl-test-suite-under-windows-10-cosimulation-fails/49/4

The issue is that windows differentiates between OS-wide file handles and per-process file descriptors. The Cosimulation class passes the pipe correctly as a file handle, but the myhdl.vpi side doesnt convert this to a file descriptor (yet).

long story short: changing the init_pipe function in myhdl.c fixes this for me:
```
wpipe = _open_osfhandle(atoi(w), 0);
rpipe = _open_osfhandle(atoi(r), 0);
```

## System information
```
MyHDL Version: 0.11
Python Version: 3.7.2
```

philipaxer, 2019-09-30:
The following snippet should fix the issue:
```
  #ifdef _WIN32
    wpipe = _open_osfhandle(atoi(w), 0);
    rpipe = _open_osfhandle(atoi(r), 0);
  #else
    wpipe = atoi(w);
    rpipe = atoi(r);
  #endif
```

daniel3097, 2019-10-09:
Hi all,

I faced the same issue with exact the same version - but solved it as following:

1. diff between myhdl.c modelsim and modelsim-win gave a hint what needs to be changed for windows plattform (mostly the pipe handling)
2. then I manually applied the diff (search and replace) to the myhdl.c for Icarus

To my big surprise this was working for a simple counter.v example (I did not yet had the time to run all the tests, fix the makefile or update the README file).

Attached you find the modified file. 
[myhdl.c.txt](https://github.com/myhdl/myhdl/files/3709313/myhdl.c.txt)

I am neither a Windows nor a Linux nor a C programmer the use is at your own risk. 
Also sorry for not providing proper patches - but as a google search showed I am not the first one that had this issue.

cfelton, 2019-10-13:
@philipaxer thanks for bringing this to our attention.

cfelton, 2019-10-13:
@philipaxer what posix emulation system were you using on Windows, e.g. cygwin?

josyb, 2019-10-14:
I thought the idea was that cosimulation *just* works in native Windows; i.e. without all that cygwin and other ...

philipaxer, 2019-10-14:
I am using msys2 and the mingw that comes with it. No cygwin involved.
Additionally, I can also confirm issue #273, but the underlying proble is probably in the icarus sources.
regards
Philip

josyb, 2020-04-19:
@philipaxer 
> The following snippet should fix the issue:
> 
> ```
>   #ifdef _WIN32
>     wpipe = _open_osfhandle(atoi(w), 0);
>     rpipe = _open_osfhandle(atoi(r), 0);
>   #else
>     wpipe = atoi(w);
>     rpipe = atoi(r);
>   #endif
> ```

Will you make a PR  for this?