myhdl/myhdl #324: myhdl + icarus: static signals not correct in Python

bluecmd, 2019-11-15:
Hi,

I'm using MyHDL with Icarus and I have a statement that is:

```
assign bar0_mm_writedata = 32'hdeadbeef; 
```

When reading this from my Cosimulation I read "0" on that signal. If I tell Icarus to graph the signal it correctly displays 0xdeadbeef.

Same issue is with:

```
  logic [31:0] wd = 32'hdeadbeef;
  assign bar0_mm_writedata = wd;
  always @(posedge clk) begin
    wd <= 32'hdeadbeef;
  end
```

But if I change the initial value to change the signal value is correctly propagated to MyHDL:
```
  logic [31:0] wd = 32'h0;
  assign bar0_mm_writedata = wd;
  always @(posedge clk) begin
    wd <= 32'hdeadbeef;
  end
```

## System information
```
MyHDL Version: 0.11
Python Version: 3.7.3
```

chiplukes, 2019-11-15:
As far as I'm concerned this is the single biggest issue with co-simulation that I run into (usually with AXI stream interfaces where RDY is assigned to 1 in Verilog).  My usual fix is to make sure that the initial value of the signal in the MyHDL testbench matches the unchanging verilog wire/reg value.

May be related to [#238](https://github.com/myhdl/myhdl/issues/238)

bluecmd, 2019-11-15:
+1. My case was an Avalon interface setting ready to static one :-).

closed 2024-10-05 by josyb