myhdl/myhdl #326: Convert to verilog and keep parameter name

kermit0124, 2019-11-30:
Hello, I'v try the little sample of the convert function.
I used the example, the "width" be change by value "8" and replace it for all verilog "width"...
So, I want to generate the verilog RTL code that keep the "width" as a verilog parameter in the module. How can I do?
Thanks!

--------
    def convert(hdl, width=8):
       B = Signal(intbv(0)[width:])
       G = Signal(intbv(0)[width:])
       inst = bin2gray(B, G)
    inst.convert(hdl=hdl)
    convert(hdl='Verilog')

------------------------------

josyb, 2019-12-24:
Currently, you can't.
One of the ideas behind MyHDL is that the generated VHDL and Verilog code is to be read by synthesis tools, rather than by humans.

kermit0124, 2019-12-28:
I see. Thanks a lot of!

closed 2019-12-28 by kermit0124

nanduraj1, 2020-04-07:
@josyb , If parameters we preserved, we could have slowly moved to myhdl, by writing one module a a time with myhdl and then instantiating it on our existing SV codebase

josyb, 2020-04-07:
I did it the other way round; I moved from writing VHDL 'library' blocks to writing these in MyHDL and connecting large(r) projects in MyHDL. E,g. one of my projects generates 33000 lines of VHDL code. 
I thought about having parameters e.g. for hierarchical conversion. But until now I do not miss this functionality.

Pusiol, 2021-07-21:
How can you have two modules in a project with the same architecture but different properties without using parameters? (Eg: two fixed decimators with different rates)

josyb, 2021-07-21:
@Pusiol 

MyHDL expands all instantiations and then flatten everything; so you will end up with two written out decimators but with different parameters and signal widths etc.
An example:
```python
'''
Created on 21 jul. 2021

@author: josy
'''

from myhdl import (block, Signal, intbv, modbv, always_comb, always_seq,
                   instances, ResetSignal)

@block
def counter(Clk, Reset, Q):
    count = Signal(modbv(0)[len(Q):])

    @always_seq(Clk.posedge, Reset)
    def docount():
        count.next = count + 1

    @always_comb
    def mkoutput():
        Q.next = count

    return instances()

@block
def contrived_random_generator(Clk1, Reset1, Clk2, Reset2, Clk3, RandomBit):
    Q1 = Signal(intbv(0)[7:])
    counter1 = counter(Clk1, Reset1, Q1)
    Q2 = Signal(intbv(0)[5:])
    counter2 = counter(Clk2, Reset2, Q2)
    metachain = Signal(intbv(0)[2:])

    @always_seq(Clk3.posedge, reset=None)
    def crg():
        metachain.next[0] = Q1[0] ^ Q2[0]
        metachain.next[1] = metachain[0]
        RandomBit.next = metachain[1]

    return instances()

if __name__ == '__main__':
    Clk1 = Signal(bool(0))
    Reset1 = ResetSignal(0, 1, False)
    Clk2 = Signal(bool(0))
    Reset2 = ResetSignal(0, 1, False)
    Clk3 = Signal(bool(0))
    RandomBit = Signal(bool(0))

    dfc = contrived_random_generator(Clk1, Reset1, Clk2, Reset2, Clk3, RandomBit)
    dfc.convert(hdl='VHDL')
```
will generate:
```vhdl
-- File: contrived_random_generator.vhd
-- Generated by MyHDL 0.11
-- Date: Wed Jul 21 19:22:37 2021

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_011.all;

entity contrived_random_generator is
	port(
		Clk1      : in  std_logic;
		Reset1    : in  std_logic;
		Clk2      : in  std_logic;
		Reset2    : in  std_logic;
		Clk3      : in  std_logic;
		RandomBit : out std_logic
	);
end entity contrived_random_generator;

architecture MyHDL of contrived_random_generator is

	signal Q1             : unsigned(6 downto 0);
	signal metachain      : unsigned(1 downto 0);
	signal Q2             : unsigned(4 downto 0);
	signal counter0_count : unsigned(6 downto 0);
	signal counter1_count : unsigned(4 downto 0);

begin

	CONTRIVED_RANDOM_GENERATOR_COUNTER0_DOCOUNT : process(Clk1) is
	begin
		if rising_edge(Clk1) then
			if (Reset1 = '1') then
				counter0_count <= to_unsigned(0, 7);
			else
				counter0_count <= (counter0_count + 1);
			end if;
		end if;
	end process CONTRIVED_RANDOM_GENERATOR_COUNTER0_DOCOUNT;

	Q1 <= counter0_count;

	CONTRIVED_RANDOM_GENERATOR_COUNTER1_DOCOUNT : process(Clk2) is
	begin
		if rising_edge(Clk2) then
			if (Reset2 = '1') then
				counter1_count <= to_unsigned(0, 5);
			else
				counter1_count <= (counter1_count + 1);
			end if;
		end if;
	end process CONTRIVED_RANDOM_GENERATOR_COUNTER1_DOCOUNT;

	Q2 <= counter1_count;

	CONTRIVED_RANDOM_GENERATOR_CRG : process(Clk3) is
	begin
		if rising_edge(Clk3) then
			metachain(0) <= (Q1(0) xor Q2(0));
			metachain(1) <= metachain(0);
			RandomBit    <= metachain(1);
		end if;
	end process CONTRIVED_RANDOM_GENERATOR_CRG;

end architecture MyHDL;
```

Pusiol, 2021-07-21:
@josyb Well, I see, it works. But synthesizers (at least xilinx vivado) work best with small, well-delimited modules. In addition, parameter support would allow some level of customization within the synthesis tool itself, which speeds up the workflow and allows a myhdl solution to be combined with legacy vhdl/verilog ips or used by third parties with some flexibility without having to install myhdl tools and recompile modules again.

That said, I think it would be very useful to be able to define parameters in modules, specifying the type of data to be overridden and its default value. This does not contradict the philosophy that files generated by myhdl should not be modified, because the parameters would be overwritten in the files where the modules are instantiated. 

Do you think this is too big an effort? My training in compilers was not very extensive so I don't know exactly where the change should be made, but with some guidance I can try to do that.

josyb, 2021-07-21:
>  But synthesizers (at least xilinx vivado) work best with small, well-delimited modules.
In the end Vivado will have a flattened netlist as well - so I don't think that matters. I have generated large VHDL files and never saw a problem with that.

> In addition, parameter support would allow some level of customization within the synthesis tool itself, which speeds up the workflow
If Xilinx had done a decent job in its IP integrator and implemented the callback function we could generate the V* file on the fly. I did this for Quartus' Qsys. But today I also use Xilinx FPGA only. I can do perfectly without IP integrator (except for the MicroBlaze stuff).

Parameters may be a bit more work than we think; I dabbled on that some time ago and recently as well. Perhaps one day.
I have a *Parameters* branch on my GitHub MyHDL repo. You can compare if you like. But it is only initial stuff and not even the small counter example I have there is completely parameterized; there is one *magic* number left :(

Pusiol, 2021-07-22:
Thank you, i will read the myhdl core and try something.