myhdl/myhdl #332: Support of conversion of slice type

LarsRlrs, 2020-03-31:
Hi,

the request is to support the slice type (for Python2 and Python3, if possible)

```

MySlice=slice(3,0)
from myhdl import *

@block
def inc(Sin, Sout):

    @always_comb
    def combed():
        Sout.next=Sin[MySlice]

    return instances()

def convert_inc(hdl):

    S1,S2=[Signal(intbv(0)[8:]) for k in range(2)]
    
    inc_1 = inc(S1,S2)

    inc_1.convert(hdl=hdl)
```

Best regards,
Lars

hackfin, 2020-03-31:
This branch has some basic support for named slices, but VHDL output support only:
https://github.com/hackfin/myhdl/tree/upgrade

LarsRlrs, 2020-08-31:
Any chance to merge this into the main repo for Python3?

hackfin, 2020-08-31:
I've redrawn #331, and I think it's somewhat void to invest into the 'legacy' structure with respect to conversion, eventually, the whole conversion business from my side is likely going to be ported/follow the path currently pursued on the https://github.com/hackfin/myhdl/tree/jupyosys branch (introducing '@blackbox' and hierarchy preservation) and uses some layer of abstraction to generate VHDL/Verilog/C++ from. Eventually, this might offload completely to a yosys plugin for VHDL (using their internal RTLIL representation).