myhdl/myhdl #341: verilog code conversion not showing in the respective folder

LavanyaMa, 2020-05-02:
The below shown is my code, but after execution, it is neither giving output nor showing any error. please help me out in this.

from myhdl import *
@block
def xor11(z1,x1,x2):
    @always_comb
    def comb1():
        z1.next=x1^x2
    return instances()
@block
def xor2(z2,x3,x4):
    @always_comb
    def comb2():
        z2.next=x3^x4
    return instances()
@block
def xnor11(z3,x1,x2):
    @always_comb
    def comb3():
        z3.next=not (x1^x2)
    return instances()
@block
def xnor2(z4,x3,x4):
    @always_comb
    def comb4():
        z4.next=not (x3^x4)
    return instances()
@block
def mux2(mux2out,xor11,xnor11,xor2):
    @always_comb
    def comb5():
        if xor2==True:
            mux2out.next=xnor1
        else:
            mux2out.next=xor1
    return instances()
@block
def sum1(sumout,mux2,Cin):
    @always_comb
    def comb6():
        if Cin==True:
            sumout.next=not mux2
        else:
            sumout.next=mux2
    return instances()
@block
def carry(carout,Cin,x4,mux2):
    @always_comb
    def comb7():
        if mux2==True:
            carout.next=Cin
        else:
            carout.next=x4
    return instances()
@block
def cout(cop,x1,x3,xor11):
    @always_comb
    def comb8():
        if xor1==True:
            cop.next=x3
        else:
            cop.next=x1
    return instances()

if __name__ == '__main__':

    def convert():
    
        x1=Signal(bool(0))
        x2=Signal(bool(0))
        x3=Signal(bool(0))
        x4=Signal(bool(0))
        Cin=Signal(bool(0))
        z1=Signal(bool(0))
        z2=Signal(bool(0))
        z3=Signal(bool(0))
        z4=Signal(bool(0))
        mux2out=Signal(bool(0))
        sumout=Signal(bool(0))
        carout=Signal(bool(0))
        cop=Signal(bool(0))
        xor11=Signal(bool(0))
        xnor11=Signal(bool(0))
        xor2=Signal(bool(0))
        mux2=Signal(bool(0))
    
    
        xor11_verilog=xor11(z1,x1,x2)
        xor11_verilog.convert(hdl="Verilog", name="xor11",path="Verilog", initial_values=True)
        
        xnor11_verilog=xnor11(z3,x1,x2)
        xnor11_verilog.convert(hdl="Verilog",name="xnor11",path="Verilog",initial_values=True)
        
        xor2_verilog=xor2(z2,x3,x4)
        xor2_verilog.convert(hdl="Verilog", name="xor2",path="Verilog",initial_values=True)
        
        xnor2_verilog=xnor2(z4,x3,x4)
        xnor2_verilog.convert(hdl="Verilog",name="xnor2",path="Verilog", initial_values=True)
        
        mux2_verilog=mux2(mux2out,xor11,xnor11,xor2)
        mux2_verilog.convert(hdl="Verilog", name="mux2",path="Verilog",initial_values=True)
        
        sum1_verilog=sum1(sumout,mux2,Cin)
        sum1_verilog.convert(hdl="Verilog",name="sum1",path="Verilog", initial_values=True)
        
        carry_verilog=carry(carout,Cin,x4,mux2)
        carry_verilog.convert(hdl="Verilog",name="carry",path="Verilog", initial_values=True)
        
        cout_verilog=cout(cop,x1,x3,xor11)
        cout_verilog.convert(hdl="Verilog", name="cout",path="Verilog",initial_values=True)
    
        convert()

NicoPy, 2020-05-02:
Please, format your code correctly using markdown formatting.
Which MyHDL version do you use ?

LavanyaMa, 2020-05-03:
[myhdl code.docx](https://github.com/myhdl/myhdl/files/4569615/myhdl.code.docx)

Thanks for the reply. I am using MyHDL 0.11

josyb, 2020-05-03:
Try "un"-indenting the convert() call on the very last line ...

josyb, 2020-05-03:
But that will then start throwing errors.

Is this a *homework* assignment?

LavanyaMa, 2020-05-03:
Thanks for the reply. ya I tried unindenting got the error message. what to
do now? please help me in this regard, trying hard from past one week

NicoPy, 2020-05-03:
Please edit your post and format your code correctly : https://www.markdownguide.org/extended-syntax/#syntax-highlighting

LavanyaMa, 2020-05-03:
> Please edit your post and format your code correctly : https://www.markdownguide.org/extended-syntax/#syntax-highlighting

I have attached the word document describing each block, its inputs and outputs. Is this not sufficient. what is the necessity of marking down. Sorry I am new to markdown

josyb, 2020-05-03:
You can add code to your post by enclosing the code  between to sets of triple backticks:

\`\`\`

multiple lines of code 

\`\`\`

LavanyaMa, 2020-05-03:
@josyb 
```
# The below code is for the 4:2 compressor using xor-xnor and mux gates where x1,x2,x3,x4,Cin are the inputs with sum, carry and cout outputs

 

from myhdl import *
#xor block where x1,x2 inputs and z1 output
@block
def xor11(z1,x1,x2):
    @always_comb
    def comb1():
        z1.next=x1^x2
    return instances()
#xor block where x3,x4 inputs and z2 output
@block
def xor2(z2,x3,x4):
    @always_comb
    def comb2():
        z2.next=x3^x4
    return instances()
#xnor block where x1,x2 inputs and z3 output
@block
def xnor11(z3,x1,x2):
    @always_comb
    def comb3():
        z3.next=not (x1^x2)
    return instances()
#xnor block where x3,x4 inputs and z4 output
@block
def xnor2(z4,x3,x4):
    @always_comb
    def comb4():
        z4.next=not (x3^x4)
    return instances()
#The functions are passed as the arguments i.e., xor11,xnor11,xor2 are the functions and mux2out is the output. The outputs of xor11,xnor11,xor2 are the inputs to mux
@block
def mux2(mux2out,xor11,xnor11,xor2):
    @always_comb
    def comb5():
        if xor2==True:
            mux2out.next=xnor1
        else:
            mux2out.next=xor1
    return instances()
# The output of mux2 and Cin are the inputs and sumout is the output
@block
def sum1(sumout,mux2,Cin):
    @always_comb
    def comb6():
        if Cin==True:
            sumout.next=not mux2
        else:
            sumout.next=mux2
    return instances()
# The output of mux2, Cin, x4 are the inputs and carout is the output
@block
def carry(carout,Cin,x4,mux2):
    @always_comb
    def comb7():
        if mux2==True:
            carout.next=Cin
        else:
            carout.next=x4
    return instances()
# The output of xor11, x1, x3 are the inputs and cop is the output
@block
def cout(cop,x1,x3,xor11):
    @always_comb
    def comb8():
        if xor1==True:
            cop.next=x3
        else:
            cop.next=x1
    return instances()

if __name__ == '__main__':
#The below is the code for myhdl to verilog conversion

    def convert():
    
        x1=Signal(bool(0))
        x2=Signal(bool(0))
        x3=Signal(bool(0))
        x4=Signal(bool(0))
        Cin=Signal(bool(0))
        z1=Signal(bool(0))
        z2=Signal(bool(0))
        z3=Signal(bool(0))
        z4=Signal(bool(0))
        mux2out=Signal(bool(0))
        sumout=Signal(bool(0))
        carout=Signal(bool(0))
        cop=Signal(bool(0))
        xor11=Signal(bool(0))
        xnor11=Signal(bool(0))
        xor2=Signal(bool(0))
        mux2=Signal(bool(0))
    
    
        xor11_verilog=xor11(z1,x1,x2)
        xor11_verilog.convert(hdl="Verilog", name="xor11",path="Verilog", initial_values=True)
        
        xnor11_verilog=xnor11(z3,x1,x2)
        xnor11_verilog.convert(hdl="Verilog",name="xnor11",path="Verilog",initial_values=True)
        
        xor2_verilog=xor2(z2,x3,x4)
        xor2_verilog.convert(hdl="Verilog", name="xor2",path="Verilog",initial_values=True)
        
        xnor2_verilog=xnor2(z4,x3,x4)
        xnor2_verilog.convert(hdl="Verilog",name="xnor2",path="Verilog", initial_values=True)
        
        mux2_verilog=mux2(mux2out,xor11,xnor11,xor2)
        mux2_verilog.convert(hdl="Verilog", name="mux2",path="Verilog",initial_values=True)
        
        sum1_verilog=sum1(sumout,mux2,Cin)
        sum1_verilog.convert(hdl="Verilog",name="sum1",path="Verilog", initial_values=True)
        
        carry_verilog=carry(carout,Cin,x4,mux2)
        carry_verilog.convert(hdl="Verilog",name="carry",path="Verilog", initial_values=True)
        
        cout_verilog=cout(cop,x1,x3,xor11)
        cout_verilog.convert(hdl="Verilog", name="cout",path="Verilog",initial_values=True)
    
        convert()

```

NicoPy, 2020-05-03:
You have defined a function named *xor11* at the beginning of the module.
You also defined a signal named *xor11* in the *convert()* function. The signal definition masks the function definition.
You did the same error for other functions/signals (xor2, xnor11...).

LavanyaMa, 2020-05-04:
If the xor11,xor2,xnor11,xnor2 signals are not defined in the convert function, the complete spyder file if i run, it is not throwing error. But when a part of code like
```
xor11_verilog=xor11(z1,x1,x2)
        xor11_verilog.convert(hdl="Verilog", name="xor11",path="Verilog", initial_values=True)
```
is run then the following error is obtained.

  File "<ipython-input-12-afba357acbb1>", line 1, in <module>
    xor11_verilog=xor11(z1,x1,x2)

NameError: name 'z1' is not defined

NicoPy, 2020-05-05:
It looks like you need to learn Python before using MyHDL...

LavanyaMa, 2020-05-05:
I have written python code, where it is giving correct output for the 4: 2 compressor. But, when I am trying to write the code, there is problem with arguments which are the functions

NicoPy, 2020-05-06:
I don't open .docx files since it is not secure.
If you want help, you need to post your code here. Before posting your code, reduce it to a minimal excerpt showing the problem.

LavanyaMa, 2020-05-06:
myhdl code which is shown above has been written first in python as shown below.
```
class xornor1:
    def __init__(self,x1,x2,x3,x4,cin):
        self.fval=x1
        self.sval=x2
        self.x3=x3
        self.x4=x4        
        self.cin=cin
    def xor1(self):
        if(self.x1==self.x2):
            return 0
        else:
            return 1
    def xnor1(self):
        if(self.x1==self.x2):
            return 1
        else:
            return 0
    def xor2(self):
        if(self.x3==self.x4):
            return 0
        else:
            return 1
    def xnor2(self):
        if(self.x3==self.x4):
            return 1
        else:
            return 0
    
x1=xornor1(1,0,1,1,1)
print(x1.xor1())
print(x1.xnor1())   
x2=xornor1(1,0,1,1,1)
print(x2.xor2())
print(x2.xnor2())
  
Mux2=((x1.xor1() and x2.xnor2()) or (x1.xnor1() and x2.xor2()))
print(Mux2)
Sum=((x1.cin and (not Mux2)) or ((not x1.cin) and Mux2))
print(Sum)
Carry=(((not Mux2) and x1.x4) or (Mux2 and x1.cin))
print(Carry)
Cout=((x1.xnor1() and x1.fval) or (x1.xor1() and x1.x3))
print(Cout)
```

closed 2020-07-01 by LavanyaMa