myhdl/myhdl #343: ConversionError: non-boolean argument in logical operator

LavanyaMa, 2020-06-12:
I am using myhdl 0.11 version, Please kindly solve this issue
```
from myhdl import *

@block
def final42(x1,x2,x3,x4,cin,sum1,carry,cout):
    
    @always_comb
    def logic():
        xor1=x1 ^ x2
        xor2=x3 ^ x4
        xnor1=not(x1 ^ x2)
        xnor2=not(x3 ^ x4)
        mux2=((xor1 and xnor2) or (xnor1 and xor2))
        sum1.next=((cin and (not mux2)) or ((not cin) and mux2))
        carry.next=(((not mux2) and x4) or (mux2 and cin))
        cout.next=((xor1 and x1) or (xor1 and x3))
    return logic

x1=Signal(bool(0))
x2=Signal(bool(0))
x3=Signal(bool(0))
x4=Signal(bool(0))
cin=Signal(bool(0))
sum1=Signal(bool(0))
carry=Signal(bool(0))
cout=Signal(bool(0))
        
final_42=final42(x1,x2,x3,x4,cin,sum1,carry,cout)
final_42.convert(hdl='VERILOG',name="final42",path=".",testbench=False,initial_values=True)
```
i am getting the following error when i run the above code:
```
  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 704, in runfile
    execfile(filename, namespace)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 108, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/Vamshi/.spyder-py3/final42.py", line 35, in <module>
    final_42.convert(hdl='VERILOG',name="final42",path=".",testbench=False,initial_values=True)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\_block.py", line 342, in convert
    return converter(self)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_toVerilog.py", line 177, in __call__
    genlist = _analyzeGens(arglist, h.absnames)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_analyze.py", line 170, in _analyzeGens
    v.visit(tree)

  File "C:\Users\Vamshi\Anaconda3\lib\ast.py", line 253, in visit
    return visitor(node)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_analyze.py", line 1119, in visit_Module
    _AnalyzeBlockVisitor.visit_Module(self, node)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_analyze.py", line 1070, in visit_Module
    self.generic_visit(node)

  File "C:\Users\Vamshi\Anaconda3\lib\ast.py", line 261, in generic_visit
    self.visit(item)

  File "C:\Users\Vamshi\Anaconda3\lib\ast.py", line 253, in visit
    return visitor(node)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_analyze.py", line 1099, in visit_FunctionDef
    self.visit(n)

  File "C:\Users\Vamshi\Anaconda3\lib\ast.py", line 253, in visit
    return visitor(node)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_analyze.py", line 528, in visit_Assign
    self.visit(value)

  File "C:\Users\Vamshi\Anaconda3\lib\ast.py", line 253, in visit
    return visitor(node)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_analyze.py", line 446, in visit_BoolOp
    self.visit(n)

  File "C:\Users\Vamshi\Anaconda3\lib\ast.py", line 253, in visit
    return visitor(node)

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_analyze.py", line 450, in visit_BoolOp
    "non-boolean argument in logical operator")

  File "C:\Users\Vamshi\Anaconda3\lib\site-packages\myhdl\conversion\_misc.py", line 148, in raiseError
    raise ConversionError(kind, msg, info)

ConversionError: non-boolean argument in logical operator
```

NicoPy, 2020-06-12:
The `final42` function must be defined like the following :
```python
@block
def final42(x1,x2,x3,x4,cin,sum1,carry,cout):

    xor1=Signal(bool(0))
    xor2=Signal(bool(0))
    xnor1=Signal(bool(0))
    xnor2=Signal(bool(0))
    mux2=Signal(bool(0))
    
    @always_comb
    def logic1():
        xor1.next=x1 ^ x2
        xor2.next=x3 ^ x4
        xnor1.next=not(x1 ^ x2)
        xnor2.next=not(x3 ^ x4)

    @always_comb
    def logic2():
        mux2.next=((xor1 and xnor2) or (xnor1 and xor2))

    @always_comb
    def logic3():
        sum1.next=((cin and (not mux2)) or ((not cin) and mux2))
        carry.next=(((not mux2) and x4) or (mux2 and cin))
        cout.next=((xor1 and x1) or (xor1 and x3))

    return logic1, logic2, logic3
```
_Undefined variables_ can't be used in `@always_comb` functions.
Inside `@always_comb` functions, a signal can't be used as input __and__ output hence the multiple `@always_comb` functions.

LavanyaMa, 2020-06-13:
@NicoPy 

Thank you very much. It is working fine.

closed 2020-06-13 by LavanyaMa