myhdl/myhdl #344: how to assign input variable to weights of a neural network

LavanyaMa, 2020-07-01:
```
import numpy as np
import keras
model = keras.models.Sequential()
model.add(keras.layers.Dense(2, activation='relu', input_shape=(2,),use_bias=False)) # hidden layer
model.add(keras.layers.Dense(1,activation='softmax',use_bias=False)) # output layer
w1=np.zeros((2,2)) # two input neurons for two neurons at the hidden layer
w2=np.zeros((2,1)) # two input neurons for one output neuron
def __init__(self,x1,x2):
        self.x1=x1
        self.x2=x2
        w1[0,0]=not x2
        w1[0,1]=0
        w1[1,0]=0
        w1[1,1]=not x1
w2[0,0]=1
w2[1,0]=1
model.set_weights([w1, w2])
x = np.array([
[0, 0],
[0, 1],
[1, 0],
[1, 1],
])
model.predict(x)
```
[image]

I want to implement 2 input XOR gate for the above shown neural network. I want to assign weights directly as not x1, not x2 to get correct output. I tried with different Activation functions. When linear, sigmoid, softmax activation functions are used at output, all outputs obtained are 0's, 0.5's, 1's respectively. Kindly help me in getting correct output. I am using python 3.6

josyb, 2020-07-01:
Dear,

These kind of issues are best discussed on: https://discourse.myhdl.org
There are more people watching there than over here.
Also I don't exactly see where this would be a MyDHL related issue.

Best regards,
Josy

LavanyaMa, 2020-07-01:
sorry this is not MyHDL code, It is python code.