myhdl/myhdl #346: Asynchronous Reset but not every FF, VHDL

LarsRlrs, 2020-08-31:
Hello,

MyHDL Version: 0.11
Python Version: 2 and 3

Converting the attached code to VHDL causes an error: No separate else clause found
I was unable to achieve the desired behavior with other types of notation (e.g. always_seq).
It is unfeasible to put such signals into separate processes. It wouldn't make sense that myhdl is more restrictive than VHDL regarding this issue.

Is there a workaround?

[test.py.txt](https://github.com/myhdl/myhdl/files/5150382/test.py.txt)

No issues with synchronous reset.

Many thanks and best regards, Lars

josyb, 2020-08-31:
This is a syndrome of  the (in-)famous Olof Kindgren reset-approach. (http://olofkindgren.blogspot.com/2017/11/resetting-reset-handling.html)
I suggest you use two processes, one with a reset for out1 and a second without reset for out2.
Also always use `@always_seq,` for the second process specify `reset=None` and your intentions a clear to everyone.

Regards,
Josy

LarsRlrs, 2020-08-31:
@josyb The code above is a simplified example. In reality, internal signals, some of which shall have a reset and some of which must not have a reset, are interacting with each other.  This is true for example for arrays that are expected to be mapped to distributed/LUT RAM and shift registers.
If I follow your recommendation, two processes will not be enough. Separating this into two processes forces me to go from 1-process-notation to 2-process-notation. So I will end up with three to four processes and a lot more signals instead of one process. Readable, easily understandable code is of great importance; resources and timing are as well. 

Regarding Olof Kindgrens reset-approach: I would prefer not to discuss whether certain design styles are good or bad. My understanding is that myhdl is supposed to increase freedom and possibilities.

It seems like Verilog is ok, the problem is with conversion to VHDL.
Is it intentional that conversion of this code to VHDL fails?

Best regards, Lars

NicoPy, 2020-08-31:
Your code fails because the management of reset is automatically handled by MyHDL. 
When using a `ResetSignal`, MyHDL automatically adds a _reset statement_ that resets __all__  the signals assigned in the _process_. You can't use the reset signal in the _process_ by yourself.
However, you are not forced to use a `ResetSignal`. You can manage the reset yourself as you would do with VHDL by using a standard `Signal`.

LarsRlrs, 2020-08-31:
@NicoPy I have modified the example according to your recommendation and attached the code in this comment. The code fails as well. Conversion for synchronous resets works, conversion for asynchronous resets to VHDL seems to fail.

Thanks for the help at all.
Best regards Lars

[test.py.txt](https://github.com/myhdl/myhdl/files/5151971/test.py.txt)

josyb, 2020-08-31:
@LarsRlrs I always write 3-process state machines, one combinatorial, one synchronous with reset and another synchronous but without reset.  (Sometimes the third is not there ...) I never needed a fourth process. 
Quite often the people writing 1-process state machines still end up with a second asynchronous process (where  they copy a lot of code of the first process ....).
I object to Olof Kindgren's claim that his blog is the better reset way ... quote: "I _will make a bold statement. You are probably doing your reset wrong in your RTL code._"
You can read my ideas on this here: https://gist.github.com/josyb/efec12907bb82a06e42758c039767c7f

But you have identified the problem: VHDL can't handle asynchronous resets as OK describes (for Verilog).
The example in test.py implicitly specifies the Reset with an `edge`; VHDL doesn't like this; Verilog fakes it by treating the edge as a continuous signal.

Best regards,
Josy

josyb, 2020-08-31:
> My understanding is that myhdl is supposed to increase freedom and possibilities.
Yes, but MyHDL is not Verilog in disguise -- although one might think so because of the `@always` decorators.

NicoPy, 2020-08-31:
@LarsRlrs Can you provide the resulting VHDL code you expect ?
Please embed the code in the body of your message (using markdown code tag), not in an external file. It will be much easier to read this thread.

LarsRlrs, 2020-08-31:
Version1:
```
from myhdl import *
@block
def inc(clk, RESETn, in1, in2, out1, out2):
    @always(clk,RESETn)
    def seq():
        if clk.posedge:
            out1.next=in1 or in2
            out2.next=in1 and in2      
        if RESETn==0:
            out1.next=0
    return instances()

def convert_inc():
    clk=Signal(bool(0))
    RESETn=Signal(bool(0))
    
    in1,in2,out1,out2=[Signal(bool(0)) for k in range(4)]
    inc_1 = inc(clk, RESETn, in1, in2, out1, out2)

    inc_1.convert('VHDL')
```
Version 1 conversion (conversion result is as expected but does not simulate correctly in myhdl):

```
architecture MyHDL of inc is
begin

INC_SEQ: process (clk, RESETn) is
begin
    if rising_edge(clk) then
        out1 <= stdl(bool(in1) or bool(in2));
        out2 <= stdl(bool(in1) and bool(in2));
    end if;
    if (RESETn = '0') then
        out1 <= '0';
    end if;
end process INC_SEQ;

end architecture MyHDL;
```

Version 2:
```
from myhdl import *
@block
def inc(clk, RESETn, in1, in2, out1, out2):
    @always(clk,RESETn)
    def seq():
        if bool(clk.posedge) and clk==1:
            out1.next=in1 or in2
            out2.next=in1 and in2      
        if RESETn==0:
            out1.next=0
    return instances()

def convert_inc():
    clk=Signal(bool(0))
    RESETn=Signal(bool(0))
    
    in1,in2,out1,out2=[Signal(bool(0)) for k in range(4)]
    inc_1 = inc(clk, RESETn, in1, in2, out1, out2)

    inc_1.convert('VHDL')
```

Version 2 conversion (simulation seems to be ok but conversion is not ok; synthesis will use CE):

```
architecture MyHDL of inc is
begin

INC_SEQ: process (clk, RESETn) is
begin
    if (rising_edge(clk) and (clk = '1')) then
        out1 <= stdl(bool(in1) or bool(in2));
        out2 <= stdl(bool(in1) and bool(in2));
    end if;
    if (RESETn = '0') then
        out1 <= '0';
    end if;
end process INC_SEQ;

end architecture MyHDL;
```

Anything with @always(clk.posedge....) makes things worse.

NicoPy, 2020-09-01:
@LarsRlrs That's what you get. But what is the output you expect to have?

LarsRlrs, 2020-09-01:
@NicoPy 
The version1 conversion result is the output that I expect to have. But the version1 myhdl code does not simulate correctly. It seems that this is executed twice for each clk rising edge.

It is my understanding that the notation in myhdl should include `@always(clk.posedge)`, but I found no way to get to the expected VHDL conversion result with this. The reason seems to be that myhdl-conversion is searching for an `else` to the `if RESETn==0`, but there is no `else` and so myhdl-conversion raises an error.
```
    @always(clk.posedge, RESETn.negedge)
    def seq():
        out1.next=in1 or in2
        out2.next=in1 and in2
        
        if RESETn==0:
            out1.next=0
```
leads to `No separate else clause found
`
And
```
    @always(clk.posedge, RESETn)
    def seq():
        out1.next=in1 or in2
        out2.next=in1 and in2
        
        if RESETn==0:
            out1.next=0
```
leads to `base type error in sensitivity list`
VHDL conversion result for synchronous reset is working.
VHDL conversion result for asynchronous reset is not worling.

Again, here is the expected/desired conversion result (VHDL with asynchronous reset, but not every FF):
```

architecture MyHDL of inc is
begin

INC_SEQ: process (clk, RESETn) is
begin
    if rising_edge(clk) then
        out1 <= stdl(bool(in1) or bool(in2));
        out2 <= stdl(bool(in1) and bool(in2));
    end if;
    if (RESETn = '0') then
        out1 <= '0';
    end if;
end process INC_SEQ;

end architecture MyHDL;
```

josyb, 2020-09-01:
Does your desired construct synthesize?
In all literature the standard VHDL asynchronous reset follows this template:
```
process (Clock, Reset)
begin
  if Reset = '0' then
    -- reset register, Q <= 0
  elsif RISING_EDGE(Clock) then
    -- increment register, Q <= Q + 1;
  end if;
end process;
```

Again, _MyHDL != Verilog_. 
IMO the only way is to use two processes; one with a (either synchronous or asynchronous) reset and the second without. The `@always_seq` construct was specifically added to simplify the choice of  reset signal (active high or low; synchronous or asynchronous; or simply None) without having to update the MyHDL code depending on which FPGA vendor is targeted.
I understand that this conflicts with the one-process state machine approach.

LarsRlrs, 2020-09-01:
@josyb 
> Does your desired construct synthesize?

Yes.

> In all literature the standard VHDL asynchronous reset follows

The same literature that states one shall use Python for HDL? ;)
It took until VHDL-2019 for VHDL to include interfaces.
Writing the reset at the end is much older than 2017. We are not waiting for the literature :)

> Again, _MyHDL != Verilog_.
> IMO the only way is to use two processes; one with a (either synchronous or asynchronous) reset and the second without.

Verilog synchronous: works
Verilog asynchronous: works
VHDL synchronous: works
VHDL asynchronous: does not work

Could this be fixed so that mhdl-conversion does not insist on the else-clause of the Reset but uses the stuff before the Reset for rising clock edge?

> The `@always_seq` construct was specifically added to simplify the choice of reset signal (active high or low; synchronous or asynchronous; or simply None) without having to update the MyHDL code depending on which FPGA vendor is targeted.

The FPGA vendor Xilinx recommends to reset as few FFs as possible.

> I understand that this conflicts with the one-process state machine approach.

Your recommendation to use more than one process is cleaner from a design guide perspective and this almost makes me want to rewrite code. But it really increases the code size. I would have to use an additional combinatorial process because two sequential processes alone will not do (additional clock cycle from one sequential process to the other).

josyb, 2020-09-01:
> Could this be fixed so that mhdl-conversion does not insist on the else-clause of the Reset but uses the stuff before the Reset for rising clock edge?

Probably, but I wonder what priority it will get and who is going to make the PR.

>The FPGA vendor Xilinx recommends to reset as few FFs as possible.

Xilinx also recommends _synchronous_ resets?

> But it really increases the code size. I would have to use an additional combinatorial process because two sequential processes alone will not do (additional clock cycle from one sequential process to the other).

And then you have the 3-process approach. I don't find that it increases the (source) code size that much. And it does not increase resource usage - win some, lose some.

LarsRlrs, 2020-09-01:
> Probably, but I wonder what priority it will get and who is going to make the PR.

Hopefully, it is easy and not much effort to fix.

> Xilinx also recommends _synchronous_ resets?

Yes. But I can't do that at the moment.

> And then you have the 3-process approach. I don't find that it increases the (source) code size that much. And it does not increase resource usage - win some, lose some.

I'd say, up to 50% and beyond is significant. Starting with one processes, you need one additional combinatorial signal for every interaction between functionality that is then separated into two sequential processes. Multiple interactions per state and a couple of states et voila: "ldlq", "ldlqw", "sellqw" (https://gist.github.com/josyb/efec12907bb82a06e42758c039767c7f). Much more for modules with a couple of counters and ShiftRegs.

NicoPy, 2020-09-01:
@LarsRlrs 
From what I understand, `@always` intended use is like this :
```python
    @always(clk, RESETn)
    def seq():
        if RESETn == 0:
            out1.next=0
        elif clk.posedge :
            out1.next=in1 or in2
            out2.next=in1 and in2

    return instances()
```
You will notice there is an `else` ;)
It translates to the following :
```vhdl
INC_SEQ: process (clk, RESETn) is
begin
    if (RESETn = '0') then
        out1 <= '0';
    elsif rising_edge(clk) then
        out1 <= stdl(bool(in1) or bool(in2));
        out2 <= stdl(bool(in1) and bool(in2));
    end if;
end process INC_SEQ;
```

For a reset at the end of the process, your version1 converts correctly but does not simulate. Maybe this is the way to go and try to make simulation work corectly.

LarsRlrs, 2020-09-01:
@NicoPy 
Just one comment: The code that you wrote, makes out2 dependent from RESETn.

NicoPy, 2020-09-02:
@LarsRlrs Yes. I was showing the way it is intended to be used (from my understanding).

Another code that suits your requirements is the following  :
```python
    @always(clk.posedge, RESETn.negedge)
    def seq():
        if clk.posedge :
            out1.next=in1 or in2
            out2.next=in1 and in2
        
        if RESETn == 0:
            out1.next=0
        else :
            pass
```
MyHDL complains there is no else ? Well... just add an else ;)

However, the conversion result is surprising :
```vhdl
INC_SEQ: process (clk, RESETn) is
begin
    if rising_edge(clk) then
        out1 <= stdl(bool(in1) or bool(in2));
        out2 <= stdl(bool(in1) and bool(in2));
    end if;
    if (RESETn = '0') then
        out1 <= '0';
    elsif rising_edge(clk) then
        null;
    end if;
end process INC_SEQ;
```
I have not checked if it simulates correctly.

LarsRlrs, 2020-09-03:
@NicoPy 

> I have not checked if it simulates correctly.

It seems that it does not simulate correctly.
Maybe among other things, some combinatorial processes are not being informed correctly about signals that are written in this sequential process. Also: Not sure, if there is a difference between Python2 and Python3 (worse in Python3).

NicoPy, 2020-09-03:
> It seems that it does not simulate correctly.

This is not a surprise.
Anyway, the `@always`parameters are not good. `.negedge`should not be used here. At least for `RESETn'.
I still think the way to go is to fix the simulation.