myhdl/myhdl #349: Verilog Conversion and Simulation Handle tuples of enums differently

venks1, 2020-12-02:
I would like to index an array which is populated with enums.   I can get this to work in both simulation and conversion with different syntax.   

The files below are `a_sim.py` for simulation and `a_vlg.py` for conversion.   They both work fine.  But if you look at the declaration of the `imap` variable - it is different.    It would be preferable (or even consistent) to allow the same syntax to work for both.   

a_sim.py is below:

    from myhdl import (enum, always_comb, instances, block, Signal, intbv)
    
    alop = enum('NOP', 'LD', 'AND', 'OR', 'XOR', 'ADD', 'SUB', 'SHR')
    
    @block
    def dec(a, b):
        imap = (alop.LD, alop.AND, alop.OR, alop.XOR, alop.LD, alop.ADD, alop.SUB, alop.SHR)
        @always_comb
        def dr():
            pre = imap[a[3:]]
            b.next = pre
        return instances()
    
    @block
    def tb():
        a = Signal(intbv(0)[3:])
        b = Signal(alop.NOP)
        d = dec(a, b)
        return instances()
    
    inst = tb()
    inst.run_sim(10)

a_vlg.py is below:

    from myhdl import (enum, always_comb, instances, block, Signal, intbv)
    
    alop = enum('NOP', 'LD', 'AND', 'OR', 'XOR', 'ADD', 'SUB', 'SHR')
    
    @block
    def dec(a, b):
        imap = tuple([int(i) for i in [alop.LD, alop.AND, alop.OR, alop.XOR, alop.LD,
                                      alop.ADD, alop.SUB, alop.SHR]])
        @always_comb
        def dr():
            pre = imap[a[3:]]
            b.next = pre
        return instances()
    
    a = Signal(intbv(0)[3:])
    b = Signal(alop.NOP)
    d = dec(a, b)
    d.convert(hdl='verilog')

## System information
```
MyHDL Version: 0.11
Python Version: 3.8
```

NicoPy, 2020-12-05:
MyHDL can simulate almost every code. This is not true for conversion. 

The construct you try to use is (currently) not supported in conversion.
The workaround you use for conversion should not work (you assign a `int` to a `enum`).

Any reason you don't use a construct like the following ?
```python
if a == 0 : 
    b.next = alop.LD
elif a == 1 :
    b.next = alop.AND
...
```

venks1, 2020-12-05:
Thanks for the answer. Understood. That was a subset of the problem, but the reasons are just the standard ones - readablity (when all on one screen) and conciseness.   The int conversion strangely does work (pretty well) for conversion.  And tuples seem to be allowed.  Feel free to change the title and make this an enhancement rather than a bug.

NicoPy, 2020-12-06:
> The int conversion strangely does work (pretty well) for conversion. And tuples seem to be allowed. Feel free to change the title and make this an enhancement rather than a bug.

This _"works"_ when the `enum` encoding is `binary` (default). But with `one_hot` or `one_cold` encoding...

venks1, 2020-12-07:
Fair enough.  So maybe this isn't a good use of the myhdl enum class.   I recoded the program to use the [Python enum](https://docs.python.org/3/library/enum.html) instead.   But that has the same problem too.  Simulation passes, but synthesis fails unless I convert the map to be a tuple of ints (rather than a tuple of enums).

venks1, 2020-12-07:
I just converted all uses to int(enum) or intbv(int(enum)) and the same source file works in both cases.  I changed the title to sound like an enhancement.  If you agree, please remove the bug and add enhancement tags.   Otherwise feel free to close this.

josyb, 2020-12-07:
Use ```IntEnum``` the you can do without the ```int()``` casting

NicoPy, 2020-12-07:
> 
> 
> Use `IntEnum` then you can do without the `int()` casting

Verilog looks good but VHDL is broken.

josyb, 2021-02-16:
Really? IntEnum doesn't work in VHDL?