myhdl/myhdl #367: shift_left conversion issue in vhdl

davekeeshan, 2022-04-13:
Summary

When trying to introduce a shift left code, where I want to initialize a value by shift the number 1 into a vector:

    full_length = 16
    q     = Signal(modbv(0)[full_length:])

    
    @always(clk.posedge)
    def assign_clk():
        q.next = 1 << full_length-1

This results in code that does not synthesis inside vivado:

    q <= to_unsigned(shift_left(1, (16 - 1)), 16);

I get these errors:

    [Synth 8-2778] type error near %s ; expected type %s
    [Synth 8-2029] formal arg has no actual or default value ["C:/storage/projects/myhdl/sl.vhd":35]
    [Synth 8-2235] indexed name prefix type unsigned expects 1 dimensions ["C:/storage/projects/myhdl/sl.vhd":35]

If however I manually rewrite the vhdl code to this:

    q <= shift_left(to_unsigned(1, 16), (16 - 1));

It at least gets through synthesis.

I think the issue is that the shift_left function expects a signed or and unsigned value and if it does not get that it fails. It also returns the type of the input, so cast the 1 to an unsigned means that the value returned in unsigned.

```
  function SHIFT_LEFT (ARG: UNSIGNED; COUNT: NATURAL) return UNSIGNED;
  -- Result subtype: UNSIGNED(ARG'LENGTH-1 downto 0)
  -- Result: Performs a shift-left on an UNSIGNED vector COUNT times.
  --         The vacated positions are filled with '0'.
  --         The COUNT leftmost elements are lost.
  -- Id: S.3
  function SHIFT_LEFT (ARG: SIGNED; COUNT: NATURAL) return SIGNED;
  -- Result subtype: SIGNED(ARG'LENGTH-1 downto 0)
  -- Result: Performs a shift-left on a SIGNED vector COUNT times.
  --         The vacated positions are filled with '0'.
  --         The COUNT leftmost elements are lost.
```
I assume there might be a similar issue with shift_right, but have not tested it yet. 

The verilog version synthesizes successfully in vivado.

## System information
```
MyHDL Version: 0.11
Python Version: 3.8.9
```

josyb, 2022-04-13:
Actually IMO in your case the expression should be (simply) evaluated and produce:
`q <= to_unsigned( 0x8000, 16);`
This will need some deep thinking: and some extra work.
The  best workaround is to avoid shifts:
`q.next = 2 ** (FULL_LENGHT - 1)` (The pedantic me uses  full upper case names for constants, aka macros in C)

davekeeshan, 2022-04-13:
@josyb I am not too sure if that is exactly the right thing to do, maybe in this explicit case. But it is moving some of the code compilation away from the synthesis tool and placing it in the python.

However I do have another example that it probably won't work on and it is when in a loop:

    for i in 0 to 16 loop
        q[i].next <= 1 << i;

(I have an array of q values)

(I use upper cases CONSTANTS in VHDL too, but don't in python, it is weird to mix the languages sets, python is case sensitive and I use python coding guidelines there)

davekeeshan, 2022-04-13:
@josyb 

Oh I should point out this is just code to show the bug, I have much more involved code hanging off the example I was working on.

josyb, 2022-04-14:
PEP-8 tells you to use uppercase for constants :)  
And if otherwise I would still use uppercase like in C. 
And I never use the `<<` and `>>` operators (I should check) so I haven't encountered this bug yet. Or perhaps I have once, long ago, and just use the workarounds since.

Back to our muttons;
MyHDL conversion to VHDL is complicated by the fact that the OA tries to write output text as soon as possible; sometimes this messes up the order and adds unnecessary or even more complicating `resize', `to_unsigned` etc.
So it is on the *todo* list.

davekeeshan, 2022-04-14:
I am coming to myhdl as a ASIC designer that has been using python to prototype code for years.  Bit shifting has become so common in that work it is ridiculous (i will use them to divide instead of integer/float divides, where appropriate).  I imagine other python people with a HW background would place bit shift operations in the level of very commonly used. So a bug here is potentially a huge barrier to entry.

I am trying to write a basic subtracting divider in myhdl, more proof of concept more than any thing.  I have a line like this:

    if (n[i] >> (C_FULL_LENGTH-2-i)) >= d[i]:

I want to compare the upper bits of n with a value, d, as a test.  I have used this in the myhdl simulation, and it appears to work correctly.  When I generate the vhdl I get this:

    if (shift_right(n(i), ((C_FULL_LENGTH - 2) - i)) >= d(i)) then

However when I try to simulate it in vivado i do not get the same result. 

After exploring, I think the issue is that shift_right is a barrel shift, it loops bits around the bottom and back in the top.  Whereas in python, the bits get dropped off the end.  Obviously as I was writing python, that was what I expected, but the fact that they don't match is more significant.  Even if the python was changed to match the RTL, it would be weird, but it would be consistent.

I checked the verilog version, it appears to also barrel shift too and give the same wrong result as vhdl, relative to the python result.

josyb, 2022-04-14:
My route: TTL -> PalAsm -> Abel -> (Altera) AHDL -> VHDL -> MyHDL
You may now understand that `<< >>` are *strangers* to me :)

But we will definitely look into this deeper.
You can start by posting a self-contained module so I can see your intentions more clearly and  start from there.

Best regards,
Josy

davekeeshan, 2022-04-14:
I was actually just working on that:

[https://github.com/daxzio/div_gen](https://github.com/daxzio/div_gen)

Strangers are just friends you haven't met yet!  

As a flow maintainer, I usually explore tools to see if they are mature enough to be brought into a flow in house and it is robust enough to handle what we throw at it.  There has been an explosion in scala/chisel the past few years (thanks sifive!) but I think that a python equivalent has the potential to be huge, hence my interest and the work.

josyb, 2022-04-14:
>> but I think that a python equivalent has the potential to be huge, hence my interest and the work.

That is the plan.

davekeeshan, 2022-04-20:
OK I was able to investigate this issue, it isn't a barrel shifter. It is this line:

    n[i+1].next = (n[i] - (d[i] << (C_FULL_LENGTH-2-i)))

The current code becomes:

    n((i + 1)) <= (n(i) - shift_left(d(i), ((24 - 2) - i)));

However that shift_left is only over the length of `d(i)`, which in this case is significantly smaller than `n(i)`.  If i change the vhdl manually to this:

    n(i + 1) <= (n(i) - shift_left(resize(d(i), 24), ((24 - 2) - i)));

So that when `d(i)` gets shifted left, it has enough head room to keep the important bits (at least for this equation)

josyb, 2022-04-20:
To have the convertor add the `resize` before the shift  it has to know the intended size which could (will) be cumbersome.
One could argue that making `d[i]` the same size as `n[i]` is the appropriate thing to do.
I would be inclined to change the simulator to behave as the converted VHDL; if you shift more than the width the bits will drop off (in HW: there are not enough registers).
We will have to ask others for their opinion: @hgomersall

davekeeshan, 2022-04-20:
Doesn't the resize just have to match what on the left side of the equation? 

What ever it is it should be documented as there is what is obvious to a VHDL person and what is obvious to a python person and this seems to lie perfectly in between.

hgomersall, 2022-04-21:
Does VHDL specify that lengths have to be compatible?

Just so I understand, the problem is that the converted code from `n[i+1].next = (n[i] - (d[i] << (C_FULL_LENGTH-2-i)))` gives an error on the VHDL side, which is due to mismatched lengths of `d[i]` and `n[i]`, which is a problem at the subtraction?

How do we handle this for a simple addition?

davekeeshan, 2022-04-21:
@hgomersall The issue here is that the shift length is more than the length of `d[i]` so the resultant value is 0. I am guessing this is also an problem for addition.

Does any one know chisel and understand what they do here?  I am trying to review their code base, but it is in not 100% clear. But i did pick this up from an online source:

https://raw.githubusercontent.com/wiki/schoeberl/chisel-book/unit2.pdf

- Left and right shifts
- Left shift extends bit width
- Right shift reduces bit width

hgomersall, 2022-04-21:
If you shift a bit vector by more than its length, of course it's going to be 0: "Explicit is better than implicit". In this case that's absolutely the right thing to do. Anything else makes the contrary behaviour unnecessarily harder to express.

I'm a bit puzzled how this relates to the original issue which seemed to be a genuine conversion bug...

(bear with me, I'm trying to get in the picture!)

hgomersall, 2022-04-21:
> Does any one know chisel and understand what they do here? I am trying to review their code base, but it is in not 100% clear. But i did pick this up from an online source:
> 
> https://raw.githubusercontent.com/wiki/schoeberl/chisel-book/unit2.pdf

I can't understand that properly. Is it generally safe to extend? Should you sign extend, or perhaps restrict to unsigned? You're absolutely right that the behaviour needs to be thoroughly documented. Extending seems to be opening a potential can of worms...

davekeeshan, 2022-04-21:
well i think because in python a shift doesn't stop at the length of the vector, it will extend (32 bit, 64 bit limited?)

Ultimately the main issue is that they don't match.

    d = Signal(modbv(0)[8:]))
    n = Signal(modbv(0)[16:]))

    (d = 1)

    n.next = d << 12

What would you expect n.next to be?  in python the value lands, in vhdl it is shifted outside it range.

As a python programmer, I am used to what python is doing, (and I assume the goal of `myhdl` is to attract python people, RTL people already know how to RTL) but as a RTL programmer I understand there needs to be limits, these are real bits.  Ultimately I think it should be governed by the size of the LHS of the function. Or minimally an excpetion/warning when a shift if rotating through it range.  It took me a while to debug, I can see this being a recurrent problem.

davekeeshan, 2022-04-21:
https://www.chisel-lang.org/api/latest/chisel3/UInt.html#%3C%3C(that:Int):chisel3.Bits

I have to admit, I don't fully know what they mean by `this` and `that`

Note
       
    The width of the returned [Bits](https://www.chisel-lang.org/api/latest/chisel3/Bits.html) is width of this + that.

hgomersall, 2022-04-21:
It's interesting - a shifted intbv is not actually sized. This means there is an internal representation which is needs to be mapped by the converter. If you do:
```python
a = intbv(1)[8:]
b = a << 10
print(b.len())
```
you get `0`. This is obviously fine in the python world, but how it should be represented in converted code is the issue. It's a while since I thought about this. It would be useful to think how this is handled in other places...

hgomersall, 2022-04-21:
We use left shifts in our code base and haven't encountered any issues. We convert to Verilog mostly, but we do run our test suite with VHDL. In our case, the source and destination vectors are the same length.

davekeeshan, 2022-04-21:
I don't have access to a decent verilog compiler, but this may only affect VHDL. I did run them through the vivado synthesis tool.  This issue has moved on, but there was also a syntactical issue in formatting that vivado barfed on, also related to shift_left

hackfin, 2022-04-23:
If I'm not terribly mistaken, the root cause for all kinds of resize() issues with more complex expressions is, that operations on `intbv` do not track the size, moreover, mostly result in just integers (like `intbv(4) + 1`) is of type `int`).
With Verilog this is somewhat fine, as expressions mostly resolve implicitely, with VHDL being strict, there are plenty of scenarios where it goes wrong or the conversion fixes turn out to be messy and not orthogonal to each other.

The easy workaround for the shift left with a *static* (positive integer) shift value is using concat and LSB zero padding, as this accounts for the length, obviously. Likewise slicing for *logic* right shifts. When dynamic, shift_left/shift_right make sense, but the maximum possible shift value needs again to be accounted for. Only then, a barrel shifter would be inferred by the synthesis (if recognized).

The above modbv case is an interesting one, actually. As it's supposed to be a modulo int, an overflow shift should wrap around (like it does with addition). So, doing a shift op on a modbv value should consequently do a rotate, but 'master' myhdl doesn't carry it out this way.
Other than that, `d << 12` should be reporting an effective size of 20, and warn early on the size mismatch, IMO. Doing an intbv operation out of range would only throw an overflow exception at simulation runtime, which some may have found insufficient for some verification demands .

josyb, 2022-04-24:
> If I'm not terribly mistaken, the root cause for all kinds of resize() issues with more complex expressions is, that operations on intbv do not track the size, moreover, mostly result in just integers (like intbv(4) + 1) is of type int).

Yes, I noticed that the shiftleft produces an unconstrained intbv which can be greater than the original input intbv.max. As I said before I think we should investigate this, and fix it so that the simulation behaves like the converted VHDL code.

> The easy workaround for the shift left with a static (positive integer) shift value is using concat and LSB zero padding, as this accounts for the length, obviously. Likewise slicing for logic right [...]

Actually all of the shiftlefts in the repo-example div_gen are static. So no barrel shifters needed here. The workaround you propose is what I actually do, as a I am weary of the `<<` and `>>` operators. The generated chain of `resize` always make me wonder. Also I feel that the VHDL resize function is confusing; I always have to check whether it is for signed or unsigned vectors; in my VHDL days I used self-made `zeroextend` and `signextend` functions for clarity.

> The above modbv case is an interesting one, actually. As it's supposed to be a modulo int, an overflow shift should wrap around (like it does with addition). So, doing a shift op on a modbv value should consequently do a rotate, but 'master' myhdl doesn't carry it out this way.

`modbv` is somewhat more complex than a modulo int; as it also has a defined minimum value. I wonder how a leftshift of a `modbv(min=3, max=11)' would work and convert. (maybe an academic exercise ...) 

> Other than that, d << 12 should be reporting an effective size of 20, and warn early on the size mismatch, IMO. Doing an intbv operation out of range would only throw an overflow exception at simulation runtime, which some may have found insufficient for some verification demands .

In the early days only Verilog conversion was supported. And as you say Verilog complains less. 

We should revisit _intbv.py and make it return properly constrained intbvs. This way the simulation will match the generated V* code and issue a warning (or raise an exception?) if the shifted value does not fit.

hgomersall, 2022-04-25:
It wouldn't be unreasonable to specify that shifting implies a cast to unsigned, much like a slice does. When you shift, you shift _raw bits_ - it's up to you dear programmer to make of that what you will.

hgomersall, 2022-04-25:
I've actually toyed with the idea of having out-of-range error signals attached to an `intbv` so synthesized values can be range checked at runtime.

davekeeshan, 2022-04-25:
OK, first up  I didn't realise I was using `modbv` over `intbv`, probably because I copied some code somewhere, it muddies the conversation a little. I rewrote my code with `intbv` and it performs in python simulation the same way, arguably they should be different (? see below)

Also what is the ethos, or goal of myhdl?  Is it just to try an emulate another RTL language or is it to give allow access to higher level programming techniques?  I think one of the reasons that Chisel is getting popular is that it is simplifying the code that gets written, even if the underlying code is a spaghetti monster. (I had the (mis)fortune to have to synthesis a NoC from Arteris in to a FPGA, all in verilog, but talk about write-only code, ugh!).  Unfortunately it is the job of this code/group to potentially write and maintain the code that writes the spaghetti code, that is no fun.

Let's break out some real examples.  Below is what I think should be done, but if people here disagree, cool, but lets work out what the correct answers should be, because the conversation is beginning to get a bit abstract and hard to pin back to an implementation (that might be the limit of my imagination).

As, I think LHS guides the results, the choice of `d` as a `intbv` or `modbv` I think is irrelevant, it gets recast.

```
d = Signal(intbv(0)[8:]))  (or modbv(0)[8:])
n = Signal(intbv(0)[16:]))

(d = 0x81)

n.next = d << 1   (n.next = 0x0102)
n.next = d << 12  (n.next = 0x1000)
n.next = d << 15  (n.next = 0x8000)
n.next = d << 16  (n.next = 0x0000)
```
In the case with the LHS as a modbv, I am not entirely sure, this is _**a**_ possible solution, only putting it forward and something to work against.
```
d = Signal(intbv(0)[8:])) (or modbv(0)[8:])
n = Signal(modbv(0)[16:]))

(d = 0x81)

n.next = d << 1   (n.next = 0x0102)
n.next = d << 12  (n.next = 0x1008)
n.next = d << 15  (n.next = 0x8040)
n.next = d << 16  (n.next = 0x0081)
```
There are further examples of what happens when `n` and `d` are used to create `n.next`, but I would like to start with these examples to at least get a understanding of where the thinking is going.

hgomersall, 2022-04-25:
I really don't think `modbv` should imply a barrel shift. In the case of `modbv` being an `intbv` modulo a power of 2, it's easy, the `modbv` should be the same as the `intbv`. In the case of not being a power of two, I would suggest it being shifted intbv with wrapping applied after the shift (same as for powers of 2, but in that case, the wrapped value is the same as the non-wrapped value).

In the case of a constant shift this can all be inferred at conversion time, with resizing done appropriately. In the case of a signal driven shift, it's a bit tricker. How does VHDL handle this case?

hgomersall, 2022-04-25:
Re what is the point of MyHDL - I think I differ a little from @josyb on the end goal, but agree on where we are. @josyb (correct me if I'm wrong) that MyHDL should be a syntactically pleasant way of writing HDL with lots of pythono niceties, like much better testing than offered by V*.

My view is that MyHDL gets super interesting when some higher level concepts can be introduced - lets say unrolling loops prior to conversion, so dicts of signals can be used. I'm not talking about inferring HDL from normal code, but allowing a greater range on constructs that can be represented in Verilog or VHDL, with the converter mapping to a lower level representation.

davekeeshan, 2022-04-25:
Just before I respond to the comments, I was thinking of the simpler case, which is sorta where this issue started, where there is no `d` and all we are talking about is shifting a constant value:
```
n = Signal(intbv(0)[16:]))

n.next = 0x81<< 1   (n.next = 0x0102)
n.next = 0x81<< 12  (n.next = 0x1000)
n.next = 0x81<< 15  (n.next = 0x8000)
n.next = 0x81<< 16  (n.next = 0x0000)
```
As you can see, it is as if in the previous case the `d` has just been cast to an int

Maybe my understand (again) of `modbv` is mixed up, jumping between languages leaves me frazzled sometimes. Is there no configuration that does a barrel shifter? I am not sure if that is a shame or not, I know one sees it a lot in hashing algorithms. Just to be clear in the example I am currently looking at, I do not want a barrel shift.

> Promoting the arguments of the RHS to the size of the LHS could be an option, but it may be much more work than we now think. Making the `<<` and `>>` behave the same for the simulated MyHDL code and the simulated V* code seems the most reasonable option, as @hgomersall says: you shift too much, you get zero.

Sure, but where the zero kicks in is still slightly what we are discussing, I am not sure if it has been fully decided yet. However the easy choice now, to aid implementation is potentially, functionally different than what one might do if they had the time, changing functionality down the road is arguably worse, but the reality is you end up living with the compromise.  Maybe a third class later, that inherits from `intbv` that does it the way I expect (`extbv`, I don't know I am terrible at names), and you lock `intbv` down now to get the python simulation to match the vhdl simulation and leave it that way for ever more.  As usual document so that etc etc ...

> BTW: the V* code generated by MyHDL can be big too (I have generated modules of 30000 lines) but it is still very readable.

Noted, but that could change in the future on how this gets implemented could result in some long lines that can take some mental contortions to debug. Again this would be the problem for the people in this group, it should be transparent to the end user

> MyHDL is not an HLS. I don't think Chisel is either

No but cleaning up things like bitshifting, which is such a HW idea, would be lovely, I know it is still the bane of my existence when I have to implement it in VHDL, especially after I have already debugged a reference design in python.

hgomersall, 2022-04-25:
It would be nice if individual types could dictate their conversion, then it could be delegated to specific types as you suggest @davekeeshan . That currently isn't the case, but replacing the conversion code has been debated to death but I think with only @hackfin doing anything to that end. It would be interesting to know how that fits with his new work...

hgomersall, 2022-04-25:
> @hgomersall: That wouldn't cover the arithmetic right shift, e.g. for a 9 bit value, pythonic:
> 
> `assert hex((-0x20 >> 3) & 0x1ff) == '0x1fc'`

That's interesting. Your point is that the sign bit is not shifted with a right shift? i.e. It doesn't convert an unsigned prior to shifting. We could just define the right behaviour is the python behaviour, with length constraints. I'd be happy with that.

I appreciate the subtlety about what the length constraint should be...

hackfin, 2022-04-25:
The Python '>>' is always arithmetic, thus the MSB always sticks at '1' for negative values. That's also the intbv behaviour for a `intbv()[N:].signed()` instance.
Reading up on the barrel shifter above: not sure what the exact definition is, these only need to be inferred when shifting occurs by **dynamic** signal values. A BS can be configured with or without bit-rotation.

For the 'new kernel work', I had disabled '<<' and '>>' shifting with the standard `Signal` type and made an ad-hoc workaround with a derived Signal type that infers to a `barrelshifter` primitive for dynamic occurences, while making sure old code only uses concatenation/slicing macros only for static shifts. With Verilog HDL translation on the doorstep, I'm still experimenting with the language built-ins (shift_right, >>, >>>) and some HDL specific handling of the intbv derivatives. The static vector sizes are tracked by the kernel operations so far and are close to 'strict as VHDL' (warn often, fail early).
So my current code is: restrict ambiguous/flaky features on the default data types down to a rock-solid minimum and allow developers to derive and implement how they think it should work. This is prone to potential wild growth of custom datatypes, on the other hand, static type hints in the interface (recommended) prevent some shooting into the foot.

hgomersall, 2022-04-25:
> For the 'new kernel work', I had disabled '<<' and '>>' shifting with the standard `Signal` type and made an ad-hoc workaround with a derived Signal type that infers to a `barrelshifter` primitive for dynamic occurences

At risk of derailing the conversation, why would you use a barrel shifter for the dynamic case? That's very at odds with python and current behaviour.

hackfin, 2022-04-25:
Why's that at odds?
I'm using a VHDL variant of this implementation (augmented MyHDL style):

https://github.com/hackfin/myhdl.v2we/blob/master/examples/bshift_compact.ipynb

It's not formally proven, but should cover logic/arithmetic and rotating shifts just fine, just the latter at a limitation to bitvector sizes to (2**x). I recall xst inferring barrel shifters automatically for `shift_left(a, to_integer(sh))`.

Not sure about gist sustainability, but playing back the @davekeeshan 'configuration', there's still a lot of funny stuff going on with these modbvs:

https://gist.github.com/hackfin/174d853a11807028fb987881b83ace1d#file-myhdl_shifting-ipynb

or as a binder launch:

https://mybinder.org/v2/gist/hackfin/174d853a11807028fb987881b83ace1d/HEAD

davekeeshan, 2022-04-25:
@hackfin I am looking at the python myself.  I think the code performs ok-ish with the examples I gave (i gave them as examples of what could be done, not what is being done, just so there was a common reference point)

in my work it seems to get werid when one tries to do something like:

`n.next = n + d << x`

A fairly surface looks at the code is that if you do an add using a `intbv`, what is being returned is a boring old `int` so a lot of the width handling gets lost.

There are a few other places, this seems like the fix required for where we started in intbv:

```
    @property
    def mask(self):
        return self._max-1

    def __lshift__(self, other):
        if isinstance(other, intbv):
            shift = other._val
        else:
            shift = other
        return intbv((int(self._val) << shift) & self.mask, _nrbits=len(self))
```

from:

```
    def __lshift__(self, other):
        if isinstance(other, intbv):
            return intbv(int(self._val) << other._val)
        else:
            return intbv(int(self._val) << other)
```

josyb, 2022-04-25:
> A fairly surface looks at the code is that if you do an add using a intbv, what is being returned is a boring old int so a lot of the width handling gets lost.

The fact that the `intbv` at the RHS returns an `int` is what our BDFL intended. This may shed some light on his thoughts: [These Int's Are Made for Countin'](https://www.jandecaluwe.com/hdldesign/counting.html) He also has/had a strong opinion on Chisel.

> There are a few other places, this seems like the fix required for where we started in intbv:

    @property
    def mask(self):
        return self._max-1

    def __lshift__(self, other):
        if isinstance(other, intbv):
            shift = other._val
        else:
            shift = other
        return intbv((self._val << shift) & self.mask, _nrbits=self._nrbits)

That is what I had/have in mind; constrain returned intbv.  There is some additional error checking to add for the case that the sihifted intbv is unconstrained.

davekeeshan, 2022-04-25:
@josyb 
> The fact that the `intbv` at the RHS returns an `int` is what our BDFL intended. This may shed some light on his thoughts: [These Int's Are Made for Countin'](https://www.jandecaluwe.com/hdldesign/counting.html) 

Let me review.  I have been looking at the code and have seen areas that i think could be improved, but if they have a design paradigm behind them, it might be like that on purpose.  I am not for or against either, it is just that there was some extra info encoded into the `intbv` class that gets lost, on the surface that seems strange to me. However I would like to help the code if I can.

> He also has/had a strong opinion on Chisel.

I have a strong opinion too, i never want to use it, so that is what is motivating me to investigate this work.

hackfin, 2022-04-25:
The above masking will go very wrong when `self._max` is not a power of two. Plus, keep in mind that extending a signal may be desired (as in your example). So, 'fixing' intbv that way would break many, many reference tests. But, you're totally free to derive a foobv and make it operate the way you want. Or to be in full control of the resize behaviour with intbv, split up `n.next = n + d << x` into separate signal assignments, you might want to be flexible with pipelining anyhow.

Rather than packing special behaviour into overriden operator methods for 'foobv', you might also go for infix operator tricks, like explicitely defining a rotating shift op, like:

`a <<rotl>> b`

`intbv` does just the right job when the target HDL is Verilog and the implicit behaviour just works. With VHDL requiring explicit sizing, the lack of tracking in intbv expression chains becomes tricky during conversion and causes a few arithmetic scenarios fail. That's just the status quo.

davekeeshan, 2022-04-25:
> The above masking will go very wrong when `self._max` is not a power of two. Plus, keep in mind that extending a signal may be desired (as in your example). So, fixing 'intbv` that way would break many, many reference tests. 

I feel like i am trying to ride two horses here. I have given 2 possible ways of doing it, and i am still waiting on a comment or decision on which of the two it is.  Based on the decision the inbv could be broken right now, and need to be fixed or not. I know which of the two I'd pick, but it isn't up to me.

Any way, it should be easy to define a suite of expected inputs and outputs, what are they? And lets design to that!

> But, you're totally free to derive a foobv and make it operate the way you want. Or to be in full control of the resize behaviour with intbv, split up `n.next = n + d << x` into separate signal assignments, you might want to be flexible with pipelining anyhow.

Yes i am doing this right now to understand how hard it is (I don't want or expect people to do work that I won't, it just takes a while to get up to speed on a new code base).  

The `intbv` becomes the truncated shifting, you can only shift inside your space, and another class that inherits from `intbv` but performs the way I described.

> MyHDL should be a syntactically pleasant way of writing HDL with lots of pythono niceties

For me, this is that. I am expecting the python outcome, that is not what I see, this is not a complaint, it is just not what I am expecting.

> `intbv` does just the right job when the target HDL is Verilog and the implicit behaviour just works. With VHDL requiring explicit sizing, the lack of tracking in intbv expression chains becomes tricky during conversion and causes a few arithmetic scenarios fail. That's just the status quo.

Yes VHDL breaks my heart, if I could mask all it messiness it would make me very happy.

hgomersall, 2022-04-25:
We did some testing that suggests this is only an issue when a constant is on the LHS of the operator. Will report more fully tomorrow.

hackfin, 2022-04-26:
> I feel like i am trying to ride two horses here. I have given 2 possible ways of doing it, and i am still waiting on a comment or decision on which of the two it is. Based on the decision the inbv [...]

Even though a few test scenarios might seem to be missing,  `intbv` is not broken, it's just *designed* that way, and it's necessary to follow Jan's concept to understand the innards. It's been so far mostly operations with modbvs that converted erratically to VHDL (there used to be a long related list of issues), or say: There are mismatches between MyHDL behaviour and VHDL output. But the core isn't flaky.

When you want to go for a `stricly truncated intbv` approach, you potentially open up some can, there'll be a lot of open questions to discuss, like, what happens here:

```
b = Signal(intbv()[16:]).signed()
a = truncbv(0x7f)[8:].signed() # An intbv would throw an 'out of range' on a  initial `0x80`
...
b.next = a << 1

```
The left shift would put a '1' into the MSB and flip the sign, when truncated at 8 (and when signedness is preserved, which intbv currently doesnt'). So you can't just simply truncate without nasty implicit side effects. Or being explicit, the same happening with intbv, written explicitely:

```
b.next = (a << 1)[8:].signed()
```
Likewise, when a initial `-0x80` is specified, sign extension to b should work correctly, this is where implicit truncation requires case handling.

After trying a 'next generation' approaches with direct synthesis, I found the 'augmenting approach' the most convenient one, i.e. a  `a << sh` has the length of `(a.size() + 2**sh.size())`, same for other operators as `*`, etc. Then the truncation always happens at the assignment stage (a.next = v), and Signal classes may choose to be strict on overflows or just drop a warning. So in my domain, size accounting is offloaded to the operators, which interacts just fine with intbv, on the other hand, a lot of explicit notation may be required.

But, for the shifting scenarios, there's still a complete test bench to be found.

tobygomersall, 2022-04-26:
I have had a brief look into this issue to see if we are exposed and thought it might be helpful to add my findings to this discussion.

Scenarios checked:

1. A constant shifted by a constant: Fails in the VHDL case (as described above).
2. A constant shifted by signal: Fails in the VHDL case (as described above).
3. Unsigned signal shifted by a constant: Works as I would expect.
4. Unsigned signal shifted by a signal: Works as I would expect.
5. Signed signal left shifted by a constant: Works as I would expect.
6. Signed signal right shifted by a constant: Works as I would expect.

There is a lot of discussion around shifting signals but it seems to work for me (including in the signed case). I only see the problem when I try to shift constants.

I am including some workarounds for this issue for anyone who comes across this problem in the future.

We can work around the constant shifted by a constant (including a for loop variable) issue by manually setting the value or by defining the constants outside the `@always` block.

```python
const_val = 1 << 3

@always(clk.posedge) 
def shifter():
    # Manually set the value to 1 << 3
    output_a.next = 0x8

    # Use the constant defined outside the @always block
    output_b.next = const_val

return shifter
```

We can work around the constant shifted by a signal issue by bit slicing or by defining the constant as a signal (a bit shabby).

```python
signal_const = Signal(intbv(1, 0, 2))
signal_const.driven = 'reg'

@always(clk.posedge)
def constant_shifted_by_signal():
    
    # Use bit slicing
    output_f.next[input_1] = True

    # Use the constant defined as a signal
    output_e.next = signal_const << input_1

return signal_constant_shifted_by_signal
```

rafaelcorsi, 2022-07-19:
There is other issues with `shift_left` and vhdl

```py
....
LEDR = Signal(intbv(0)[10:])
foo(LEDR, ....)

def foo(phases, ...):
    ...
    @always_comb
    def comb():
        phases[4:].next = intbv(1)[4:] << activated
```
Generates 10 bits vector, not 4

```vhd
LEDR(4-1 downto 0) <= shift_left(to_unsigned(1, 10), to_integer(stepMotor0_activated));
```

----------------

another error:

```py
phases[4:].next = 1 << activated
```

Generates:

```vhd
LEDR(4-1 downto 0) <= to_unsigned(shift_left(1, to_integer(stepMotor0_activated)), 10);
```

But the `numeric_std` does not have a `left_shift( int, int)`

https://opensource.ieee.org/vasg/Packages/-/blob/586ebeb9c3fcefd1ac9a07ce749d0e01e678503e/ieee/numeric_std.vhdl#L937