myhdl/myhdl #373: AttributeError: 'Name' object has no attribute 'value'

Grissess, 2022-07-01:
Summary

## System information
```
MyHDL Version: `0.11`
Python Version: `3.9.13 (main, May 18 2022, 22:07:34) \n[GCC 11.3.0]`
```

It's probably the simplest thing, but the error message isn't helping me find it :)

I'm playing with a toy control unit:

```
@block
def cpu(addr, data, ready, valid, clk, halt, reset):
    width = len(addr)

    state = Signal(CPU_STAGE.IDLE)
    d = Signal(modbv(0)[width:])
    s = Signal(modbv(0)[width:])

    l_op, l_out = Signal(intbv(0)[4:]), Signal(modbv(0)[width:])
    logic = logic_unit(d, s, l_op, l_out)

    a_op, a_out = Signal(ARITH.ADD), Signal(modbv(0)[width:])
    arith = arith_unit(d, s, a_op, a_out)

    c_eq, c_gt, c_sn, c_iv, c_out = map(Signal, [False] * 5)
    comp = comp_unit(d, s, c_eq, c_gt, c_sn, c_iv, c_out)

    inst = Signal(intbv(0)[16:])
    npc = Signal(intbv(0)[width:])

    reg = [Signal(modbv(0)[width:]) for _ in range(16)]

    @always_seq(clk.posedge, reset=reset)
    def tick():
        # Keep the branches of this if unmerged; it helps MyHDL realize that
        # it's actually a case analysis
        if state == CPU_STAGE.IDLE:
            if not halt:
                state.next = CPU_STAGE.FETCH
        elif state == CPU_STAGE.FETCH:
            if valid:
                ready.next = False
                inst.next = data
                state.next = CPU_STAGE.EXEC
            else:
                pc = reg[0]  # PC
                addr.next = pc
                npc.next = pc + 2
                ready.next = True
        elif state == CPU_STAGE.EXEC:
            dl = inst[4:0]
            sp = inst[8:4]
            if inst[16:14] == 0b01:
                pass  # TODO: data transfer
            else:
                code = inst[16:12]
                if code == 0b0001:
                    d.next = reg[dl]
                    s.next = reg[sp]
                    l_op.next = inst[12:8]
                elif code == 0b0010:
                    d.next = reg[dl]
                    s.next = reg[sp]
                    # FIXME: the Verilog synth can't deal with this swizzle
                    #a_op.next = getattr(ARITH, ARITH._names[int(inst[11:8])])
                    a_code = inst[11:8]
                    if a_code == 0:
                        a_op.next = ARITH.ADD
                    elif a_code == 1:
                        a_op.next = ARITH.SUB
                    elif a_code == 2:
                        a_op.next = ARITH.SHL
                    elif a_code == 3:
                        a_op.next = ARITH.SHR
                    elif a_code == 4:
                        a_op.next = ARITH.ASR
                    elif a_code == 5:
                        a_op.next = ARITH.MUL
                    elif a_code == 6:
                        a_op.next = ARITH.DIV
                    elif a_code == 7:
                        a_op.next = ARITH.MOD
                elif code == 0b0011:
                    d.next = reg[dl]
                    s.next = reg[sp]
                    c_eq.next = inst[8]
                    c_gt.next = inst[9]
                    c_sn.next = inst[10]
                    c_iv.next = inst[11]
                elif code == 0b1000:
                    offset = inst[8:0]
                    cmp = inst[12:8]
                    if reg[cmp] != 0:
                        npc.next = npc.signed() + offset.signed()
                elif code == 0b1001:
                    reg[dl].next = npc
                    npc.next = reg[sp]
            state.next = CPU_STAGE.WRITEBACK
        elif state == CPU_STAGE.WRITEBACK:
            dl = inst[4:0]
            if inst[16:14] != 0b01:
                code = inst[16:12]
                if code == 0b0001:
                    reg[dl].next = l_out
                elif code == 0b0010:
                    reg[dl].next = a_out
                elif code == 0b0011:
                    if c_out:
                        reg[dl].next = 1
                    else:
                        reg[dl].next = 0
            reg[0].next = npc  # PC
            if halt:
                state.next = CPU_STAGE.IDLE
            else:
                state.next = CPU_STAGE.FETCH

    return logic, arith, comp, tick
```

The trouble is that synthesis always fails with some variant of `AttributeError` at about the same line:

```
  File "/usr/x86_64-pc-linux-gnu/lib/python3.9/site-packages/myhdl/conversion/_analyze.py", line 986, in accessIndex
    self.visit(node.slice.value)
```

I instrumented an exception handler to tell me the innermost node it can find that was handled:

```
This appears to have been during the processing of this/these AST node(s) (the innermost one(s) found):
node:
Subscript(value=Name(id='reg', ctx=Load()), slice=Name(id='dl', ctx=Load()), ctx=Load())
(line 27)
```

... which turns out to be the `pc = reg[0]  # PC` line above.

Even commenting that literal index out (and replacing it with a constant `pc = 0`), it still chokes:

```
AttributeError: 'Name' object has no attribute 'value'
----------------------------------------------------------------------
This appears to have been during the processing of this/these AST node(s) (the innermost one(s) found):
node:
Subscript(value=Name(id='reg', ctx=Load()), slice=Name(id='dl', ctx=Load()), ctx=Load())
(line 27)
```

... which is almost certainly `d.next = reg[dl]`.

The documentation implies that this _should_ be fine, in the form of a "RAM inference"; indeed, being able to access this register file by (dynamic!) index is a critical feature, and I'd rather not manually translate this thunderous function into a `verilog_code` attribute myself--not least because I don't trust myself to :)

MyHDL's simulation alone is terse, but passes its test, and seems to be indicating that the code is _probably_ at least operating as intended; it's only the synthesis that's failing. With that said, what object is `node.slice` supposed to be that has a `value` attribute which is also an AST node--or, to the point, what's the right way to synthesize a "RAM bank" register file like this?

josyb, 2022-07-02:
Graham,

You show only a (small?) part of your code, otherwise I would have run it.

The issue is in your use of *variables* `pc`, `dl`, `sp`, `code`, `a_code`, `offset`, `cmp`.
These need to be declared and used as e.g.:
```  
    pc = intbv(0)[WIDTH_A:] # declare it outside the generators else it gets *created* at every call by the simulator

    @always_seq( clk.posedge, reset).
    def  synch():
         pc[:] = reg[0]
         address.next = pc
```

`reg = [Signal(modbv(0)[width:]) for _ in range(16)]` will correctly infer a Register File; the FPGA tools will not infer a RAM however as `reg` has an active `reset`. If you actually want the FPGA tools to infer a RAM primitive you have to write a separate `block` for this and connect it with `address`, `wr` signals and co

Best regards,
Josy

Grissess, 2022-07-03:
Hello!

It is a little large, I admit (~400LOC), with not much of it contributing to this issue, but you're welcome to [look at it in full][cpu]. (It wants an argument to run tests, and the relevant one is `cpu_test`.) There was at least an earlier workaround that might also bear some attention:

```
@block
def logic_unit(p, q, op, out):
    assert len(p) == len(q) == len(out)
    assert len(op) == 4

    width = len(out)

    @always_comb
    def comb():
        res = intbv(0)[width:]
        for bit in range(width):
            res[bit] = op[
                    q(bit) | (p(bit) << 1)
            ]
        out.next = res

    return comb

# FIXME: MyHDL can't yet seem to transpile dynamic indexing
# NB: carefully escape Verilog functions ($) with ($$) due to the use of a
# template string
logic_unit.verilog_code = '''
integer bit;
always @(p, q, op) begin: LOGIC_UNIT_COMB
    for(bit = 0; bit < $$size(out); bit = bit + 1) begin
        out[bit] = op[q[bit] + (p[bit] <<< 1)];
    end
end
'''
```

In any case, I've attempted to survey `tick` to find all the temporaries in use:

```
    # Temporaries used below
    pc = intbv(0)[width:]
    dl = intbv(0)[4:]
    sp = intbv(0)[4:]
    opcode = intbv(0)[4:]
    a_code = intbv(0)[3:]
```

(I think that's a complete list, at least!)

Wherever they were in write context, I also made it a slice assignment, as indicated, e.g.:

```
                pc[:] = reg[0]  # PC
                addr.next = pc
                npc.next = pc + 2
                ready.next = True
        elif state == cpu_stage.exec:
            dl[:] = inst[4:0]
            sp[:] = inst[8:4]
            if inst[16:14] == 0b01:
                pass  # todo: data transfer
            else:
                opcode[:] = inst[16:12]
                if opcode == 0b0001:
```

But, while simulation still works, synthesis still fails:

```
  File "/usr/x86_64-pc-linux-gnu/lib/python3.9/site-packages/myhdl/conversion/_analyze.py", line 962, in visit_Subscript
    self.accessIndex(node)
  File "/usr/x86_64-pc-linux-gnu/lib/python3.9/site-packages/myhdl/conversion/_analyze.py", line 986, in accessIndex
    self.visit(node.slice.value)
  File "/usr/x86_64-pc-linux-gnu/lib/python3.9/ast.py", line 407, in visit
    return visitor(node)
  File "/usr/x86_64-pc-linux-gnu/lib/python3.9/ast.py", line 411, in generic_visit
    for field, value in iter_fields(node):
  File "/usr/x86_64-pc-linux-gnu/lib/python3.9/ast.py", line 249, in iter_fields
    for field in node._fields:
AttributeError: 'int' object has no attribute '_fields'
----------------------------------------------------------------------
This appears to have been during the processing of this/these AST node(s) (the innermost one(s) found):
node:
Subscript(value=Name(id='reg', ctx=Load()), slice=Constant(value=0), ctx=Load())
(line 14)
```

(Admittedly, _this_ is supposed to be the error on `int`s--they get visited as if they were nodes, but are not; I think I miscopied the error info above. They do have a `value` field, though!)

[cpu]: https://gist.github.com/Grissess/177cdd6245f65c3ba2630458cb2a3709

josyb, 2022-07-03:
Hi Graham,

I downloaded the *full* code and ran it; except there were a few syntax errors ...
After correcting these it  I, just like you, got stuck on the `logic_unit`. 
I also changed some names to avoid VHDL conflicts, e.g. `out` is a restricted keyword, I replaced it with `result`

This line is the culprit:
`            res[bit] = op[ q(bit) | (p(bit) << 1)]`

It is a bit subtle but the parenthesis around `q` and `p` need to be brackets.
    q(bit) is the construct to create a ShadowSignal, and although that will work in simulation, the conversion doesn't handle the dynamic index `bit`.
    but I am sure you didn't intend to create such a ShadowSignal, but just wanted to access the single bit values

The `logic_unit` module/function can be simplified to:
```
def logic_unit(p, q, op, result):
    assert len(p) == len(q) == len(result)
    assert len(op) == 4

    WIDTH = len(result)

    @always_comb
    def comb():
        for bit in range(WIDTH):
            result.next[bit] = op[ q[bit] | (p[bit] << 1)]

    return comb
```
Note the `result.next[bit] =` construct; if you want to change a slice or a single bit in a Signal(intbv()[w:]) you must write it like this to register the value that will be set at the next cycle in the simulation process. In the other case, `result[bit].next =` you will be creating an new *dangling* object and update that, but it will not be used by any other process. Unfortunately the convertors don't care ...

And then the two tests *logic_test* and *cpu_test* run fine, and the conversion to Verilog are OK (as far as the [Sigasi ](https://www.sigasi.com)plugin to Eclipse tells me; I don't do much Verilog ...)
The *arith_test* fails simulation and throws a divide by zero exception.
But I suppose you didn't get that far yet.

Regards,
Josy

Grissess, 2022-07-03:
Good to know about the ShadowSignals! I had read about them, and may have been led down the garden path by the error that occurs when I do it "the right way." Incidentally, it's our old friend:

```
  File "/usr/x86_64-pc-linux-gnu/lib/python3.9/site-packages/myhdl/conversion/_analyze.py", line 986, in accessIndex
    self.visit(node.slice.value)
AttributeError: 'Name' object has no attribute 'value'
----------------------------------------------------------------------
This appears to have been during the processing of this/these AST node(s) (the innermost one(s) found):
node:
Subscript(value=Attribute(value=Name(id='out', ctx=Load()), attr='next', ctx=Load()), slice=Name(id='bit', ctx=Load()), ctx=Store())
(line 4)
```

The code now reads, per suggestion:

```
@block
def logic_unit(p, q, op, out):
    assert len(p) == len(q) == len(out)
    assert len(op) == 4

    width = len(out)

    @always_comb
    def comb():
        for bit in range(width):
            out.next[bit] = op[q[bit] | (p[bit] << 1)]

    return comb
```

(which does look nice, I agree, and simulates perfectly well, but fails to synth as above :)

I get the feeling you got unlucky during one of the tests! The parameters are chosen randomly, and division/mod by 0 was definitely unaccounted for. I _might_ have indeed uploaded a version that had... oh, what was it, a "vim mistake" in it, I believe... but, absent that, it passes all its tests in sim. A new version of that gist has been uploaded that _actually_ passes all the tests on my machine, and should also fix that test bug.

Josy, what version of Python are you using? (Or, if you want to test my platform's version, I have `3.9.13`.) I have a distinct dread of the culprit...

Thank you for the help so far, nonetheless :)

josyb, 2022-07-04:
On my system:
```
@block
def logic_unit(p, q, op, result):
    assert len(p) == len(q) == len(result)
    assert len(op) == 4

    WIDTH = len(result)

    @always_comb
    def comb():
        # res = intbv(0)[WIDTH:]
        for bit in range(WIDTH):
            result.next[bit] = op[ q[bit] | (p[bit] << 1)]
            # the VHDL conversion creates:
            # but the 'workaround' creates another problem: producing a negative shift value in 'concat'
            # result.next[bit] = op[ concat(p[bit], q[bit]) ]
            # res[bit] = op[ q(bit) | (p(bit) << 1)]
        # result.next = res

    return comb
```

converts to 

```
// File: verilog\logic_unit.v
// Generated by MyHDL 0.11
// Date: Mon Jul  4 08:12:08 2022

`timescale 1ns/10ps

module logic_unit (
	p,
	q,
	op,
	result
);

	input [3:0] p;
	input [3:0] q;
	input [3:0] op;
	output [3:0] result;
	reg [3:0] result;

	always @(op, p, q) begin: LOGIC_UNIT_COMB
		integer bit;
		for (bit=0; bit<4; bit=bit+1) begin
			result[bit] = op[(q[bit] | (p[bit] << 1))];
		end
	end

endmodule
```
Which is entirely correct, no?
Perhaps it has to do with the actual MyHDL version installed.
I suppose you installed with: `pip install myhdl`
Which is entirely valid, but installs a (rather) stale version from PyPi
Can you re-try by installing from github: `pip install git+https://github.com/myhdl/myhdl.git@master`
I am not sure whether the above line is correct, but I am sure you can figure that out.

One tip:
When I `import random`, I add a seed: `random.seed('We want repeatable randomness')`
It makes repeatable debugging the .vcd a lot easier

Regards,
Josy

Grissess, 2022-07-04:
You got the syntax for Git remotes in `pip` right, as far as I can tell, though the granularity MyHDL reports in versions is too small to be sure anything changed:

```
hygiea ~ # pip install git+https://github.com/myhdl/myhdl.git@master
Collecting git+https://github.com/myhdl/myhdl.git@master
  Cloning https://github.com/myhdl/myhdl.git (to revision master) to /tmp/pip-req-build-djtr73f9
  Running command git clone --filter=blob:none --quiet https://github.com/myhdl/myhdl.git /tmp/pip-req-build-djtr73f9
  Resolved https://github.com/myhdl/myhdl.git to commit ae25af4d593d20a26c85fbb17c0cd98a026e8595
  Preparing metadata (setup.py) ... done
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
hygiea ~ # python
Python 3.9.13 (main, May 18 2022, 22:07:34)
[GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import myhdl
>>> myhdl.__version__
'0.11'
```

(and I'm not sure how to check to _be sure_ what was selected for the installation of a package.)

Just for fun, I decided to try Python 3.7.13 (the lowest Py3 I have on my system), and _replicated your results_. It also works in Python 3.8.13.

... which is good! I actually have the full design as a Verilog file for now, so I'll go play with that. But something changed (with the AST?) in Python 3.9 that causes it to fail; it also affects Py3.10.4. (My distro is stabilizing Py3.10 approximately now, and I'm likely to switch to that being my host `python` when that much is done. Having the old versions around is occasionally helpful, but I do bare `python` out of habit--and don't run `venv`s, for better or worse, as you can see above--so it's a mild gripe.)

So with that in mind, I'm half-tempted to submit a PR to fix the slice AST analysis--though I'd be happier to leave that to the expertise of the maintainers, since I'm not sure I'd know exactly the _right_ way to fix it :)

(Your tip is well-received, too, but I don't mind the entropy--after all, if this were a _serious_ design with _serious_ tests, I'd be beholden to actually making them both correct and complete, rather than pull random numbers out of a hat; without such a spontaneous failure, I might never have been made aware of the test harness dividing by zero :)

Thanks again!

josyb, 2022-07-04:
> (and I'm not sure how to check to be sure what was selected for the installation of a package.)

One of the first things to do is add a *subminor* field to the version number.
Second; find out how to update PyPi on a regular basis

> But something changed (with the AST?) in Python 3.9 that causes it to fail

We have mitigated those changes in PR #358
I checked with 3.9 and 3.10; all is well

Regards,
Josy

Grissess, 2022-07-04:
Aye, yes, those look like they should fix it; I guess it's down to figuring out why my local version doesn't match [`master`'s apparently reasonable code][fixed]...

And the answer may well have been that I didn't use `--upgrade` when I tried to install over; an uninstall and reinstall seems to have pulled it in from the Git remote, and all is well in the world :)

Alright, so the real qualms are just that a stale, not-yet-3.9+-ready is on PyPI (and recommended as the install target by the documentation). I believe my _synthesis_ issues are now satisfactorily resolved, but I'll leave this open myself if you want to track the versioning blemish. (I'll have no hard feelings if you close it, though!)

[fixed]: https://github.com/myhdl/myhdl/blob/0477c7e042aee72fbd2afb722e68f6dee61c102c/myhdl/conversion/_analyze.py#L999-L1002

jck, 2022-07-07:
Unfortunately, we can't setup a GitHub action for pypi publishing cause I don't have permissions to add secrets to this repo, but I can update it manually for now after we update the docs.