myhdl/myhdl #374: vhdl wrong resize length on port

rafaelcorsi, 2022-07-25:
This:

```py
def toplevel(LEDR, SW):
    x = Signal(modbv()[2:])
    y = Signal(modbv()[2:])
    s = Signal(modbv()[2:])
    c = Signal(bool())

    ic3 = adder2(x, y, s, c)
 
    @always_comb
    def comb():
        x.next = SW[2:0]
        y.next = SW[2:0]
        LEDR[2:0].next = s
        LEDR[3].next = c

LEDR = Signal(intbv(0)[10:])
SW = Signal(intbv(0)[10:])
top = toplevel(LEDR, SW)
top.convert(hdl="VHDL")
```

Generates this:

```vhd
LEDR(2-1 downto 0) <= resize(s, 10);
```

Where should be: 

```vhd
LEDR(2-1 downto 0) <= resize(s, 2);
```

I think that the error is here:

https://github.com/myhdl/myhdl/blob/0477c7e042aee72fbd2afb722e68f6dee61c102c/myhdl/conversion/_toVHDL.py#L716-L719

Changing to `pre, suf = "resize(", ", %s)" % ori.size` solved the problem. Should I prepare a PR, or there is something more complex than this?

rafaelcorsi, 2022-07-25:
in verilog it detects that the size ir correct and dont resize, this should be done in VHDL to.

```verilog
assign x = SW[2-1:0];
assign y = SW[2-1:0];
assign LEDR[2-1:0] = s;
assign LEDR[3] = c;
```

josyb, 2022-07-25:
If you write the correct assignments:
```
        LEDR.next[2:0] = s
        LEDR.next[3] = c
```
you will get correct VHDL code

The VHDL and Verilog conversion better should throw an exception when it encounters `somesignal[hi:low].next = value` but alas they don't.

josyb, 2022-07-25:
Perhaps try simulating your code; you will see that the simulation doesn't show what you'd expect.

closed 2022-07-26 by rafaelcorsi