myhdl/myhdl #377: always_comb sensitivity list is Empty

PiyushSaini09, 2022-08-29:
Hi -
I have recently started exploring myHDL for some stuff that I was not able to accomplish in COCOTB.
While trying some examples on my own for combinational blocks, I got stuck with the this error:

[image]

While reading through the docs, what I understood is that alwas_comb does not require a sensitivity list which is true even in verilog. 
I'm confused as to why this error is shown?

[image]

Snippet of the code that I'm trying to run:
```
@block
def clk_gen(clk_en, internal_clk):
    #clk signal generation
    # instantiate the signal to 1
    clk = Signal(1)
    #clk = internal_clk only when clk_en = 1
    @always_comb
    def clk_driver():
        if (clk_en == 1):
            clk.next= internal_clk
        else:
            clk.next = 1
    
    return clk_driver
```

Please help in understanding the correct behaviour of always_comb.
Thanks!

==================================================================================
python version - 3.9.12
myHDL version - 0.11
OS - Linux

josyb, 2022-08-29:
If i run your snippet:
```python
from myhdl import block, Signal, always_comb

@block
def clk_gen(clk_en, internal_clk):
    # clk signal generation
    # instantiate the signal to 1
    clk = Signal(1)

    # clk = internal_clk only when clk_en = 1
    @always_comb
    def clk_driver():
        if (clk_en == 1):
            clk.next = internal_clk
        else:
            clk.next = 1

    return clk_driver

clk_en = Signal(bool(0))
internal_clk = Signal(bool(0))

dfc = clk_gen(clk_en, internal_clk)
dfc.convert(hdl='VHDL')
```
I get a different error:
```
Traceback (most recent call last):
  File "C:\Users\josy\myhdlsupport\issues\piyushs\alwayscomb1.py", line 26, in <module>
    dfc.convert(hdl='VHDL')
  File "C:\Users\josy\myhdlsupport\myhdl\myhdl\_block.py", line 353, in convert
    return converter(self)
  File "C:\Users\josy\myhdlsupport\myhdl\myhdl\conversion\_toVHDL.py", line 202, in __call__
    siglist, memlist = _analyzeSigs(h.hierarchy, hdl='VHDL')
  File "C:\Users\josy\myhdlsupport\myhdl\myhdl\conversion\_analyze.py", line 105, in _analyzeSigs
    raise ConversionError(_error.UndefinedBitWidth, s._name)
myhdl.ConversionError: Signal has undefined bit width: clk
```
Change `clk = Signal(1)`  into `clk = Signal(bool(1))` or `clk = Signal(True)`
But then you will get a warning for the `Clk` signal:
```
C:\Users\josy\myhdlsupport\myhdl\myhdl\conversion\_toVHDL.py:464: ToVHDLWarning: Signal is driven but not read: clk
  warnings.warn("%s: %s" % (_error.UnreadSignal, s._name),
```

BTW, you can write ` if (clk_en == 1):` simply as ' if clk_en :'   - no parenthesis required either

To obtain the desired `assign clk = (clk_en)? internal_clk : 1'b1'
 you have to write it as follows:
```python
    @always_comb
    def clk_driver():
        clk.next = internal_clk if clk_en else 1
```

PiyushSaini09, 2022-08-30:
Thanks for the help!
After making the required changes, I'm still facing the same issue.

> I get a different error:

Could it be that we are using different python versions? I've python v3.9.12 installed on my machine.

josyb, 2022-08-30:
The Python version doesn't matter - we test against CPython 3.7, 3.8, 3.9 and 3.10 and also against PyPy3.9
It will be  the actual MyHDL state; I reckon that you installed from PyPi but I am testing against the master branch. Unfortunately the PyPi version has become rather stale.
I suggest you update your local MyHDL code with `pip install -e 'git+https://github.com/myhdl/myhdl.git@master`

PiyushSaini09, 2022-09-01:
Thanks for the suggestion. I'll do this.
And I was able to run this code with python version 3.10 seamlessly.
Thanks for the help!
I'm closing this issue.

closed 2022-09-01 by PiyushSaini09