myhdl/myhdl #379: ‘Constant’ object has no attribute 'signed'

BlackJackJam, 2022-10-10:
- Python 3.8.5
- myHDL 0.11

Hi, I've come across a problem with my code. More specifically, a "‘Constant’ object has no attribute 'signed'" error was raised, but I could not locate what triggered this error. I checked my code but could not find any 'Constant'.  

error:

> AttributeError: 'Constant' object has no attribute 'signed'

In fact, I do need signed representations and I supposed that definition like _intbv()[n:]_ would not work for signed numbers, so I defined the min value and max value of my variables. I don't know whether this is a factor that causes this problem.

Here's my code:

```
from myhdl import *
  
  @block
  def neuron(spike_in, firing_threshold, syn_weight, clock,   reset, spike_out, fir_res = 16, syn_res = 8):
        m = fir_res
        n = syn_res
        membrane_potential = Signal(intbv(0, min = -2**(m-1), max = 2**(m-1)-1))
        membrane_potential_next = Signal(intbv(0, min = -2**(m-1), max = 2**(m-1)-1))
        membrane_potential_ovfl_next = Signal(intbv(0, min = -2**(m-1), max = 2**(m-1)-1))
        
        @always(clock.posedge, reset.posedge)
        def spike_result():
                if reset :
                    membrane_potential.next = 0
                    spike_out.next = False
                elif (membrane_potential_next >= firing_threshold) :
                    membrane_potential.next = 0
                    spike_out.next = True
                else: 
                    membrane_potential.next = membrane_potential_next
                    spike_out.next = False

        @always_comb
        def potential_next():
            if spike_in == True:
                membrane_potential_ovfl_next.next = membrane_potential + syn_weight
            else: 
                membrane_potential_ovfl_next.next = membrane_potential
                    
        @always_comb       
        def spike_comb():
            if ((~membrane_potential_ovfl_next[m-1]) & membrane_potential[m-1] & syn_weight[n-1]):
                membrane_potential_next.next = 2**(m-1) -1
            elif (membrane_potential_ovfl_next[m-1] & (~membrane_potential[m-1]) & (~syn_weight[n-1])):
                membrane_potential_next.next = -2**(m-1)
            else:
                membrane_potential_next.next = membrane_potential_ovfl_next
                
        return spike_result, spike_comb, potential_next

  def convert_inc(hdl, fir_res_arg = 16, syn_res_arg = 8):
    _fir_res_ = fir_res_arg
    _syn_res_ = syn_res_arg
    spike_in, clock, spike_out, reset = [Signal(bool(0)) for i in range (4)]
    firing_threshold = Signal(intbv(0, min = -2**(_fir_res_ - 1), max = 2**(_fir_res_ - 1) -1))
    syn_weight = Signal(intbv(0, min = -2**(_syn_res_ - 1), max = 2**(_syn_res_ - 1) -1))
    convInst = neuron(spike_in, firing_threshold, syn_weight, clock, reset, spike_out, fir_res = _fir_res_, syn_res = _syn_res_)
    convInst.convert(hdl=hdl)

  convert_inc(hdl = 'Verilog', fir_res_arg = 6, syn_res_arg = 4)
```

Thank you~

josyb, 2022-10-11:
I updated the master branch; so you can upgrade your MyHDL installation

BlackJackJam, 2022-10-12:
Thanks! It really helps!

josyb, 2022-11-04:
Can we close this issue?

BlackJackJam, 2022-11-04:
Sure!

closed 2022-11-04 by BlackJackJam