myhdl/myhdl #385: Dynamic Variable assignment

davekeeshan, 2022-11-30:
Summary
I have a block ram memory I am trying to write.  It is meant to be a drop in replacement for the standard block rams that vivado generates, but parameterisable and syntheisable (in addition to simulatable)

I have written it already in straight verilog and vhdl, my goal now is for the myhdl equivalent.

These memories are a little weird as they have a byte level mask.  So for data getting written in, `din`, there is another signal, `wea`, which is `len(din)/8`.

In verilog land it looks like this (this is what I am trying to get to):

```
    always @(posedge clka) begin
        integer i;
        if (ena) begin
            for (i=0; i<G_WEWIDTH; i=i+1) begin
                if (wea[i]) begin
                    f_ram[addra][(8 * i)+:8] <= dina[(8 * i)+:8];
                end
            end
        end
    end
```
However when I try and write some thing that work in python/myhdl:

```
    @always(clka.posedge)
    def write():
        if ena:
            for i in range(len(wea)):
                if wea[i]:
                    f_doutb[addra].next[(8*i)-1:8*i] = dina[(8*i)-1:8*i]                    

```
The code this generates:
```
always @(posedge clka) begin: BLOCKMEM2P_WRITE
    integer i;
    if (ena) begin
        for (i=0; i<4; i=i+1) begin
            if (wea[i]) begin
                f_doutb[addra][((8 * i) - 1)-1:(8 * i)] <= dina[((8 * i) - 1)-1:(8 * i)];
            end
        end
    end
end
```
Will not go through icarus:

```
../myhdl/blockmem2p.v:50: error: A reference to a wire or reg (`i') is not allowed in a constant expression.
../myhdl/blockmem2p.v:50: error: Part select expressions must be constant.
../myhdl/blockmem2p.v:50:      : This lsb expression violates the rule: ('sd8)*(i)

```
Checking verilog spec, it will not accept a variable as the second argument of an array.

Python arrays doesn't allow you to specify a start index and an number of values to take, only `[start, stop, step]`.  Is it even possible to achieve code like this?

I have set up a working/failing example [here](https://github.com/daxzio/blockmem2p), here is a failing github action (using cocotb) compared against the working verilog only version. Checkout the [action](https://github.com/daxzio/blockmem2p/actions/runs/3584439012)

## System information
```
MyHDL Version: 
Current main branch on github

Python Version: 
3.7-3.11

```

josyb, 2022-11-30:
Hi Dave,

the MyHDL statement ` f_doutb[addra].next[(8*i)-1:8*i] = dina[(8*i)-1:8*i]`
should be: `f_doutb[addra].next[(i + 1) * 8:i * 8] = dina[(i + 1) * 8: i * 8]`
Q: have you simulated your design first, before converting?

Regards,
Josy

davekeeshan, 2022-11-30:
Sorry I typo-ed out.  Yes it should be as you said, however the error reported is the one I was getting to.

    f_doutb[addra].next[(i + 1) * 8:i * 8] = dina[(i + 1) * 8: i * 8]
To

    f_doutb[addra][((i + 1) * 8)-1:(i * 8)] <= dina[((i + 1) * 8)-1:(i * 8)];

Still gives the same icarus error. It is the i in the second argument that is the probelm, verilog does not recognise it.

I have not simulated this in myhdl yet, I had the verilog/cocotb testbench already written for something else.

I will try and get a myhdl testbench going for prosperity

josyb, 2022-11-30:
Hi Dave,

>Still gives the same icarus error

Does this mean we need to convert 
`f_doutb[addra].next[(i + 1) * 8:i * 8] = dina[(i + 1) * 8: i * 8]` 
to 
`f_doutb[addra][(8 * i)+:8] <= dina[(8 * i)+:8];`?

I also checked  a Verilog LRM (2001) and now see what goes wrong.
And it is effectively a bug then. Apologies for being too quick earlier on.
I think it can be fixed relatively easy,  the `+:` is already applied for some specific cases (see \_toVerilog.py line  1237). We should be able add this *variant*.

Regards,
Josy

davekeeshan, 2022-11-30:
That's OK, it hard to pull all the pieces together in the first place.

Your answer if mostly right, however I don't know if there is intrinsically enough information in the python to derive the verilog. That 8 amount looks sorta obvious here, but what about:

    f_doutb[addra].next[(i + 1) * 8:i * 4] = dina[(i + 1) * 8: i * 4]

While unusual, is still pythonically legal, but I am not too sure what the equivalent verilog should be with the restrictions all ready discussed.

I don't have a answer here, just trying to describe the problem.

josyb, 2022-12-01:
Or this one:
```
            for j in range(5):  # major y
                for i in range(5):  # major x
                    for l in range(4):  # minor y
                        for k in range(4):  # minor x
                            pisodata[(j * 4 + l) * 20 + i * 4 + k].next = self.Raw[j * 5 + i][(l * 4 + k + 1) * 12:(l * 4 + k) * 12]
```
VHDL has no issue with this one nor with the *asymmetrical* case you just showed above. I doubt that you can transcribe your expression into valid Verilog?
The fix for this would be to check the second part of the slice, ascertain that it is an ast.BinOP with an ast.Mult as operator and use two the two leaves to build the Verilog output (which will actually work as well for my *more complex* expression). If the MyHDL code deviates from this form: ` [(a +1) * b : a * b]` # with **a** being a variable expression but **b** a constant, then you will have no pie ... We can do some basic checking on the form of the first part of the slice but evaluating the expression may turn out to be impossible because the conversion process is done on a line per line basis without keeping context (more or less)

davekeeshan, 2022-12-01:
Yes, again I have no good answer.  The language paradigms seem completely at odds.

I am surprised how limited verilog is here (because that seems to be where the problem lies).  In fact I didn't know about the `+:` way of doing it until this week, this is the way I used to do it:

    generate
    genvar i;
        for (i = 0 ; i < G_WEWIDTH ; i=i+1)
        begin : g_write_mem
            always @(posedge clka)
            begin
                if (ena & wea[i])
                    f_ram[addra][(8*i)+7:8*i] <= dina[(8*i)+7:8*i];
            end
        end
    endgenerate

Is there a way of doing something like that in python, a generate loop, that also works with VHDL too?

josyb, 2022-12-02:
Hi Dave,

The generate variable `i` will infer a constant for each invocation, getting you of the hook. But I don't like `generate` constructs.

I experimented a bit and in stead of trying to decipher the `slice' I introduced a new way of specifying this:
```python
    @always(clka.posedge)
    def write():
        if ena:
            # storage[addra].next = storage[addra]
            for i in range(NBW):
                if wea[i]:
                    storage[addra].next[(i, 8)] = dina[(i, 8)]
```
I use a tuple to specify the index and the width; added code to __getitem__ and __setitem__ in _intbv.py; added code to parse the tuple in _toVerilog.py.
The tuple has 2 or 3 elements: index, width and optional offset; this allows us to slice a large incoming vector.

The result:
```Verilog
always @(posedge clka) begin: RAMBW_WRITE
    integer i;
    if (ena) begin
        for (i=0; i<4; i=i+1) begin
            if (wea[i]) begin
                storage[addra][i * 8 +: 8] <= dina[i * 8 +: 8];
            end
        end
    end
end
```

I used an anonymous tuple,  but contemplate to use a `namedtuple` object to better *document* what's going on.
So far I only checked this minimal example for conversion to Verilog only; I have to check simulation and add conversion to VHDL.

What are your thoughts about this?
@hgomersall Would you be so kind to read over this all and share your thoughts?

Regards,
Josy

Hint: if you specify the language after the starting three backticks you get nice syntax coloring (e.g.: \```Verilog)
Confession: I use [(i + 1) * W : i * W] quite a lot, but (luckily?) only convert to VHDL

josyb, 2022-12-05:
@hgomersall 
Yes, but I see it rather as slice assignment by a slice
First : it is a bug in the conversion of slicing: [(i + 1) * W : i * W] to Verilog fails. The output should be [i *W +: 8] but MyHDL generates simply [(i + 1) * W : i * W] 
Dave encountered this issue when trying to create a RAM with Byte-Enables.
Second: I present a workaround because trying to parse the AST to detect the `*W` seems too complicated (for me and for now ...) as there are many forms possible, e.g. : [Offset +(index +1) * WIDTH : WIDTH * index + Offset] where Offset and index can be variable expressions and WIDTH is a constant expression. Also see the *double indexed* example I showed earlier.
The workaround would be to introduce a `custom slice` method by declaring a namedtuple: `vectorelement(index, WIDTH, Offset=0)`.  BTW this construct could also be used to zip/unzip vectors e.g. in to odd and even and vice versa:
```python
for i in range(4):
    v.next[vectorelement(i,2)] = even[i]
    v.next[vectorelement(i,2,1)] = odd[i]
```