myhdl/myhdl #386: Support for switch in MyHDL

davekeeshan, 2022-12-05:
Since python 3.10, there is now support for case statements that more align with case statements in both VHDL and Verilog.

Raising this issue to track, potential, implementation in MyHDL.

[Python Switch](https://www.freecodecamp.org/news/python-switch-statement-switch-case-example/)

josyb, 2022-12-06:
Hi Dave,

I had part of this discussion with Steve, a MyHDL user in NZ already:
I copy the conversation (being a bit lazy to write a long answer)

Steve:
>Python 3.10 has a match statement - will we now get a case statement for
myHDL?? :-)

> https://learnpython.com/blog/python-match-case-statement/

Me:
>Perhaps, (probably?). And if, it will be a 'restricted' version like:
```python
match( smp ):
    case machine_states.IDLE:
        if something:
            smn.next = machine_states.BUSY
        else:
            smn.next = machine_states.IDLE 
    case machine_states.BUSY:
        ...
    case other: # optional; in VHDL we expect to cover all states explicitly; in Verilog (I don't exactly know ...)
        ... 
```
>The main (and only?) advantage is that the MyHDL code then better resembles the V* code.

Steve:
>Yes - but for decoding bit patterns also?  This is what I really want it for - creating complex if statements is less readable.  This example belwo is somewhat trivial and doesnt show the advantage clearly. but think microcontroller instruction decode for example

```python
match (concat(a,b,c)):
    case intbv(0b010):
        m.next = 0
    case intbv(0b101):
        m.next = 2
    case other:
        m.next = 3
```
Me:
>That would also work.
But for the case of microprogramming I would be tempted to try a look-up table.

Steve:
>Yes, it better resembles the v* code but it reads better than a set of if statements (IMHO), and probably compiles to a better result also.

Me:
>the 'if-elif-else chain' is converted into a V* case; the 'match', by nature, will convert to a V* case as well; but there will be no difference between the two results ... 

I have no issue with a future implementation of Python's `match` construct but I believe we can find a few other things to improve

davekeeshan, 2022-12-06:
Yes it is good to track this, 3.10 is only just starting to catch on. 

Many years ago there was a difference in RTL between synthesising a fully qualified if statement and a fully qualified case statement, but those two worlds might be a lot closer now based on what the compiler tools recognise, but having the option to do either I think is still important, for synthesis. 

https://electronics.stackexchange.com/questions/73387/difference-between-if-else-and-case-statement-in-vhdl

On a side note do you have a priority list for features?

josyb, 2022-12-07:
>On a side note do you have a priority list for features?

Not really, I can think of one but first we need a place to post such a list so that you and others can contribute
The first one I have is to add a subminor to MyHDL version number which gets incremented at every merge. This is useful to give support as then it is easier to find out that their issue is already fixed (or not)

davekeeshan, 2023-02-03:
Thank you for merging the code, However I think that is only part of this issue.  I would like to discuss the deprecation path.  I personally and very keen to make the if -> if and match -> case as the default.  I don't think it should functionally change anything from a testing perspective, however the synthesisable code is more what I am aiming for here