myhdl/myhdl #388: verliog generation of constants is strange

olisnr, 2022-12-07:
myHDL is perfekt to simultate and generate verilog for FPGAs. 
myHDL constants will be converted in the verilog code to numbers. this is not so nice, but works in most cases. But today i had problems whit one constant in a verliog if-statment. in myHDL the constant was "CLK_DiV =  4" and the if-statement "elif clkdiv == CLK_DiV // 2:". So i would expect an if-statment that compares to 2 in verilog, unfortunatly the generator produced: "(-'h1): begin" and the syntesizer optimized this out... i tried different things:
            elif clkdiv == CLK_DiV // 2: # gives "(-'h1): begin"
            elif clkdiv == math.ceil(CLK_DiV / 2): # gives error: "Not supported: true division - consider '//'"
            elif clkdiv == 2: # gives "'h2: begin"
            elif clkdiv == CLK_PH2: # gives "'h2: begin" this works and using "CLK_PH2 = CLK_DiV // 2"

## System information
```
MyHDL Version: 0.11 from "pip install git+https://github.com/myhdl/myhdl.git@master"
Python Version:  3.9
```
my full code is:

#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""
Created 2021

@author: omunz
"""
import math

import random
from myhdl import block, always, instance, Signal, delay, StopSimulation, intbv, instances, ResetSignal, always_comb, always_seq

RESET_ACTiVE =  0
ADC_CHiPs =     4
ADC_DATA_BiTs = 18
ADC_SCLK_CONV = 16  #500ns
CLK_DiV =       4   #max 32MHz SCLK
CLK_PH1 =       0
CLK_PH2 =       CLK_DiV // 2

CLK_DIV_WiDE = math.ceil(math.log2(CLK_DiV))
CLK_CNTR_WiDE = math.ceil(math.log2(ADC_DATA_BiTs + ADC_SCLK_CONV + 3))

@block
def qaud_adc_spi(    clk,
                        reset_n,
                        enable,
                        
                        sclk,
                        conv,
                        to_adc,
                        from_adcs, 
                        
                        data_out0, 
                        data_out1, 
                        data_out2, 
                        data_out3, 

                        data_out_new
                    ):

    #intern
    shifts =    [Signal(intbv(0)[ADC_DATA_BiTs:]) for _ in range(ADC_CHiPs)]
    # cntr =      Signal(intbv(0, max=ADC_DATA_BiTs + ADC_SCLK_CONV + 3))
    # clkdiv =    Signal(intbv(0, max=CLK_DiV))
    cntr =      Signal(intbv(0)[CLK_CNTR_WiDE:]) #bessere darastellung in gtkWave
    clkdiv =    Signal(intbv(0)[CLK_DIV_WiDE:])

    @always_seq(clk.posedge, reset=reset_n) #only one clk all synch
    def shift():
        data_out_new.next = 0
        to_adc.next = 1
        if enable or cntr != 0:
            clkdiv.next = (clkdiv + 1) % CLK_DiV
            if clkdiv == CLK_PH1:
                conv.next = cntr < ADC_SCLK_CONV
                cntr.next = cntr + 1
                sclk.next = 0
                if cntr == ADC_SCLK_CONV + ADC_DATA_BiTs + 1: #trasfer end
                    data_out0.next = shifts[0]
                    data_out1.next = shifts[1]
                    data_out2.next = shifts[2]
                    data_out3.next = shifts[3]
                    data_out_new.next = 1
                    cntr.next = 0
                    clkdiv.next = 0
#            elif clkdiv == CLK_DiV // 2: # gives "(-'h1): begin"
#            elif clkdiv == math.ceil(CLK_DiV / 2): # gives error: "Not supported: true division - consider '//'"
#            elif clkdiv == 2: # gives "'h2: begin"
            elif clkdiv == CLK_PH2: # gives "'h2: begin"
                if cntr > ADC_SCLK_CONV:
                    for i in range(ADC_CHiPs):
                        shifts[i].next[ADC_DATA_BiTs:1] = shifts[i][ADC_DATA_BiTs-1:]
                        shifts[i].next[0] = from_adcs[i]
                    sclk.next = 1
        else:
            clkdiv.next = 0

    return instances()

olisnr, 2022-12-08:
[QUAD_ADC_SPi.zip](https://github.com/myhdl/myhdl/files/10184278/QUAD_ADC_SPi.zip)

olisnr, 2022-12-08:
why do You think its not a bug? shoulden't generate all "elif clkdiv == " varinants the same "'h2: begin"? why is  "(-'h1): begin" ok?

josyb, 2022-12-08:
I didn't say that `(-'h1):` is OK.

I am checking the code; the VHDL conversion is correct, so it looks like it is bug after all. It is a bit odd that this behavior hasn't been reported before (perhaps it has, one can only remember so much). I guess that most use the `clkdiv == CLK_PH2` method.
> MyHDL constants will be converted in the verilog code to numbers. this is not so nice, but works in most cases.

I feel sympathetic to your wish for symbolic constants - in which case you would have used  `clkdiv == CLK_PH2` :)
A workaround to do this is to declare:`CLK_PH2 = Signal(intbv(CLK_DiV // 2)[CLK_DIV_WiDE:])`
But unfortunately this will generate a warning that CLK_PH2 is not driven, and we rather have no warnings ...
I am experimenting on generating constant declarations in VHDL (as this is what I use) and then do the same for Verilog (which I don't know much of) 

Quite often the reported bugs aren't a bug, so it would be good etiquette to leave the qualification of bug to the moderators.

josyb, 2022-12-08:
> my experience whit the conversion is: bigger designs, that are exported as one big VHDL-module wont compile, in the FPGA-SW, while the same design exported as verilog will. so i switched to verilog.

I have generated VHDL files of 33000 lines in length - no issue
And I will never switch to Verilog - SystemVerilog perhaps

josyb, 2022-12-08:
> if i shoulden't report as bug, should i write "maybe bug" in the title? and mark as question?

The title as you put is fine.

olisnr, 2022-12-08:
> I have generated VHDL files of 33000 lines in length - no issue And I will never switch to Verilog - SystemVerilog perhaps

i also wouldn't switch to verilog - as long there is myHDL :) but i think it would be interesting to switch from Python to Nim, because of the compile-time error checking.

josyb, 2022-12-08:
> but i think it would be interesting to switch from Python to Nim, because of the compile-time error checking.

That would be a total new project, and a massive as well ...
The thing about MyHDL (note the capital M at the beginning) is that the simulation (MyHDL is first a simulation language, conversion comes second) is actually running Python: our BFDL has ingeniously crafted MyHDL and the inner workings are quite deep matter - at least to me they are/were, I am now fully appreciating his original work.

Back to our muttons: I will have a look what goes wrong in the Verilog conversion; if something works for VHDL it should evenly work for Verilog and vice versa

olisnr, 2022-12-20:
thanks for Your work. the "(4 / 2): begin" seems to be working whit "synplify for microchip".

closed 2023-04-26 by olisnr