myhdl/myhdl #393: MIgrate PyPi release to github actions

davekeeshan, 2022-12-10:
I release this is probably a trackable issue.

Work that needs to be done to enable myhdl PyPi release from github

- [ ] Get PYPI_API_TOKEN from PyPi
- [ ] Setup github actions for release
- [ ] Demo with a minor rev
- [ ] (Minor) Cleanup readme used by PyPi (currently says that there is only python 3.4 support)

davekeeshan, 2022-12-10:
@josyb In PyPi

[image]

Just pick one for the project:

[image]

davekeeshan, 2022-12-16:
The secret need to be recorded here, it needs to be call `PYPI_API_TOKEN`:

[image]

(Don't worry this secret has been edited, but it looks enough like what it will look)
[image]
After you set it, you can never see it again via github interface, even if you entered it.

davekeeshan, 2022-12-16:
Here is a checkin with a tag that has a working pypi release:
[image]

The one with the tag is a failure since I don't have the proper PYPI_API_TOKEN

[image]

But the rest looks good

josyb, 2022-12-17:
[image]

josyb, 2022-12-17:
@davekeeshan 
I still would like to see a sub-minor added to the version as: 0.11.42, which we increment at each successive merge
Can we have a file in the GitHub repository, say `_version.py`, which is protected from being updated by any PR but gets updated on a successful merge?

davekeeshan, 2022-12-17:
> @davekeeshan I still would like to see a sub-minor added to the version as: 0.11.42, which we increment at each successive merge Can we have a file in the GitHub repository, say `_version.py`, which is protected from being updated by any PR but gets updated on a successful merge?

@josyb I am not too sure if this is possible with actions.  An action runs on an event, like a pull request (the emphasis here is on the word request).  It runs, completes and reports a pass/fail, and then it is finsihed.  Now the merge happens after this, or at all, you as the repo owner may decide to refuse the merge even.

Let me think about this, it may be possible some how. Howevr, if you don't mind can we at least do the first one or two manually, there are a few niggles the the flow I need to resolve (closing the loop with PyPi) before I would even want to attmpet that.

I am just about to do another merge request.  It cleans up the action a little bit, no task is run for the PyPi release if it is not a tag.  Also I have updated the make file so that the command `make release`, will read the contents of the `__version__` variable in `myhdl/__init__.py` and use this value to set the tag.  If you want to change or bump the tag to anything else (major or minor) Edit the file to the new value (don't worry about committing it, as it will be done as part of the make, however you can commit if you want) or you can run the command `make release TAG=0.11.3`, it edits `myhdl/__init__.py` checks in that file, adds a tag and pushes the lot.