myhdl/myhdl #405: Constant value inversion (~0x1, ~0x4523) not working in VHDL

davekeeshan, 2023-01-19:
When in python some one used an inversion of a constant (~0x1, ~0x4523 etc) and `ANDs` it to another vector, the VHDL version fails
```.py
    a   = Signal(modbv(0)[8:])
    z   = Signal(modbv(0)[8:])
    
    z.next = a & ~0x1
```
This work fine in verilog as it sytax is almost identical:
```v
    assign z = (a & (~1));
```

However the current resolving for VHDL does not work:
```vhd
    z <= (a and to_unsigned(not(1), 8));
```
The not on the constan value, the 1 here, does not work

Recommend recalculating the correct inverted value and storing that instead:
```vhd
    z <= (a and to_unsigned(254, 8));
```

With special attention made to make sure that the width is correct for the assigning signal

## System information
```
MyHDL Version: master
Python Version: 3.10
```

davekeeshan, 2023-01-19:
Example of failing test [link](https://github.com/daxzio/myhdl/actions/runs/3960940762/jobs/6785777865#step:10:487)

josyb, 2023-01-19:
I rather see that the *not* is kept - to show the intent of the designer :)
You will see / have seen that the VHDL conversion is complex.

davekeeshan, 2023-01-19:
@josyb 

I respect that in some ways, but when I see it it is usually as an act of lazyness by the original coder. The whole act of generating code from code is a translation. I would veer to the side of readable code versus turgid. 

Can you send on an example of the more complex vhdl to do this?

josyb, 2023-01-19:
`z <= a and not(to_unsigned(1, 8));`
but as I said the VHDL convertor takes work to make it correct;
e.g. I am suffering to make this work: 
```python
    last_slot = Signal(False)
    next_pc = Signal(intbv(0)[WIDTHA:])

    def precalc_logic():
        lit_count = 1
        lit_offset = 0
        for i in range(OPCODES):
            if ((ir >> (i * ISA)) & ISA_MASK) == 0x10:
                lit_count = lit_count + 1
                if (i <= slot):
                    lit_offset = lit_offset + 1
        pc_offset.next = pc + lit_offset
        instr.next = (ir >> (slot * ISA)) & ISA_MASK
        last_slot.next = (slot >= (OPCODES-1))
        next_slot.next = 0 if last_slot else (slot + 1)
        next_pc.next = pc + (lit_count if last_slot else 0)
``` 
The last line converts to:
 ` next_pc <= resize((resize(pc, 19) + tern_op(cond => last_slot, if_true => lit_count, if_false => 0)), 18);`
which doesn't pass ...
It didn't work for Verilog either, but fixing Verilog was easy; VHDL cumbersome and not yet working ...

davekeeshan, 2023-01-19:
I don't understand why this is the same thing, there is no `~` in that code.

I think I have seen a problems like this elsewhere though.

josyb, 2023-01-19:
No, you are right, it is not the same thing.
I was trying to emphasize that VHDL takes more work to get the conversion right whereas the Verilog conversion is easier (compare the size of the two modules)
BTW: the Verilog  code: `next_pc = ($signed({1'b0, pc}) + $signed({1'b0, last_slot}) ? lit_count : 0);` it passes, but doesn't look great either: IMO we could do without that `$signed` stuff.

davekeeshan, 2023-02-03:
Code has been merged

closed 2023-02-03 by davekeeshan