myhdl/myhdl #406: Python3.7 Deprecation list

davekeeshan, 2023-02-03:
This is to track all the things that need to be changed when 3.7 is deprecated:

- [ ] Change ast.Num to ast.Constant

> Deprecated since version 3.8: Old classes ast.Num, ast.Str, ast.Bytes, ast.NameConstant and ast.Ellipsis are still available, but they will be removed in future Python releases. In the meantime, instantiating them will return an instance of a different class.

josyb, 2023-02-03:
Why is it a bug?
We have covered `ast.Num` and `ast.NameConstant`; `ast.Bytesd` and `ast.Ellipsis` are not handled
conversion/_analyze.py:
```python
    if sys.version_info >= (3, 9, 0):

        def visit_Constant(self, node):
            node.obj = None  # safeguarding?
            # ToDo check for tuples?
            if isinstance(node.value, int):
                # Num
                if node.value in (0, 1):
                    node.obj = bool(node.value)
                else:
                    node.obj = node.value
            elif node.value in (True, False, None):
                # NameConstant
                node.obj = node.value
            elif isinstance(node.value, str):
                # Str
                node.obj = node.value

    else:

        def visit_Num(self, node):
            n = node.n
            # assign to value attribute for backwards compatibility
            node.value = n
            if n in (0, 1):
                node.obj = bool(n)
            elif isinstance(n, int):
                node.obj = n
            else:
                node.obj = None

        def visit_Str(self, node):
            node.obj = node.s

        def visit_NameConstant(self, node):
            node.obj = node.value
```

davekeeshan, 2023-02-03:
In the Pull release I just did I was using the `ast.Constant` to do a test to make sure the code gets written correct.

`ast.Num` is being used currently in a number of locations in `to_VHDL.py` and `to_verilog.p`y.

I have enhanced the code to break them out for the version number

in `visit_Call` for example, there are others:
```py
            if sys.version_info >= (3, 8, 0):
                if isinstance(node.args[0], ast.Constant):
                    node.args[0].n = int(node.args[0].n)
            else:
                if isinstance(node.args[0], ast.Num):
                    node.args[0].n = int(node.args[0].n)
```

There are also tests on `ast.Str`, which I haven't figured out what the replacement is yet.

josyb, 2023-02-03:
Then the *test suite* missed out on some of these, otherwise I would have repaired them in #350, I repaired one not tested but flagged by a colleague in #380

davekeeshan, 2023-02-03:
Well I think i have addressed all the `ast.Num` in my most recent PR, I tried to match the style you were using, however please accept the merge and change accordingly as the inversion bug it is fixing is also tied up in there

Will you look at the `ast.Str` too?

There are no `ast.NameConstant` or `ast.Bytes` in the code base

josyb, 2023-02-03:
`ast.Str` is covered in _analyze
We will see later

davekeeshan, 2023-02-03:
```

grep -r ast.Str
myhdl/conversion/_toVerilog.py:        if isinstance(expr, ast.Str):
myhdl/conversion/_toVHDL.py:        if isinstance(expr, ast.Str):
myhdl/conversion/_toVHDL.py:                if isinstance(a, ast.Str):
myhdl/conversion/_toVHDL.py:        elif isinstance(node.op, ast.Mod) and isinstance(node.left, ast.Str):  # format string
```

josyb, 2023-02-03:
I have to see the context.
But it is obvious that the *tests*  don't seem to find these.

davekeeshan, 2023-02-03:
Well they are not deprecated out of python yet, so they are still valid all the way up to 3.11 it would appear, maybe they were found to be quite extensively used?

josyb, 2023-02-03:
They are removed in 3.9, see  #350

davekeeshan, 2023-02-03:
really, OK that is weird, hmmm....

A code overage tool would should what the tests are hitting and what they are now, maybe there is no code in the test suite hitting these cases, strange.

josyb, 2023-02-03:
I fixed (tried to) in #358 first, then  in #380
So there will be a few more to come, once we have some user getting caught ... we better try to add some test case for it